arendst/Tasmota #3756: Enable AD0 in sensors.bin

pmknowles, 2018-09-10:
**Have you look for this feature in other issues and in the wiki?**Y

**Is your feature request related to a problem? Please describe.**N
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
It would be useful if the sensors.bin had AD0 enabled as default rather than having to compile my own bin just for that feature

**Describe alternatives you've considered**
Compile own bin

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-10:
Make sense.

PR #3757 made with your proposed change for Theo to review it.

andrethomas, 2018-09-11:
Confirmed functional after build 7b3ceb9

closed 2018-09-11 by arendst