arendst/Tasmota #3758: Using sht3X sensor in hom assistant?

jascdk, 2018-09-10:
Hi - Home assistant won’t let me use the sensor - it is saying error . I think it is because it’s name is sht30x-something .

My bme280 is fine 

Can the name of the sht30x just be that?

ascillato, 2018-09-10:
Please, complete the troubleshooting template in order to properly help you.

ascillato, 2018-09-10:
### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

ascillato2, 2018-09-10:
Hi,

When you have time, please provide all the information requested and your HA configuration

closed 2018-09-10 by ascillato2