arendst/Tasmota #3759: Disconnect Button from Relay (Using Different MQTT Topics)

kellertobias, 2018-09-10:
I find the terminology quite confusing. Switch vs. Button. Switchmode. Nothing seems to be what I want to do.

What I plan to to: Connect Multiple Sonoffs Basic and Sonoff Touch 2ch/3ch to openHAB (or another home control system) using MQTT.

However, The buttons on the sonoffs should not necessarily control directly the relays on the same device. e.g. I have a switch for my room (a 3ch one) even so, I have only 2 outgoing wire connections to ceiling lamps. But I also have some LED strips and other lamps, I want to control using mqtt.

I want the button 1 to e.g. control the main light (relay 1 on the same device) as well as some led strip. Button 2 is an "All off" switch.

So I want different topics per device (I have written a firmware by myself, but it does not work stable, so I want to switch to tasmota):

home/ceiling/relay1 < ON or OFF
home/ceiling/relay2 < ON or OFF
home/ceiling/btn1 > TOGGLE
home/ceiling/btn1 > TOGGLE

ascillato, 2018-09-10:
Great!

Tasmota can perform all the tasks you need.

All the information you required is already in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki). Also there are examples.

Please, if you need more information that you can not find in the wiki, please search for that in the issues, in the forum and in the Tasmota chat.

Support Information:

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

Thanks

closed 2018-09-10 by ascillato2

jgalak, 2019-02-23:
I have essentially the same problem - I have a NodeMCU flashed with Tasmota.  There's a switch on one GPIO pin (designated as Switch 1) and an RGB LED on 3 other pins (designated as PWM1, 2, 3, with SetOption15 letting me control color).  I want the two to be completely separate - the state of the switch should not affect the LED in any way.

Can't figure out how to do it - I tried changing the Switch from 1 to 2, and even to 5, behavior persists.  

How can I dissassociate the two?

Please don't just post a link to the wiki - I've been all over it, and haven't figured it out.  Maybe it's me, maybe it's not clearly explained.  If it's on the wiki, where, specifically?  

Unfortunately, while the wiki has very complete info on various commands and options, it seems to have very few explanations of what the commands and options are **for** - simply stating that option X gives effect Y doesn't always help determine why one would want Y.

houseofbugs, 2019-05-29:
Seriously. Why is this so hard to get help on? Why is there not a clear article on separating the switch from a relay. All I see is the same "go read the wiki" response from everyone. It's not in there. I've been looking for 4 hours now.

ascillato, 2019-05-29:
The full explanation is in the following article:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#16-using-an-external-button-with-single-press---double-press-and-hold

You also have the Tasmota support chat on Discord for asking questions.

There is no need for complaining. Just ask nicely. Tasmota is given to you and supported by us (the Tasmota community) just for free.

jabbas, 2019-07-15:
> The full explanation is in the following article:
>
> https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#16-using-an-external-button-with-single-press---double-press-and-hold
>
> You also have [...]

Still, preparing so called "Common tasks/jobs/issues" on the Wiki will solve lot's of "how to" questions.
Even empty, will fill out with time. For hard-core users it's easy, since they know Tasmota from the ground up, for new one (even smart ones) it is not.

hrmele, 2020-04-22:
You see, I am reading this topic one year later, and because nobody answered, just linked random pages, this topic doesn't have any answers, as all mentioned links are broken...
Seriously, direct answers... One user question on a github page is not just that single user doubt.

tomlut, 2020-05-08:
Link is broken.

meingraham, 2020-05-08:
July 2019. Docs have moved twice since then.

https://tasmota.github.io/docs/Rules/#button-with-single-press-double-press-and-hold

bibikiwi, 2021-07-25:
https://tasmota.github.io/docs/Buttons-and-Switches/#setoption114

to detach all switches :

     setoption114 1

to detach only a specific switch (e.g. switch 3)

     switchmode3 15

UnbridledGames, 2021-10-11:
Thats great and all but detaching all SWITCHES from the relays does nothing when your wall switch has a button, and not an actual switch.

I've been trying for HOURS to make the BUTTON ONLY SEND a mqtt, and ignore the relay. Literally nothing I have tried has worked. I found this, read it all, saw Option114 and got excited... and it only applies to switches and not buttons.

I want a button to instantly send a MQTT without waiting to see if I double or triple click it. And not touch the relay. And nothing I try will do that.

My GOAL is to have a button that instantly toggles the relay when the button is RELEASED, but if it's held for more than X time before it's released, it does something else. I saw an issue elsewhere that talked about the option that disables the multi-click and hold "because to instantly toggle the relay you can't then check to see if it's held".

Yes you can. Toggle the relay if the button is RELEASED before the hold duration. The relay doesn't need to flip when the button goes down. No one holds these buttons down unless they are HOLDing them. Toggle the relay when the button is released.

I'm thinking I'm going to have to custom compile my own build. This is a SIMPLE thing.

barbudor, 2021-10-11:
@UnbridledGames This is a closed issue that was dealing with the specific problem of it's author
DO NOT HIJACK a closed issue

If you have a qualified bug in Tasmota to report, create a new Issue and fill ALL required information (otherwise the issue will be automatically closed and rejected). In case of doubt if the bug in Tasmota or PEBCAK, follow the next recommendation.
Issues are meant for bug report only.

If you want Help in configuration or Usage, create a Discussion or join Discord support server. In every case we will need a clear explanation of your setup and goal (so most likely same as for an Issue).

sfromis, 2022-05-12:
Do not reincarnate old threads to ask unrelated questions. And do not ask questions in "issues", "discussions" or the Discord server is better for that.