arendst/Tasmota #3761: i am using sonoff touch 3 gang module

akshaydudhagara, 2018-09-11:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-09-11:
Hi,

> i am using sonoff touch 3 gang module and i want to know about sonoff touch is normally pull up/pull down..?? 

internal pull up resistor by default.

> when i press to touch than what it will give me as output means it gives me high / low when i am trying to read that pin..??

low when press. Please, check this with a DMM.

closed 2018-09-11 by ascillato2

akshaydudhagara, 2018-09-12:
ok thanks, but i m getting some different kind of result when i am debugging it.
i got a result as continues changing its state without touch from 0 to 1.
so what should i do..?