arendst/Tasmota #3762: Sonoff Tasmota MQTT reporting wrong MAC address

PerfectCrime, 2018-09-11:
The reported APMac value from a Sonoff device is 1 higher than what arp -a reports.
I  have 3 access points at home with which I tested this behavior, all report MAC address + 1.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : Sonoff Basic/ Sonoff S20
- [X] _Provide the output of command_``status 0`` :
```
12:45:01 CMD: cmnd/son_02/status 0
12:45:01 MQT: stat/son_02/STATUS = {"Status":{"Module":1,"FriendlyName":["Garage voorkant"],"Topic":"son_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:45:01 MQT: stat/son_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"6T11:38:18","StartupUTC":"2018-09-04T23:06:43","Sleep":0,"BootCount":13,"SaveCount":131,"SaveAddress":"F9000"}}
12:45:01 MQT: stat/son_02/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-21T10:01:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:45:01 MQT: stat/son_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["3992SK_2","3992SK_2"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
12:45:01 MQT: stat/son_02/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":408,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FCAE794","00000000","23B6179E","00000000"]}}
12:45:01 MQT: stat/son_02/STATUS5 = {"StatusNET":{"Hostname":"son_02-4451","IPAddress":"192.168.1.102","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"84:F3:EB:AF:91:63","Webserver":2,"WifiConfig":3}}
12:45:01 MQT: stat/son_02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.28","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AF9163","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:45:01 MQT: stat/son_02/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 11 10:45:01 2018","Local":"Tue Sep 11 12:45:01 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:07","Sunset":"20:03"}}
12:45:01 MQT: stat/son_02/STATUS10 = {"StatusSNS":{"Time":"2018-09-11T12:45:01"}}
12:45:01 MQT: stat/son_02/STATUS11 = {"StatusSTS":{"Time":"2018-09-11T12:45:01","Uptime":"6T11:38:18","Vcc":3.239,"POWER":"OFF","Wifi":{"AP":1,"SSId":"3992SK_2","RSSI":90,**"APMac":"58:EF:68:73:29:69"**}}}
```

**To Reproduce**
Check MAC address of access point with ping and than arp -a in a command prompt.
Check reported MAC address bySonoff with web console cmnd/sonoff/status 0

**Expected behavior**
MAC addresses to be equal

**Screenshots**
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-11:
Hi, Sorry, I could not reproduce your issue.

My Sonoff MAC and router MAC from status 0:
```
08:41:39 MQT: stat/living/STATUS = {"Status":{"Module":18,"FriendlyName":["Living"],"Topic":"living","ButtonTopic":"living","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:41:39 MQT: stat/living/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:12","StartupUTC":"2018-09-11T11:34:27","Sleep":0,"BootCount":36,"SaveCount":81,"SaveAddress":"3F9000"}}
08:41:39 MQT: stat/living/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-10T17:30:48","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
08:41:39 MQT: stat/living/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"192.168.1.104","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["04008009","55818000","00000000"]}}
08:41:39 MQT: stat/living/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":2580,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","0C0003B0","23021796","000000C0"]}}
08:41:39 MQT: stat/living/STATUS5 = {"StatusNET":{"Hostname":"living-5913","IPAddress":"192.168.1.126","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"XX:XX:XX:59:17:19","Webserver":2,"WifiConfig":5}}
08:41:39 MQT: stat/living/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_591719","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:41:39 MQT: stat/living/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 11 11:41:39 2018","Local":"Tue Sep 11 08:41:39 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"07:38","Sunset":"19:18"}}
08:41:39 MQT: stat/living/STATUS10 = {"StatusSNS":{"Time":"2018-09-11T08:41:39"}}
08:41:39 MQT: stat/living/STATUS11 = {"StatusSTS":{"Time":"2018-09-11T08:41:39","Uptime":"0T00:07:12","Vcc":2.993,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"XX:XX:XX:80:38:2C","Channel":1,"RSSI":72}}}
```

APMAC From my Router:
[image]

Sonoff MAC from Router:
[image]

ascillato, 2018-09-11:
Please, re test your findings.

closed 2018-09-11 by ascillato2

PerfectCrime, 2018-09-11:
@meingraham ARP queries the routing table.
I have a Linksys Velop mesh network with 3 access points. Very well possible they have more than 1 MAC address as they communicate with each other on different data channels. Thanks for your observation!