arendst/Tasmota #3763: How to set devices_present

galak, 2018-09-11:
I'm looking to support the following dimmer switch:

https://www.amazon.com/gp/product/B07FXYSVR1/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
https://fccid.io/2ANJ7-SD01

Here's what the GPIO setup looks like:

```
  { "MJ-SD01",         // MJ-SD01
     GPIO_KEY2_NP,     // GPIO00 Button [UP Button]
     GPIO_KEY3_NP,     // GPIO01 Serial RXD / Button [DOWN Button]
     GPIO_USER,        // GPIO02
     GPIO_LED5,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     GPIO_LED4,        // GPIO05 (LED4)
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_LED3,        // GPIO12 [LED3]
     GPIO_USER,        // GPIO13
     GPIO_LED2,        // GPIO14 (LED2)
     GPIO_KEY1_NP,     // GPIO15 (ON/OFF Button)
     GPIO_LED1,        // GPIO16 (LED1)
     0                 // ADC0 Analog input
  }
```

I'm trying to figure out how devices_present should get set, since we have 3 buttons/keys but just adding ends up getting devices_present = 0.  Not sure if I should add a new `light_type` or what?

Jason2866, 2018-09-11:
Dimmers are not supported from Tasmota.

ascillato, 2018-09-11:
Hi, devices_present are the amount of outputs like relays, pwm, and/or led strips.

As you set in your template a device with 3 buttons and 5 leds, why you need that variable to change?

ascillato, 2018-09-11:
If what you want is a full support for that device, you need to make a new driver, because the dimming is being done by another MCU in that device. So you will need to make all the communication protocol with that MCU.

ascillato, 2018-09-11:
Take a look on how sonoff dual is implemented. That will give you a hint on how you can work with devices_present variable and TX, RX pins

ascillato, 2018-09-11:
If you need more information, you can find us in Discord Chat

galak, 2018-09-11:
> If you need more information, you can find us in Discord Chat

what channel?

ascillato, 2018-09-11:
Tasmota Discord Chat https://discord.gg/Ks2Kzd4

galak, 2018-09-11:
> If what you want is a full support for that device, you need to make a new driver, because the dimming is being done by another MCU in that device. So you will need to make all the communication protocol with that MCU.

This dimmer doesn't seem to use a MCU like some of the other ones I've seen.

galak, 2018-09-11:
> Hi, devices_present are the amount of outputs like relays, pwm, and/or led strips.
> 
> As you set in your template a device with 3 buttons and 5 leds, why you need that variable to change?

In ButtonHandler() we have:
```
uint8_t maxdev = (devices_present > MAX_KEYS) ? MAX_KEYS : devices_present;
```

This ends up getting maxdev set to 0.  With that set to 0, its ends up means that the button handling doesn't do anything for the 3 buttons on the board.

ascillato2, 2018-09-11:
From the Chat. You can set relays in order to have outputs defined or use rules to trigger events when using the buttons/switches.

closed 2018-09-11 by ascillato2