arendst/Tasmota #3764: Sonoff Bridge mqtt controled by rfkey for DC motor

mrsadow, 2018-09-11:
Dear people,

I have some issues with setting up  mqtt switch for dc motor in home assistant.

I have a sonoff bridge that i use to send mhz433 signals to control the dc motor.

I have learned the sonoff bridge to send signals for go up(rfkey1) and go down(rfkey2)

The problem is: when i toggle the switch its sends like 1 signal the dc motor works for 1 second or less.

With the remote controle i can move it by holding the button.

how can modify the code when i toggle the switch to ON that the code stays sending the signal and when i toggle it off the signal will stop. (like holding button) 

I use the following code:

  - platform: mqtt
    name: "Magic box up"
    state_topic: "stat/sonoffbridge/rfkey"
    command_topic: "cmnd/sonoffbridge/rfkey"
    payload_on: "rfkey1"
    payload_off: "rfkey1OFF"
    state_on: "ON"
    state_off: "OFF"
    optimistic: false
    qos: 2
    retain: true
    
  - platform: mqtt
    name: "Magic box down"
    state_topic: "stat/sonoffbridge/rfkey"
    command_topic: "cmnd/sonoffbridge/rfkey"
    payload_on: "rfkey2"
    payload_off: "rfkey2OFF"
    state_on: "ON"
    state_off: "OFF"
    optimistic: false
    qos: 2
    retain: true

I hope some one can help me with this.

ascillato, 2018-09-11:
Hi,

> when i toggle the switch its sends like 1 signal the dc motor works for 1 second or less.

which switch? in Home Assistant UI ?

> how can modify the code when i toggle the switch to ON that the code stays sending the signal and when i toggle it off the signal will stop. (like holding button)

which code? your Home Assistant config or Tasmota ?

If Tasmota, you can use rules to do that.

closed 2018-09-11 by ascillato2

benzino77, 2018-09-12:
I do believe that @mrsadow wants to send RFCode for a period of time (to close or open roller shutters). The same code, just like the remote controller does  when the key is pressed down. So maybe it is not an issue but definitely can be considered as a feature request. 

I think that it can be compared to BlinkCount/BlinkTime - send RFCode X times with Y milliseconds delay between them. 

Or even better: keep sending RFCode (with a defined delay between them) until "RFCodeStop" command is sent.

Closing or opening roller shutters controlled by 433MHz remote will be easier in any Home Automation software. 

Hint:
As a limited workaround something like this can be used:
`backlog RFKey1;delay 3;RFKey1;delay 3;RFKey1;delay 3;RFKey1;delay 3;RFKey1;delay 3`
or
`backlog RFCode 12345;delay 4;RFCode 12345;delay 4;RFCode 12345;delay 4;RFCode 12345;delay 4`