arendst/Tasmota #3768: WIFI 5 vs. 6 problem

tomasroud, 2018-09-11:
I upgraded firmware to 6.1 and I must downgraded to 5.11.

With 5.11 after restart is device connected to wifi in 2 seconds,
with 6.21 it is 1 minute and more.

I tryed every method WIFI_WPSCONFIG, WIFI_MANAGER, WIFI_WAIT ... everything is same :(
Certainly with another CFG_HOLDER for each config.

If I start the log with 6.21, the better one:

00:00:00 CFG: Loaded from flash at F9, Count 3
00:00:00 APP: Boot Count 2
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A3B6F7, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 CFG: Saved to flash at F8, Count 4, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Checking connection...
00:00:10 WIF: Connected
00:00:10 HTP: Web server active on sonoff-5879 with IP address 10.0.125.227
00:00:12 MQT: Attempting connection...
00:00:12 MQT: Connected

Please where is the problem?

5.11 config:
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "10.0.0.1"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.0.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "10.0.0.10"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "JABLONSKY"      // [Ssid1] Wifi SSID
#define STA_PASS1              "xxxxx"  // [Password1] Wifi password
#define STA_SSID2              ""      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)

6.21 config
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "10.0.0.1"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.0.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "10.0.0.10"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "JABLONSKY"                // [Ssid1] Wifi SSID
#define STA_PASS1              "xxxxx"                // [Password1] Wifi password
#define STA_SSID2              ""                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WAIT         // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT, WIFI_SERIAL)
#define WIFI_CONFIG_NO_SSID    WIFI_WAIT    // Default tool if wifi fails to connect and no SSID is configured

ascillato2, 2018-09-11:
Hi,

I can not reproduce your issue.

Please, to avoid any misconfiguration, erase all flash with esptool.py (as wiki explains) and flash again by cable with last version v6.2.1.3

Frogmore42, 2018-09-12:
And provide the output from the status 0 command, either via serial or web console.

tomasroud, 2018-09-12:
Flash is erased and new firmware flashed, the result is here:
00:00:00 CFG: Loaded from flash at FB, Count 33
00:00:00 APP: Boot Count 32
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A3B6F7, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 CFG: Saved to flash at FA, Count 34, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Checking connection...
00:00:10 WIF: Connect failed with AP incorrect password
00:00:10 WIF: Patch issue 2186
00:00:10 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:11 WIF: Checking connection...
00:00:11 WIF: Attempting connection...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Checking connection...
00:00:15 WIF: Connected
00:00:15 HTP: Web server active on sonoff-5879 with IP address 10.0.125.227
00:00:15 MQT: Attempting connection...
00:00:16 MQT: Connected

tomasroud, 2018-09-12:
Can be the problem in my network? I have 8 APs with same SSID, maybe the device try to connect to first found AP with lower signal and then reconnect to AP with better signal. I try to set SSID to nearest AP different.

tomasroud, 2018-09-12:
That is the problem!!! Same SSID for more than 1 AP. Here is the log for unique SSID (one from so 10 restarts). As you can see, connection is established in few seconds.

00:00:00 CFG: Loaded from flash at F9, Count 19
00:00:00 APP: Boot Count 16
00:00:00 SRC: Restart
00:00:00 Project sonoff boiler_socket (Topic sonoff, Fallback DVES_A3B6F7, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 CFG: Saved to flash at F8, Count 20, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 JABLONSKY2 in mode 11N as sonoff-5879...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-5879 with IP address 10.0.125.227
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: switch/sonoff/tele/LWT = Online (retained)
00:00:07 MQT: switch/sonoff/cmnd/POWER = 
00:00:07 MQT: Subscribe to switch/sonoff/cmnd/#
00:00:07 MQT: Subscribe to switch/sonoffs/cmnd/#
00:00:07 MQT: Subscribe to cmnd/DVES_A3B6F7/#

tomasroud, 2018-09-12:
And here is the log when I changed SSID back. So this log is catched by serial connection and is when there are few APs with same SSID.
The device self restarted few times to establish connection and it takes so 2 minutes :(

00:00:10 WIF: Connect failed with AP incorrect password
00:00:10 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:20 WIF: Connect failed with AP incorrect password
00:00:22 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4010429c
~ld

00:00:00 Project sonoff boiler_socket (Topic sonoff, Fallback DVES_A3B6F7, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:10 WIF: Connect failed with AP incorrect password
00:00:10 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:20 WIF: Connect failed with AP incorrect password
00:00:22 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4010429c
~ld

00:00:00 Project sonoff boiler_socket (Topic sonoff, Fallback DVES_A3B6F7, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP1 JABLONSKY in mode 11N as sonoff-5879...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-5879 with IP address 10.0.125.227
00:00:05 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: switch/sonoff/tele/LWT = Online (retained)
00:00:06 MQT: switch/sonoff/cmnd/POWER = 
00:00:06 MQT: switch/sonoff/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1","FallbackTopic":"DVES_A3B6F7","GroupTopic":"sonoffs"}
00:00:06 MQT: switch/sonoff/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5879","IPAddress":"10.0.125.227"}
00:00:06 MQT: switch/sonoff/tele/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: switch/sonoff/stat/RESULT = {"POWER":"OFF"}
00:00:06 MQT: switch/sonoff/stat/POWER = OFF (retained)

ascillato, 2018-09-12:
Great that you found a solution for your issue.

Thanks for share