arendst/Tasmota #3769: insert true/false for tasmota to accept of power

madjack84, 2018-09-11:
**Is your feature request related to a problem? Please describe.**
iobroker does only send true/false or 0/100 (for lits) but tasmota firmware only accepts 1/0 or ON/OFF.
extend MQTT message set.
Already saw that in many posts, but with no good answer.

**Describe the solution you'd like**
Please add true/false as 
'cmnd/myDevice/power' -m 'true'
for setting POWER to 1

**Describe alternatives you've considered**
rules don't work properly
"Rule1":"1","Once":"0","StopOnError":"0","Free":435,"Rules":"on event#power=false do power1 0 endon on event#power=true do power1 1 endon"
as iobroker can't generate a message with payload power=false

ascillato2, 2018-09-11:
Hi,

I tested the following rules :

```
rule 1
rule on event#mypower=false do power1 0 endon on event#mypower=true do power1 1 endon
```

using MQTT as:

`cmnd/sonofftopic/event mypower`
with payload `true`
and payload `false`

and works fine.

madjack84, 2018-09-13:
Thanks, the workaround works well :)
However this command is to be used for every iobroker user for every sonoff device.
I think adding true besides on and false besides off in the code is a minor minor thing and adds lots of convenience for many users :)

ascillato, 2018-09-13:
True. You are right

By now, this has been added to the wiki as there was nothing about IOBroker there.

Thanks.

madjack84, 2018-09-14:
I tried to finde the code line for adding true/false myself.
Unfortunately searching for on or off results in plenty (>300) hits ;)

in sonoff.ino method following MqttDataHandler I can't find the on/off interpretation in the payload conversion section... Do you have a hint where to find it?
I can't debug it unfortunately myself :(

    if (!strcmp(dataBuf,"?")) data_len = 0;
    int16_t payload = -99;               // No payload
    uint16_t payload16 = 0;
    long payload32 = strtol(dataBuf, &p, 10);
    if (p != dataBuf) {
      payload = (int16_t) payload32;     // -32766 - 32767
      payload16 = (uint16_t) payload32;  // 0 - 65535
    } else {
      payload32 = 0;
    }
    backlog_delay = millis() + (100 * MIN_BACKLOG_DELAY);

    int temp_payload = GetStateNumber(dataBuf);
    if (temp_payload > -1) { payload = temp_payload; }

arendst, 2018-09-15:
My understanding is that iobroker uses node.js

There must be hundreds of solutions to change true to 1/On and false to 0/Off by node.js

madjack84, 2018-09-15:
oh sorry for the misleading comment :/
I meant to integrate into sonoffs arduino SW stack not into ioBroker node.js.
Think sonoff acceptance of true/false is the better solution regarding architecture

nicomania, 2019-11-20:
> Hi,
> 
> I tested the following rules :
> 
> ```
> rule 1
> rule on event#false do power1 0 endon on event#true do power1 1 endon
> ```
> 
> using MQTT as:
> 
> `cmnd/sonofftopic/event`
> with payload `true`
> and payload `false`
> 
> and works fine.
> 
> Console Output:
> 
> ```
> 18:42:05 MQT: cmnd/living/event = true
> 18:42:05 SRC: MQTT
> 18:42:05 RSL: Topic Recibido cmnd/living/event, Tamaño de Datos 4, Datos true
> 18:42:05 RSL: Grupo 0, Índice 1, Comando EVENT, Datos true
> 18:42:05 MQT: stat/living/RESULT = {"Event":"Done"}
> 18:42:05 RUL: EVENT#TRUE performs "power1 1"
> 18:42:05 SRC: Rule
> 18:42:05 RSL: Topic Recibido /power1, Tamaño de Datos 1, Datos 1
> 18:42:05 RSL: Grupo 0, Índice 1, Comando POWER, Datos 1
> 18:42:05 MQT: stat/living/RESULT = {"POWER1":"ON"}
> 18:42:05 MQT: stat/living/POWER1 = ON
> ```

Hello ascillato,

I have the same problem than madjack84. But for some reason the rule you mentioned does not work for me. I paste this in the console:
`rule1 `
`on event#false do power1 0 endon `
`on event#true do power1 1 endon`

and it accepts it:
`21:17:51 CMD: rule1  on event#false do power1 0 endon  on event#true do power1 1 endon`
`21:17:51 MQT: stat/OG/Ecklampe/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":446,"Rules":"on event#false do power1 0 endon  on event#true do power1 1 endon"}`

But on every "true" (or "false") trigger, it just switches off
`21:19:08 MQT: stat/OG/Ecklampe/RESULT = {"POWER":"OFF"}`
`21:19:08 MQT: stat/OG/Ecklampe/POWER = OFF`

Can you tell me, what I am doing wrong? 
Tasmota 6.7.1

Thank you and best regards

ascillato, 2019-11-20:
Please, use command weblog 4 to have more debug information in the console, and then try the event true and event false. Please, paste the output of the console. Thanks

nicomania, 2019-11-21:
Thank you for the fast response. Now I have the following:
`07:40:45 CMD: rule1  on event#false do power1 0 endon  on event#true do power1 1 endon`
`07:40:45 SRC: WebConsole from 192.168.178.53`
`07:40:45 CMD: Gruppe 0, Index 1, Befehl "RULE", Daten "on event#false do power1 0 endon  on event#true do power1 1 endon"`
`07:40:45 MQT: stat/OG/Ecklampe/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":446,"Rules":"on event#false do power1 0 endon  on event#true do power1 1 endon"}`

and at action:
`07:41:44 MQT: empfangenes topic "cmnd/OG/Ecklampe/POWER", Datengröße 4, Daten "true"`
`07:41:44 SRC: MQTT`
`07:41:44 CMD: Gruppe 0, Index 1, Befehl "POWER", Daten "true"`
`07:41:44 MQT: stat/OG/Ecklampe/RESULT = {"POWER":"OFF"}`
`07:41:44 MQT: stat/OG/Ecklampe/POWER = OFF`

ascillato, 2019-11-21:
Thanks for posting.

The problem is that you are sending POWER TRUE to Tasmota. That is not supported.

These rules just add another way of turning relays that is sending the commands:
EVENT TRUE
EVENT FALSE

nicomania, 2019-11-21:
As far as I can see, iobroker is sending "true" to the ID "power" and not "power true". But I might be wrong.
`07:41:44 CMD: Gruppe 0, Index 1, Befehl "POWER", Daten "true"`

and also here:
`07:41:44 MQT: empfangenes topic "cmnd/OG/Ecklampe/POWER", Datengröße 4, Daten "true"`
The datasize is 4 bytes/chars => "true"

Whatever, this should mean, tasmota is incompatible with iobroker? I do not see anything that could be configured to make that work. I can only export the mqtt path of my tasmota device to Alexa cloud adapter of iobroker. Thats it.

I am very new to mqtt, but I cannot believe, that two of those promising and well known products (Tasmota and iobroker) should be incompatible over the worldwide standard mqtt.

Thank you for your time.

ascillato, 2019-11-21:
You need to change in IOBROKER the ID **power** to ID **EVENT**. Just that.

nicomania, 2019-11-21:
the ID is not editable.

ascillato, 2019-11-21:
**madjack84** did that and these rules work for his setup

nicomania, 2019-11-21:
Well, maybe @madjack84 can tell us, where the magic rename button is. The edit pen on the right is not.
[image]

ascillato2, 2019-11-21:
Ok, seems that madjack84 did some modification to iobroker to make this works.

Reopening this request to see if there is someone interested on working on this and provide a PR for supporting IOBROKER. So far, IOBROKER is not supported by Tasmota and there is no information about it on the wiki.

madjack84, 2019-11-22:
Hey all,

you have to send an Event, but not Power itself 

The rule I use:

rule 1
rule on event#false do power1 0 endon on event#true do power1 1 endon

The mqtt message:

cmnd/sonofftopic/event/power1 true

Thats the workaround and it works well.

arendst, 2019-11-23:
Strange. ``cmnd/device/power true`` should always have been working...

I'll investigate.

arendst, 2019-11-23:
Just tested and it works as expected.
```
00:00:26 CMD: power
00:00:26 SRC: Serial
00:00:26 CMD: Group 0, Index 1, Command "POWER", Data ""
00:00:26 MQT: stat/wemos4/POWER = {"POWER":"OFF"}
00:00:26 MQT: stat/wemos4/POWER = OFF
00:00:30 CMD: power true
00:00:30 SRC: Serial
00:00:30 CMD: Group 0, Index 1, Command "POWER", Data "true"
00:00:30 MQT: stat/wemos4/POWER = {"POWER":"ON"}
00:00:30 MQT: stat/wemos4/POWER = ON
00:00:30 CFG: Saved to flash at F7, Count 86, Bytes 4096
00:00:39 CMD: power False
00:00:39 SRC: Serial
00:00:39 CMD: Group 0, Index 1, Command "POWER", Data "False"
00:00:39 MQT: stat/wemos4/POWER = {"POWER":"OFF"}
00:00:39 MQT: stat/wemos4/POWER = OFF
00:00:39 CFG: Saved to flash at F6, Count 87, Bytes 4096
```

arendst, 2019-11-23:
Unless you are using another language than English where the words "True" or "False" are translated in something completely different :-)

I'll add "True" and "False" to the parameter decode list.

Nice issue.

ascillato2, 2019-11-24:
Closing this issue as it has been added. Thanks

closed 2019-11-24 by ascillato2

nicomania, 2019-11-24:
I will wait for the next release and give it a try. Thank you all for the quick support.

ascillato, 2019-11-24:
It is already compiled at http://thehackbox.org/tasmota

nicomania, 2019-11-25:
Works like a charm :-) Thank you!