arendst/Tasmota #3770: Wall Switch Button Led Colour

mesab0ogie, 2018-09-11:
Im recently flashed Tasmota to a Sonoff T1 variant wall switch and minus all the initial first timer hiccups, i got everything working fine and dandy now. However one thing i noticed is that the button LEDs dont follow the stock colours when buttons are pushed before Tasmota was flashed.

[URL=http://s697.photobucket.com/user/mesab0ogie/media/20180912_043501.jpg.html][IMG]http://i697.photobucket.com/albums/vv331/mesab0ogie/20180912_043501.jpg[/IMG][/URL]

Out of the box the switch has its buttons in blue when off.

[URL=http://s697.photobucket.com/user/mesab0ogie/media/20180912_043437.jpg.html][IMG]http://i697.photobucket.com/albums/vv331/mesab0ogie/20180912_043437.jpg[/IMG][/URL]

When a button is pressed it will change to red to show its turned on, while the other button remains blue.

[URL=http://s697.photobucket.com/user/mesab0ogie/media/20180912_043508.jpg.html][IMG]http://i697.photobucket.com/albums/vv331/mesab0ogie/20180912_043508.jpg[/IMG][/URL]

On Tasmota however, when a button is turned on, it turns red but the other button will turn off its led.

Its a minor annoyance but i was wondering if there is a way for me to change this.

[image]

ascillato2, 2018-09-11:
Hi,

Have your tried commands ledstate and ledpower ? Please, see wiki for more information on commands.

mesab0ogie, 2018-09-12:
Hi,

Tried the ledpower and ledstate commands and they dont get the the blue lights on. Is this because the original Sonoff T1s have no lights on the buttons in the off state? And these switches have 2 coloured LED lights(red and blue) as opposed to Sonoff T1 which only has blue leds. I was wondering if there is something i can modify in the code to get it to control these leds.

ascillato, 2018-09-13:
Hi,

have you managed to solve your issue?

mesab0ogie, 2018-09-13:
Just got home a couple of hours ago and messed with the switch again. Was still getting nowhere so i decided to try and flash a brand new switch with tasmota and try again and voila - the ledstate command works! Still not sure what was wrong with the other switch(still cant get the leds to turn on). Might be faulty. Thank you for your help ascillato! Really appreciate it

ascillato, 2018-09-13:
Ok, great.

You could turn the blue on using ledstate and ledpower? Or the blue is on all the time? 

If your issue is solved, please close it. Thanks.

mesab0ogie, 2018-09-13:
Yes changing ledstate and ledpower to On makes the blue stay on all the time. Thank you.