arendst/Tasmota #3773: AP1 -> AP2

synekvl, 2018-09-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : S26 with the last Tasmota
- [ ] Provide the output of command ``status 0`` :
```
I'm experiencing problem with one of my toys, the S26. All others, and this one before too, have assigned two AP - Home as AP1 and Office as AP2. Running them in the office all shows at the beginning "Fails connect to AP1" and try to connect alternative AP2. This S26 after failed attempt with AP1 goes to WiFi Manager.

As I mentioned there was no problem before, it happened during weekend (maybe some interruption on WiFi signal, but all others recovered except this one).

I tried flashed it with new CFG_HOLDER via OTA but the problem is still there. Generally having such problem I flash devices via cable, but in case of S26 there is necessary to solder wires to pads.... :-(

So, is there some command to persuade S26 to connect AP1 and if it fails to go to connect AP2 ? Commands "AP x" work.

andrethomas, 2018-09-12:
You did not give us a status 0 output, so not sure what your current WifiConfig value is.

From https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi

Looks like you need to configure:
WifiConfig 4

synekvl, 2018-09-12:
Thank you for your suggestion, nevermind it has to be something different as cmnd/WiFiConfig on S26 gives value "2" (WiFiManager") , the same as on another devices without this "problem" ....

ascillato, 2018-09-12:
Ok. Type in Tasmota console, wificonfig 4

closed 2018-09-12 by ascillato2