arendst/Tasmota #3775: Sonoff Tasmota become unresponsive

gmallants, 2018-09-12:
Hello,

I control all my heater with Sonoff Basic and MQTT with Domoticz. I have no problem at all with two modules (surprising, the most distant from my router), one other is a bit messy (office room) and two others are totally messy (bedroom and kitchen).

When it's OK, console show (living room and entrance) : 
```
09:49:20 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:49:20 MQT: stat/sonoff/POWER1 = ON
09:50:50 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:50:50 MQT: stat/sonoff/POWER1 = ON
09:51:05 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:05 MQT: stat/sonoff/POWER1 = ON
09:51:30 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:30 MQT: stat/sonoff/POWER1 = ON
09:51:32 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:32 MQT: stat/sonoff/POWER1 = ON
09:51:38 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:38 MQT: stat/sonoff/POWER1 = ON
```

When it's messy (office room), console show : 
```
09:39:45 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
09:39:45 RSL: stat/sonoff/POWER1 = ON
09:39:45 MQT: Attempting connection...
09:39:52 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
09:40:02 MQT: Attempting connection...
09:40:09 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
09:40:19 MQT: Attempting connection...
09:40:23 MQT: Connected
09:40:23 MQT: tele/sonoff/LWT = Online (retained)
09:40:23 MQT: cmnd/sonoff/POWER = 
09:41:16 MQT: Attempting connection...
09:41:23 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
09:41:33 MQT: Attempting connection...
09:41:40 MQT: Connected
09:41:40 MQT: tele/sonoff/LWT = Online (retained)
09:41:40 MQT: cmnd/sonoff/POWER = 
09:42:54 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:42:54 MQT: stat/sonoff/POWER1 = ON
09:44:17 MQT: Attempting connection...
09:44:18 MQT: Connected
09:44:23 RSL: tele/sonoff/LWT = Online
09:44:28 RSL: cmnd/sonoff/POWER = 
09:44:50 MQT: Attempting connection...
09:44:50 MQT: Connected
09:44:50 MQT: tele/sonoff/LWT = Online (retained)
09:44:50 MQT: cmnd/sonoff/POWER = 
09:46:12 MQT: Attempting connection...
09:46:12 MQT: Connected
09:46:12 MQT: tele/sonoff/LWT = Online (retained)
09:46:13 MQT: cmnd/sonoff/POWER = 
09:46:38 MQT: Attempting connection...
09:46:39 MQT: Connected
09:46:39 MQT: tele/sonoff/LWT = Online (retained)
09:46:39 MQT: cmnd/sonoff/POWER = 
09:47:54 APP: Serial logging disabled
09:48:39 MQT: Attempting connection...
09:48:59 MQT: Connect failed to 192.168.0.30:1883, rc -4. Retry in 10 sec
09:49:20 MQT: Attempting connection...
09:49:21 MQT: Connected
09:49:21 MQT: tele/sonoff/LWT = Online (retained)
09:49:21 MQT: cmnd/sonoff/POWER = 
09:49:56 MQT: Attempting connection...
09:50:03 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
09:50:28 MQT: Attempting connection...
09:50:34 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
09:50:45 MQT: Attempting connection...
09:51:05 MQT: Connected
09:51:06 MQT: tele/sonoff/LWT = Online (retained)
09:51:06 MQT: cmnd/sonoff/POWER = 
09:51:32 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:32 RSL: stat/sonoff/POWER1 = ON
09:51:32 MQT: Attempting connection...
09:51:33 MQT: Connected
09:51:33 MQT: tele/sonoff/LWT = Online (retained)
09:51:33 MQT: cmnd/sonoff/POWER = 
09:51:39 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
09:51:39 MQT: stat/sonoff/POWER1 = ON
```

For the two others modules, I have no access at all at this time ! Do you have an idea where my problem is located ?

Thanks,

---

- Hardware used : 
Modem/Router : Freebox Crystal (in kitchen)
Computer : Raspberry PI B3 with Rasbian Stretch and Mosquitto MQTT Broker 1.5.0
Module : 5 * Sonoff Basic (with temperature sensor), last 6.2.1 firmware

- Output of command ``status 0`` (office room, second console output) : 
```
09:55:12 RSL: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff-Bureau"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:55:12 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:23:16","StartupUTC":"2018-09-12T08:31:56","Sleep":0,"BootCount":50,"SaveCount":303,"SaveAddress":"F5000"}}
09:55:12 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:55:12 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["EmiGaet","indebuurt2"],"TelePeriod":0,"SetOption":["54000009","55818000","00000001"]}}
09:55:12 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":6,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
09:55:12 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Bureau","IPAddress":"192.168.0.41","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"212.27.40.241","Mac":"DC:4F:22:0D:D0:A3","Webserver":2,"WifiConfig":5}}
09:55:12 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DD0A3","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:55:12 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 08:55:12 2018","Local":"Wed Sep 12 09:55:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:22","Sunset":"19:10"}}
09:55:12 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T09:55:12","AM2301":{"Temperature":26.0,"Humidity":43.1},"TempUnit":"C"}}
09:55:12 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T09:55:12","Uptime":"0T00:23:16","Vcc":3.152,"POWER1":"ON","Wifi":{"AP":1,"SSId":"EmiGaet","RSSI":80,"APMac":"2A:82:51:07:DD:08"}}}
```

gmallants, 2018-09-12:
> Hello,
> 
> I control all my heater with Sonoff Basic and MQTT with Domoticz. I have no problem at all with two modules (surprising, the most distant from my router), one other is a bit messy (office room) and two others are totally messy (bedroom and kitchen).
> 
> When it's OK, console show (living room and entrance) :
> 
> ```
> 09:49:20 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:49:20 MQT: stat/sonoff/POWER1 = ON
> 09:50:50 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:50:50 MQT: stat/sonoff/POWER1 = ON
> 09:51:05 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:51:05 MQT: stat/sonoff/POWER1 = ON
> 09:51:30 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:51:30 MQT: stat/sonoff/POWER1 = ON
> 09:51:32 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:51:32 MQT: stat/sonoff/POWER1 = ON
> 09:51:38 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:51:38 MQT: stat/sonoff/POWER1 = ON
> ```
> When it's messy (office room), console show :
> 
> ```
> 09:39:45 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:39:45 RSL: stat/sonoff/POWER1 = ON
> 09:39:45 MQT: Attempting connection...
> 09:39:52 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
> 09:40:02 MQT: Attempting connection...
> 09:40:09 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
> 09:40:19 MQT: Attempting connection...
> 09:40:23 MQT: Connected
> 09:40:23 MQT: tele/sonoff/LWT = Online (retained)
> 09:40:23 MQT: cmnd/sonoff/POWER = 
> 09:41:16 MQT: Attempting connection...
> 09:41:23 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
> 09:41:33 MQT: Attempting connection...
> 09:41:40 MQT: Connected
> 09:41:40 MQT: tele/sonoff/LWT = Online (retained)
> 09:41:40 MQT: cmnd/sonoff/POWER = 
> 09:42:54 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:42:54 MQT: stat/sonoff/POWER1 = ON
> 09:44:17 MQT: Attempting connection...
> 09:44:18 MQT: Connected
> 09:44:23 RSL: tele/sonoff/LWT = Online
> 09:44:28 RSL: cmnd/sonoff/POWER = 
> 09:44:50 MQT: Attempting connection...
> 09:44:50 MQT: Connected
> 09:44:50 MQT: tele/sonoff/LWT = Online (retained)
> 09:44:50 MQT: cmnd/sonoff/POWER = 
> 09:46:12 MQT: Attempting connection...
> 09:46:12 MQT: Connected
> 09:46:12 MQT: tele/sonoff/LWT = Online (retained)
> 09:46:13 MQT: cmnd/sonoff/POWER = 
> 09:46:38 MQT: Attempting connection...
> 09:46:39 MQT: Connected
> 09:46:39 MQT: tele/sonoff/LWT = Online (retained)
> 09:46:39 MQT: cmnd/sonoff/POWER = 
> 09:47:54 APP: Serial logging disabled
> 09:48:39 MQT: Attempting connection...
> 09:48:59 MQT: Connect failed to 192.168.0.30:1883, rc -4. Retry in 10 sec
> 09:49:20 MQT: Attempting connection...
> 09:49:21 MQT: Connected
> 09:49:21 MQT: tele/sonoff/LWT = Online (retained)
> 09:49:21 MQT: cmnd/sonoff/POWER = 
> 09:49:56 MQT: Attempting connection...
> 09:50:03 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
> 09:50:28 MQT: Attempting connection...
> 09:50:34 MQT: Connect failed to 192.168.0.30:1883, rc -2. Retry in 10 sec
> 09:50:45 MQT: Attempting connection...
> 09:51:05 MQT: Connected
[... 8 lines cut ...]
> 09:51:39 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 09:51:39 MQT: stat/sonoff/POWER1 = ON
> ```
> For the two others modules, I have no access at all at this time ! Do you have an idea where my problem is located ?
> 
> Thanks,
> 
> * Hardware used :
>   Modem/Router : Freebox Crystal (in kitchen)
>   Computer : Raspberry PI B3 with Rasbian Stretch and Mosquitto MQTT Broker 1.5.0 (ETH connected to the router)
>   Module : 5 * Sonoff Basic (with temperature sensor), last 6.2.1 firmware
> * Output of command `status 0` (office room, second console output) :
> 
> ```
> 09:55:12 RSL: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff-Bureau"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
> 09:55:12 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:23:16","StartupUTC":"2018-09-12T08:31:56","Sleep":0,"BootCount":50,"SaveCount":303,"SaveAddress":"F5000"}}
> 09:55:12 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
> 09:55:12 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["EmiGaet","indebuurt2"],"TelePeriod":0,"SetOption":["54000009","55818000","00000001"]}}
> 09:55:12 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":6,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
> 09:55:12 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Bureau","IPAddress":"192.168.0.41","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"212.27.40.241","Mac":"DC:4F:22:0D:D0:A3","Webserver":2,"WifiConfig":5}}
> 09:55:12 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DD0A3","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
> 09:55:12 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 08:55:12 2018","Local":"Wed Sep 12 09:55:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:22","Sunset":"19:10"}}
> 09:55:12 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T09:55:12","AM2301":{"Temperature":26.0,"Humidity":43.1},"TempUnit":"C"}}
> 09:55:12 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T09:55:12","Uptime":"0T00:23:16","Vcc":3.152,"POWER1":"ON","Wifi":{"AP":1,"SSId":"EmiGaet","RSSI":80,"APMac":"2A:82:51:07:DD:08"}}}
> ```

gmallants, 2018-09-12:
Also, in Mosquitto logs I show this : 
1536726843: New connection from 192.168.0.43 on port 1883.
1536726843: New client connected from 192.168.0.43 as DVES_0DC646 (c1, k15, u'DVES_USER').
1536726848: Client DVES_0DD0A3 has exceeded timeout, disconnecting.
1536726848: Socket error on client DVES_0DD0A3, disconnecting.
1536726848: Client DVES_6735EF has exceeded timeout, disconnecting.
1536726848: Socket error on client DVES_6735EF, disconnecting.
1536726849: New connection from 192.168.0.41 on port 1883.
1536726849: New client connected from 192.168.0.41 as DVES_0DD0A3 (c1, k15, u'DVES_USER').
1536726867: New connection from 192.168.0.42 on port 1883.
1536726867: New client connected from 192.168.0.42 as DVES_6735EF (c1, k15, u'DVES_USER').
1536726874: Client DVES_0DD0A3 has exceeded timeout, disconnecting.
1536726874: Socket error on client DVES_0DD0A3, disconnecting.
1536726875: New connection from 192.168.0.41 on port 1883.
1536726875: New client connected from 192.168.0.41 as DVES_0DD0A3 (c1, k15, u'DVES_USER').
1536726897: Client DVES_6735EF has exceeded timeout, disconnecting.
1536726897: Socket error on client DVES_6735EF, disconnecting.
1536726899: New connection from 192.168.0.42 on port 1883.
1536726904: New client connected from 192.168.0.42 as DVES_6735EF (c1, k15, u'DVES_USER').
1536726912: Client DVES_0DD0A3 has exceeded timeout, disconnecting.
1536726912: Socket error on client DVES_0DD0A3, disconnecting.
1536726930: New connection from 192.168.0.41 on port 1883.
1536726930: New client connected from 192.168.0.41 as DVES_0DD0A3 (c1, k15, u'DVES_USER').
1536726952: Client DVES_0DD0A3 has exceeded timeout, disconnecting.
1536726952: Socket error on client DVES_0DD0A3, disconnecting.
1536726972: New connection from 192.168.0.41 on port 1883.
1536726975: New client connected from 192.168.0.41 as DVES_0DD0A3 (c1, k15, u'DVES_USER').
1536726997: Client DVES_0DD0A3 has exceeded timeout, disconnecting.
1536726997: Socket error on client DVES_0DD0A3, disconnecting.

andrethomas, 2018-09-12:
I see the build time on your status 0 output indicates 2018-09-09T16:50:26

There was a bug in the releases around that time - please download the binary file again and
flash it with the new binary file - This may solve what you are experiencing.

Please let us know the result.

Thanks

ascillato, 2018-09-12:
Your issue is pubsubclient indeed. Please change to TasmotaMQTT that works great. No hungs. You can select that mqtt library from the user_config.h file and recompile

gmallants, 2018-09-12:
@ascillato Thanks, there is a guide somewhere to explain you how to do that ? Why not set TasmotaMQTT by default ?

@andrethomas I'll try that also ;-)

ascillato, 2018-09-12:
Guides for flashing are in the wiki.

And for setting TasmotaMQTT please, uncomment TasmotaMQTT and comment pubsubclient in the file user_config.h 

Thanks

closed 2018-09-12 by ascillato2

gmallants, 2018-09-12:
Hello,

I successfully make my build and upload it on devices, except the three one that are not accessible ATM. I'll keep you inform when it's done.

KR,

gmallants, 2018-09-12:
Hello,

I update the module in the kitchen and disable MQTT for a test and it's not better. It's really hard to access the WEB UI of the module and everything is so slow.

I've a 100% signal but three problematic modules are on the same wall of the building ! Maybe I've to try to move them a bit and/or use an other WiFi router.

Any other clue otherwise ?

KR,

andrethomas, 2018-09-12:
Other than complete erase and reflash then I cannot think of any other reason if you are using the officially released binary downloaded firmware.

Information on erase and reflash is on the wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

Good luck

ascillato, 2018-09-12:
You can use an app for you smartphone like wifianalyser to check if there are too much wifi networks. If that the case, you will need to reconfigure your router

andrethomas, 2018-09-12:
@ascillato 
[image]

gmallants, 2018-09-12:
@andrethomas Unfortunately I don't have a COM programmer.
@ascillato I'll check but I'm in appartement and there is Wifi on every chanel, I already put the best one in place.

andrethomas, 2018-09-12:
`Edit : the building is from the war and made of metal, this can cause troubles I suppose...`

Try swapping the units around to see if the same one is causing the problem or if it is a location problem.

gmallants, 2018-09-12:
I also notice this in the  console (for the same module) : 

14:11:20 MQT: domoticz/in = {"idx":6,"nvalue":1,"svalue":"","Battery":48,**"RSSI":10**}
...
14:13:14 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T14:13:14","Uptime":"0T00:04:16","Vcc":3.082,"POWER1":"ON","Wifi":{"AP":1,"SSId":"EmiGaet",**"RSSI":96**,"APMac":"2A:82:51:07:DD:08"}}}

Don't understand the first line, Sonoff don't have any battery...

gmallants, 2018-09-13:
Hello,

I find the issue, I move three module 10 centimeters off the wall and now everything work well. I think even with good signal, something make perturbation inside the wall.

Hope this thread can help others people in the futur. Thanks for your help.

KR,