arendst/Tasmota #3776: Alexa hue emulation works but 'isnt responding please check network connection and power supply'

bjhutchins, 2018-09-12:
alexa is finding devices but i just seem to control them!?
Seen people have previously had similar trouble but Whats going on?!

[image]

ascillato, 2018-09-12:
Hi,

There is a lot of information missing in order to help you properly. Please, complete the troubleshooting template.

Also, inform us if you are using a precompiled version or compiled by yourself. Which version, which esplib version, which hardware, which compiler, etc etc.

ascillato2, 2018-09-12:
Hi,

Sorry, stills all the information requested is missing.

Please, when you have time, complete the information requested in order to properly help you.

Thanks

closed 2018-09-12 by ascillato2

bjhutchins, 2018-09-12:
i dont understand what else you need to know? sorry