arendst/Tasmota #3777: Support PEAP authentication

qingz2004, 2018-09-12:
**Have you look for this feature in other issues and in the wiki?**
Yes.
**Is your feature request related to a problem? Please describe.**
I'd like to connect to my company WiFi network with PEAP authentication .

**Describe the solution you'd like**
Add support to connect to PEAP WiFi network.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2018-09-12 by ascillato2

Stevolus, 2021-05-28:
Sorry I know this is necromancing, just want to show I am also interested in this for security reasons.

ascillato, 2021-05-28:
Hi, Tasmota uses the Arduino wifi libraries and those uses the espressif sdk.

At this moment the sdk provided by the esp8266 manufacturer does not fully support PEAP, so there is nothing Tasmota can do about that.

See related issue in Arduino Repository
esp8266/Arduino#3442

Please, ask in the linked issue of the Arduino Core. Thanks.