arendst/Tasmota #3779: Restore Configuration Command Does not Work between 6.11C & 6.22

MTGS69, 2018-09-12:
**Describe the bug**
Admittedly, I do not use the recommended USER_CONFIG_OVERRIDE method when upgrading to a new version and thus this issue maybe a result of that and my fault.  Furthermore, I can work around this so this is more of an FYI and an opportunity to gain a better understanding of the issue for future use.  When a new version comes out I update the settings in the USER_CONFIG file as required (WIFI and MQTT usernames & Passwords, timezone, etc) which takes a couple minutes.  I also use the config file to set a static IP to something not used by any other device in my network (e.g. 192.168.1.169).  Next I compile a minimal and standard bin file.  I then open the web interface, save the configuration, upgrade the firmware to the minimal build, reboot into web interface at the reserved 192.168.1.169 address, upgrade the firmware to the standard build, reboot to 192.168.1.169, and finally load the config file which sets everything back to the original settings including the original set static IP.  I've been using this same process for 20 devices for many (~10) upgrades without problem.  However, in the latest build (6.2 both main and development branches) the process fails when trying to upgrade from 6.12C.  Everything works fine up to the step where I upload the new standard build bin file.  However, when I try to restore the configuration it says it succeeded but nothing in the configuration actually changes. 

I tried erasing a test chip, installing the old 6.21C build and restoring the same saved config files that fail to update 6.22 and they work fine.  I've also tried multiple devices and all have the same problem.

Again, not a major issue for me.  I can create a node-red flow to save and restore most of the settings.  I'm just curious why a procedure I have used on many devices across many upgrades failed this time.

Thanks in advance!

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_  Yes
- [ ] _Hardware used_ :  SONOFF Basics and WEMOS D1 Minis 
- [ ] _Provide the output of command_``status 0`` :  Provides no value to issue
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
Upload a 6.21C build, change the configuration, save the configuration, upload a 6.2 build and then try and restore the configuration.

**Expected behavior**
The saved configuration is restored but it reverts back to the user_config values of the new build.

I am happy to attach the bin files and/or my modified user config files.  However, they contain the username and password information to my network so I do not want to do so unless that would be useful.

Frogmore42, 2018-09-12:
Seems like a complicated process.  What exactly are you trying to achieve with it?

There are reasons to build the FW yourself:
Pre-configure Wi-Fi and MQTT are examples.  But that requires using the user configuration override so the settings actually get used.  If you are not doing that, you should probably use one of the pre-compiled binaries.

If you use a consistent cfg_holder, your settings will be maintained during and after an update.  The documentation on this is inconsistent, but my recommendation is to always use the one that is in the Git repo and NOT change it.  If you change it, you need to keep very good track of what you changed it to, when, and why.  The reason some of the documentation says to change it is because it seems like it only gets used on the first flash of a device.  That is sort of true, but you can always reset the device and tell it to use the default (in the code) settings.

I have never tried saving and restoring settings so I don't know any issues with the process.

MTGS69, 2018-09-13:
@Frogmore42  you said something that after many flashes "the hard way" finally clicked in my head.  Specifically, if the CFG_HOLDER variable remains constant between builds your old configured settings will not be reset.  I read the troubleshooting wiki on this subject a long time ago but somehow thought that meant the changes you make to Section 1 would be ignored BUT it would not keep your previous settings.  Seems obvious to me now but I think I know why I fell into this trap.  To test the functionality I erased a test Wemos D1, loaded an old version of the firmware from a few months back, compiled a version of the latest dev firmware but ONLY changed the CFG_HOLDER to match what I used in the old version.  That almost worked but with one little problem.  My login information is stored by Chrome, has never changed since the first flash, but I could not login.  I then realized that the admin username is not part of Section 1 and therefore is not part of the persistent unless CFG_HOLDER value is changed information.  Rather, it is defined in Section 2 as "admin".  I think this threw me off track back when I was a just getting started with this and just got in the habit of increment the CFG_HOLDER for each new version and making the same adjustments I always make.  So the confusion was all on me.  With that said.....I have to ask....why is the web username not included in Section 1?  I changed it from the default value of "admin" as an added security measure.  In other words, they have to know your password AND your username.......and way too many products use "admin" as a default and do not let you change (e.g. my Samsung IP Cams)?  Anyhow thanks for the help!

MTGS69, 2018-09-13:
One additional question.....I cannot get the previous version (the one that all 20 of my devices are currently running to update) even though I set the CFG_HOLDER to the value I used to create that build.  I am thinking that may have something to do with commenting out features I do not use.  It has never been obvious to me if you comment out a non indented feature if you should comment out the indented sub-feature(s) under that.  To be safe.....I mostly did NOT comment them out.  However, in the currently installed build....it looks like I was a more aggressive and commented out more than usual.  I'm thinking....if you comment out features in an old build and then try and then try a firmware upgrade with an new build (same CFG_HOLDER) that might cause problems with the upgrade in certain cases (OK for Section 1 but not Section 2 or vice-versa).  Although I plan on following the "approved" method more closely going forward......there are certain features like mDNS discovery I'd prefer to turn off.

Frogmore42, 2018-09-13:
The web username is in the code itself, because you can't change it at run time.  This issue has Theo's thinking on that.
https://github.com/arendst/Sonoff-Tasmota/issues/2798

The bigger problem would likely be people having to remember both their username and password.

Frogmore42, 2018-09-13:
Section 2 is just what bits of code / features get compiled into the code.  What you need to comment out is going to depend on how the actual defines are used.  Good practice would be to only need to comment out the main ones, but sometimes mistakes are made and wierd things will happen if one of the sub ones is not commented out.

The serial log will give an indication of what is going wrong.  There is also memory outside of what Tasmota uses that the core uses to save things.  I found I had some serious stability issues when I switched core versions and had to completely erase flash to get to a working state.  But, I don't recall which version the device was before I updated it and found it unstable.

I also don't update my devices to the latest FW just because.  I only update them when there is something important in the new version that I need.  I believe I need/want lwip2, but I am taking it slow to make sure it will work better than what I am currently using.  I now fully expect to have to serial flash each one, so I am going even slower.

ascillato2, 2018-09-13:
Hi,

Closing as your questions has been answered. Please, ask to reopen if needed. Thanks.

closed 2018-09-13 by ascillato2