arendst/Tasmota #3780: Unstable ping

fotiDim, 2018-09-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : Sonoff S20
- [x] Provide the output of command ``status 0`` :
```
19:54:28 CMD: status 0
19:54:28 RSL: STATUS = {"Status":{"Module":8,"FriendlyName":["Toaster"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:54:28 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:38:31","StartupUTC":"2018-09-12T18:15:57","Sleep":0,"BootCount":14,"SaveCount":32,"SaveAddress":"F8000"}}
19:54:28 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:54:28 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["zouzounospito","zouzounospito"],"TelePeriod":300,"SetOption":["00008001","55A18000","00000000"]}}
19:54:28 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":11,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
19:54:28 RSL: STATUS5 = {"StatusNET":{"Hostname":"toaster","IPAddress":"192.168.0.203","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:7F:EA:10","Webserver":2,"WifiConfig":5}}
19:54:28 RSL: STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 18:54:28 2018","Local":"Wed Sep 12 19:54:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:22","Sunset":"19:10"}}
19:54:28 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-09-12T19:54:28"}}
19:54:28 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-09-12T19:54:28","Uptime":"0T00:38:31","Vcc":3.153,"POWER":"OFF","Wifi":{"AP":1,"SSId":"zouzounospito","RSSI":78,"APMac":"F0:9F:C2:F4:99:E2"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

The ping is highly unstable for some reason on 2 S20s that I tried. I am using the latest sonoff.bin downloaded today from Github releases.
```
64 bytes from 192.168.0.203: icmp_seq=976 ttl=128 time=615.050 ms
64 bytes from 192.168.0.203: icmp_seq=977 ttl=128 time=107.754 ms
64 bytes from 192.168.0.203: icmp_seq=978 ttl=128 time=17.868 ms
64 bytes from 192.168.0.203: icmp_seq=979 ttl=128 time=29.991 ms
Request timeout for icmp_seq 980
Request timeout for icmp_seq 981
Request timeout for icmp_seq 982
Request timeout for icmp_seq 983
Request timeout for icmp_seq 984
Request timeout for icmp_seq 985
Request timeout for icmp_seq 986
Request timeout for icmp_seq 987
Request timeout for icmp_seq 988
Request timeout for icmp_seq 989
Request timeout for icmp_seq 990
64 bytes from 192.168.0.203: icmp_seq=991 ttl=128 time=194.596 ms
64 bytes from 192.168.0.203: icmp_seq=992 ttl=128 time=101.530 ms
64 bytes from 192.168.0.203: icmp_seq=993 ttl=128 time=115.332 ms
64 bytes from 192.168.0.203: icmp_seq=994 ttl=128 time=48.569 ms
Request timeout for icmp_seq 995
64 bytes from 192.168.0.203: icmp_seq=996 ttl=128 time=21.333 ms
64 bytes from 192.168.0.203: icmp_seq=997 ttl=128 time=18.896 ms
64 bytes from 192.168.0.203: icmp_seq=998 ttl=128 time=337.202 ms
Request timeout for icmp_seq 999
64 bytes from 192.168.0.203: icmp_seq=1000 ttl=128 time=76.171 ms
64 bytes from 192.168.0.203: icmp_seq=1001 ttl=128 time=59.231 ms
64 bytes from 192.168.0.203: icmp_seq=1002 ttl=128 time=21.367 ms
64 bytes from 192.168.0.203: icmp_seq=1003 ttl=128 time=54.954 ms
64 bytes from 192.168.0.203: icmp_seq=1004 ttl=128 time=87.545 ms
64 bytes from 192.168.0.203: icmp_seq=1005 ttl=128 time=199.862 ms
64 bytes from 192.168.0.203: icmp_seq=1006 ttl=128 time=121.973 ms
64 bytes from 192.168.0.203: icmp_seq=1007 ttl=128 time=84.987 ms
64 bytes from 192.168.0.203: icmp_seq=1008 ttl=128 time=60.911 ms
64 bytes from 192.168.0.203: icmp_seq=1009 ttl=128 time=74.111 ms
Request timeout for icmp_seq 1010
64 bytes from 192.168.0.203: icmp_seq=1011 ttl=128 time=157.315 ms
64 bytes from 192.168.0.203: icmp_seq=1012 ttl=128 time=142.097 ms
64 bytes from 192.168.0.203: icmp_seq=1013 ttl=128 time=41.044 ms
64 bytes from 192.168.0.203: icmp_seq=1014 ttl=128 time=53.302 ms
64 bytes from 192.168.0.203: icmp_seq=1015 ttl=128 time=214.672 ms
64 bytes from 192.168.0.203: icmp_seq=1016 ttl=128 time=325.445 ms
Request timeout for icmp_seq 1017
Request timeout for icmp_seq 1018
Request timeout for icmp_seq 1019
Request timeout for icmp_seq 1020

```

fotiDim, 2018-09-12:
@ascillato2 I am pinging other devices at the same time and their ping is stable. I tried the dev version you suggest using file upload and the problem remains. Is there any sleep setting that I need to change?

ascillato2, 2018-09-12:
If your status 0 is correct, you have sleep set to 0. So, there isn't anything else.

For the ping I just use the ping command from a windows console.
How are you making the ping?
Are you experiencing also a lag in the webmenu? 
You have MQTT off. Is that correct? How are you controlling your sonoff?

fotiDim, 2018-09-12:
- Mac terminal
- Yes the lag is also in web menu. Sometimes it doesn't even load
- MQTT is off. I use Wemo emulation

ascillato, 2018-09-12:
Please, use the command `weblog 4` in the Tasmota console to see if there is a disconnection of something being reported there.

ascillato, 2018-09-12:
mmm, there were some issues with Access points before. May be that is the problem. The Wifi libraries for ESP8266 devices have some issues with some access points. Please, try to connect your Sonoff directly to your main router.

fotiDim, 2018-09-12:
There is no wireless router in my case. Only those access points: https://www.ubnt.com/unifi/unifi-ap-ac-lite

ascillato, 2018-09-12:
See for example this comment in another issue: https://github.com/arendst/Sonoff-Tasmota/issues/3262#issuecomment-419724799

ascillato, 2018-09-12:
There is another thing to try. Recompile Tasmota but using esp lib core v2.4.2 instead of the v2.3.0 that you have now. May be this helps with your connection stability with your access points

fotiDim, 2018-09-12:
I can try that. Thanks for the tip.
Is there any way that I can disable mDNS without recompiling?

ascillato, 2018-09-12:
sorry, no. you need to recompile

Frogmore42, 2018-09-13:
I would try turning off Hue emulation.  It has caused problems on some people's networks.

fotiDim, 2018-09-13:
@Frogmore42 tried and did absolutely nothing. I guess recompiling is my only option.

Frogmore42, 2018-09-13:
Did you update the FW on the access points?
I would do that first.  If that doesn't help try core 2.4.2.  I am using it LWIP 2 high bandwidth.

fotiDim, 2018-09-13:
Access are update yes. I took the S20 to the office today and here with this WiFi it works fine. So clearly there is an incompatibility between Unifi access points and Sonoff or more specifically Tasmota.

I will try to downgrade and also to recompile a custom version today.

@Frogmore42 where can I find information about `LWIP 2 high bandwidth`?

d-a-v, 2018-09-13:
Not willing to step on any toes,
the ping problem is well known on the esp8266/arduino core side,
Current easiest solution is to use `WiFi.setSleepMode(WIFI_NONE_SLEEP)`
The down side of this is that it has more current consumption. This may be solved in a better way when arduino folks will update to the espressif's V3 SDK.

About `LWIP2`, it is with the [2.4.2 arduino core](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE) (variant: v2 Higher Bandwidth).

(*sidenote*: I ordered my first sonoff modules, and I love the KNX functionality)

ascillato, 2018-09-13:
> So clearly there is an incompatibility between Unifi access points and Sonoff

I'm sorry to say that you are right. There are several issues naming Unifi Access Points. Other brands work fine. I don't know what have this one in particular. May be you can play with its internal parameters.

ascillato2, 2018-09-13:
Closing issue as the problem is outside Tasmota software. (Unifi Access Points and ESP8266 core libraries).

If you find any workaround to this, please share it. Thanks.

closed 2018-09-13 by ascillato2

fotiDim, 2018-09-13:
@Frogmore42 I tried 2.4.2 but didn't fix my issue. I also tried disabling sleep and I was still getting timeouts. And yes it is a real issue. Web interface was not working and also Home Assistant lost connectivity. I tried multiple S20s. All have the same issue.

I tried now to revert 20a5395 as suggested [here](https://github.com/arendst/Sonoff-Tasmota/issues/1366#issuecomment-401038540). Things are more stable but still not perfect. I have ~5% packet loss. I am continuing to investigate.

Frogmore42, 2018-09-14:
Core 2.4.2 can be used with three varieties of lwip, which did you try?

Have you tried a simple WiFi example from the esp8266 Arduino repo?  Does it have the same issue with ping?

fotiDim, 2018-09-20:
@Frogmore42 I did a thorough test. I set up 3 x S20s, all with 2.4.2 and each one with a different lwip setting. All other settings were as default. All the S20s were plugged into the same power source and had direct eye contact to the AP. I left simultaneous pings going on for a couple of hours. Results:

LWIP_HIGHER_BANDWIDTH : 16680 packets transmitted, 16610 packets received, 0.4% packet loss
LWIP2_LOW_MEMORY: 16677 packets transmitted, 16341 packets received, 2.0% packet loss
LWIP2_HIGHER_BANDWIDTH: 16681 packets transmitted, 16296 packets received, 2.3% packet loss

I also tried an Arduino Wifi example but didn't work for me. Not sure if it's worth spending more time with it.

Is seems LWIP_HIGHER_BANDWIDTH is more stable.

Frogmore42, 2018-09-21:
LWIP (not 2) does not work across long distances, ie VPN.  LWIP2 High bandwidth is supposed to work better over VPN.  It is also the path that Espressif has said they will be supporting.  If lwip is working for your situation, use it.  There are some other security reasons to move to lwip2, but most people are not too worried about security.  If you are there are not a lot of good choices, as most IOT devices have pretty poor security.