arendst/Tasmota #3782: STATUS 8 vs. STATUS 10?

meingraham, 2018-09-12:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)

Sonoff 31
v6.1.2.3

`STATUS 8` and `STATUS 10` appear to perform the function. WHy the duplication? The Sonoff POW wiki mentions using `STATUS 8`. Is there a difference?

ascillato, 2018-09-12:
Status 0 of a POW:
```
16:53:11 CMD: status 0
16:53:11 MQT: stat/sonoffPOW/STATUS = {"Status":{"Module":43,"FriendlyName":["SonoffPOW"],"Topic":"sonoffPOW","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:53:11 MQT: stat/sonoffPOW/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:34","StartupUTC":"2018-09-12T19:52:37","Sleep":0,"BootCount":43,"SaveCount":186,"SaveAddress":"F6000"}}
16:53:11 MQT: stat/sonoffPOW/STATUS2 = {"StatusFWR":{"Version":"6.2.0.2","BuildDateTime":"2018-09-04T11:33:41","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:53:11 MQT: stat/sonoffPOW/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["04208009","55818000","00000001"]}}
16:53:11 MQT: stat/sonoffPOW/STATUS4 = {"StatusMEM":{"ProgramSize":488,"Free":512,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F80A984","0C0003B0","23021796","00000000"]}}
16:53:11 MQT: stat/sonoffPOW/STATUS5 = {"StatusNET":{"Hostname":"sonoffPOW-3696","IPAddress":"192.168.1.34","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:66:8E:70","Webserver":2,"WifiConfig":5}}
16:53:11 MQT: stat/sonoffPOW/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_668E70","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:53:11 MQT: stat/sonoffPOW/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 12 19:53:11 2018","Local":"Wed Sep 12 16:53:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"07:37","Sunset":"19:19"}}
16:53:11 MQT: stat/sonoffPOW/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:53:11 MQT: stat/sonoffPOW/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T16:53:11","ENERGY":{"Total":1.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":228,"Current":0.000}}}
16:53:11 MQT: stat/sonoffPOW/STATUS11 = {"StatusSTS":{"Time":"2018-09-12T16:53:11","Uptime":"0T00:00:34","Vcc":3.502,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":72}}}
```

Tele Status of a POW:
```
16:52:56 MQT: tele/sonoffPOW/STATE = {"Time":"2018-09-12T16:52:56","Uptime":"0T00:00:19","Vcc":3.504,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","BSSId":"18:D6:C7:80:38:2C","Channel":1,"RSSI":76}}
16:52:56 MQT: tele/sonoffPOW/SENSOR = {"Time":"2018-09-12T16:52:56","ENERGY":{"Total":1.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"Factor":0.00,"Voltage":230,"Current":0.000}}
```

Status 8 of a POW:
```
16:53:54 CMD: status 8
16:53:54 MQT: stat/sonoffPOW/STATUS8 = {"StatusSNS":{"Time":"2018-09-12T16:53:54","ENERGY":{"Total":1.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":228,"Current":0.000}}}
```

Status 8 and 10 together:
```
16:53:54 MQT: stat/sonoffPOW/STATUS8 = {"StatusSNS":{"Time":"2018-09-12T16:53:54","ENERGY":{"Total":1.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":228,"Current":0.000}}}
16:53:11 MQT: stat/sonoffPOW/STATUS10 = {"StatusSNS":{"Time":"2018-09-12T16:53:11","ENERGY":{"Total":1.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":228,"Current":0.000}}}
```

So, no difference.

I don't know. May be Theo has an explanation.

andrethomas, 2018-09-12:
Codeblock used is the same - perhaps reverse compatibility kept from an older version for posterity purposes.

andrethomas, 2018-09-12:
#define D_STATUS8_POWER "PWR"

So maybe status 8 will in future provide other information not contained in status 10

arendst, 2018-09-13:
Just  for legacy reason.

ascillato2, 2018-09-13:
Hi,

Closing as your questions has been answered. Please, ask to reopen if needed. Thanks.

closed 2018-09-13 by ascillato2

meingraham, 2018-09-13:
A note in the wiki to that effect perhaps? Just to clear up any confusion.

ascillato, 2018-09-13:
Good Idea. Can you do that? Thanks

meingraham, 2018-09-13:
Done.

andrethomas, 2018-09-13:
@meingraham Thanks ;)