arendst/Tasmota #3783: WIFI/Doesnt- stay connect

JPM-git, 2018-09-12:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Hardware used : sonoff basic R2
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

Version 5.14

00:00:00 CFG: Use defaults
00:00:00 Project sonoff2 Sonoff (Topic sonoff2, Fallback DVES_742EA0, GroupTopic sonoffs) Version 5.14.0-2_4_0
00:00:00 WIF: Connecting to AP1 Beubeu in mode 11N as sonoff2-3744...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff2-3744.local with IP address 192.168.2.25
20:00:45 MQT: Attempting connection...
20:00:45 MQT: Connected
20:00:45 MQT: tele/sonoff2/LWT = Online (retained)
20:00:45 MQT: cmnd/sonoff2/POWER = 
20:00:45 MQT: tele/sonoff2/INFO1 = {"Module":"Sonoff Basic","Version":"5.14.0","FallbackTopic":"DVES_742EA0","GroupTopic":"sonoffs"}
20:00:45 MQT: tele/sonoff2/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff2-3744","IPAddress":"192.168.2.25"}
20:00:45 MQT: tele/sonoff2/INFO3 = {"RestartReason":"Power on"}
20:00:45 MQT: stat/sonoff2/RESULT = {"POWER":"ON"}
20:00:45 MQT: stat/sonoff2/POWER = ON
20:00:46 MQT: stat/sonoff2/RESULT = {"POWER":"ON"}
20:00:46 MQT: stat/sonoff2/POWER = ON
20:00:53 MQT: tele/sonoff2/STATE = {"Time":"2018-09-12T20:00:53","Uptime":"0T00:00:14","Vcc":3.151,"POWER":"ON","Wifi":{"AP":1,"SSId":"Beubeu","RSSI":100,"APMac":"14:AA:E9:F8:B1:2C"}}
20:01:08 MQT: Attempting connection...
20:01:08 MQT: Connected
20:01:08 MQT: tele/sonoff2/LWT = Online (retained)
20:01:08 MQT: cmnd/sonoff2/POWER = 
20:01:30 MQT: Attempting connection...
20:01:30 MQT: Connected
20:01:30 MQT: tele/sonoff2/LWT = Online (retained)
20:01:30 MQT: cmnd/sonoff2/POWER = 
20:02:00 RSL: tele/sonoff2/UPTIME = {"Time":"2018-09-12T20:02:00","Uptime":"0T00:01:21"}
20:02:01 MQT: Attempting connection...
20:02:01 MQT: Connected
20:02:01 MQT: tele/sonoff2/LWT = Online (retained)
20:02:01 MQT: cmnd/sonoff2/POWER = 

Version 6.2.0

00:00:00 CFG: Use defaults
00:00:00 Project sonoff2 Sonoff (Topic sonoff2, Fallback DVES_742EA0, GroupTopic sonoffs) Version 6.2.0-2_4_0
00:00:00 WIF: Connecting to AP1 Beubeu in mode 11N as sonoff2-3744...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff2-3744.local with IP address 192.168.2.25
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/sonoff2/LWT = Online (retained)
00:00:08 MQT: cmnd/sonoff2/POWER = 
00:00:08 MQT: tele/sonoff2/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.0","FallbackTopic":"DVES_742EA0","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/sonoff2/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff2-3744","IPAddress":"192.168.2.25"}
00:00:08 MQT: tele/sonoff2/INFO3 = {"RestartReason":"Power on"}
00:00:08 MQT: stat/sonoff2/RESULT = {"POWER":"OFF"}
00:00:08 MQT: stat/sonoff2/POWER = OFF
00:00:08 MQT: stat/sonoff2/RESULT = {"POWER":"OFF"}
00:00:08 MQT: stat/sonoff2/POWER = OFF
00:00:09 MQT: stat/sonoff2/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:00:09 MQT: stat/sonoff2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:09","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
00:00:09 MQT: stat/sonoff2/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-09T23:15:16","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:00:09 MQT: stat/sonoff2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Beubeu","indebuurt2"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
00:00:09 MQT: stat/sonoff2/STATUS4 = {"StatusMEM":{"ProgramSize":510,"Free":492,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B617CE","00000000"]}}
00:00:09 MQT: stat/sonoff2/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-3744","IPAddress":"192.168.2.25","Gateway":"192.168.2.65","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.65","Mac":"60:01:94:74:2E:A0","Webserver":2,"WifiConfig":2}}
[... 32 lines cut ...]
00:00:05 MQT: stat/sonoff/POWER = OFF
00:00:05 MQT: homeassistant/light/sonoff_1/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_1/config = {"name":"Sonoff","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline"} (retained)
00:00:05 MQT: homeassistant/light/sonoff_2/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_2/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_3/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_3/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_4/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_4/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_5/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_5/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_6/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_6/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_7/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_7/config =  (retained)
00:00:05 MQT: homeassistant/light/sonoff_8/config =  (retained)
00:00:05 MQT: homeassistant/switch/sonoff_8/config =  (retained)
00:00:14 MQT: tele/sonoff/STATE = {"Time":"1970-01-01T00:00:14","Uptime":"0T00:00:14","Vcc":3.270,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Beubeu","RSSI":100,"APMac":"14:AA:E9:F8:B1:2C"}}
**(Please, remember to close the issue when the problem has been addressed)**

What can I add...every version the same behavior:
I flash
Restarts
connect to routher
after a few seconds reboot and never connects again

ascillato, 2018-09-12:
Hi,

Sorry but all your firmwares are being compiled with a ESP LIB version that it is known to be unstable and full of issues (v2.4.0)

Please, use the precompiled bins or compile your own using the esplib v2.3.0 (if you will use the sleep command) or v2.4.2 (if you will not use the sleep command)

JPM-git, 2018-09-12:
So not sure why you said that but all fw were compiled using a 2.3.0 but after updating to v2.4.2 the result is the same

JPM-git, 2018-09-12:
Not sure if it helps but on versions 6.x i-m using 
#define USE_CONFIG_OVERRIDE  
and changed 
#define CFG_HOLDER             4618 (+1 from that came)

ascillato, 2018-09-12:
from your logs:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_742EA0, GroupTopic sonoffs) Version 6.2.1-**_2_4_0_**

2_4_0 means that your are using esp lib v2.4.0 that is buggy version

Please, try with the precompiled bins from Tasmota repository

JPM-git, 2018-09-12:
So i-m doing something wrong because on arduino ide (board Manager ) I see
[image]

ascillato, 2018-09-12:
try uninstall it and install again

JPM-git, 2018-09-12:
Shell I try 2.4.2 since i don't use sleep or go for the stable 2.3.0?

ascillato, 2018-09-12:
yes please

JPM-git, 2018-09-12:
with 2.3.0 I don't have the option spiffs 1M (no SPIFFS)

[image]

ascillato, 2018-09-12:
You need to follow the instructions in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE)

And please, try the v2.4.2 that is better. If you can do an erase before upload (option available in v2.4.2)

JPM-git, 2018-09-12:
Now i'm 100% sure it's something i'm doing because i can't even compile the project:

Steps i made:

remove esp lib
close arduino
open
add esp 2.4.2
donwload 6.2.1
unpack
remove all the "old" libs from 6.2.0 from the "Sketchbook location"
added all the libs from 6.2.1 to the  "Sketchbook location"
update ide sketch settings
[image]

And now i get this erros on compilation:
Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

sonoff:1:1: error: 'esp' does not name a type

 esp/*

 ^

In file included from Z:\pessoal\Home Assistance\tasmota\Sonoff-Tasmota-6.2.1\sonoff\sonoff.ino:75:0:

settings.h:224:3: error: 'power_t' does not name a type

   power_t       power;                     // 2E8

   ^

settings.h:349:3: error: 'power_t' does not name a type

   power_t       power;                     // 01C

   ^

sonoff:298:23: error: variable or field 'SetLatchingRelay' declared void

 void SetLatchingRelay(power_t power, uint8_t state)

                       ^

sonoff:298:23: error: 'power_t' was not declared in this scope

sonoff:298:46: error: expected primary-expression before 'state'

 void SetLatchingRelay(power_t power, uint8_t state)

                                              ^

sonoff:315:21: error: variable or field 'SetDevicePower' declared void

 void SetDevicePower(power_t rpower, int source)

                     ^

sonoff:315:21: error: 'power_t' was not declared in this scope

sonoff:315:37: error: expected primary-expression before 'int'

 void SetDevicePower(power_t rpower, int source)

                                     ^

xdrv_interface:156:19: error: variable or field 'XdrvSetPower' declared void

 void XdrvSetPower(power_t mpower)

                   ^

xdrv_interface:156:19: error: 'power_t' was not declared in this scope

sonoff:125:1: error: 'power_t' does not name a type

 power_t power = 0;                          // Current copy of Settings.power

 ^

sonoff:126:1: error: 'power_t' does not name a type

 power_t blink_power;                        // Blink power state

 ^

sonoff:127:1: error: 'power_t' does not name a type

 power_t blink_mask = 0;                     // Blink relay active mask

 ^

sonoff:128:1: error: 'power_t' does not name a type

 power_t blink_powersave;                    // Blink start power save state

 ^

sonoff:129:1: error: 'power_t' does not name a type

 power_t latching_power = 0;                 // Power state at latching start

 ^

sonoff:130:1: error: 'power_t' does not name a type

 power_t rel_inverted = 0;                   // Relay inverted flag (1 = (0 = On, 1 = Off))

 ^

sonoff:298:23: error: variable or field 'SetLatchingRelay' declared void

 void SetLatchingRelay(power_t power, uint8_t state)

                       ^

sonoff:298:23: error: 'power_t' was not declared in this scope

sonoff:298:46: error: expected primary-expression before 'state'

 void SetLatchingRelay(power_t power, uint8_t state)

                                              ^

In file included from C:\Users\Perdizes\Documents\Arduino\libraries\NeoPixelBus-2.2.9\src/NeoPixelBus.h:60:0,

                 from Z:\pessoal\Home Assistance\tasmota\Sonoff-Tasmota-6.2.1\sonoff\xplg_ws2812.ino:25:

C:\Users\Perdizes\Documents\Arduino\libraries\NeoPixelBus-2.2.9\src/internal/NeoBitmapFile.h:30:22: error: expected declaration before end of line

 #pragma pack(push, 2)

                      ^

exit status 1
'esp' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ascillato, 2018-09-12:
he he, now it turns to be an Arduino issue.

I know that this would sound repetitive but: try to uninstall all Arduino and install it again from scratch.

JPM-git, 2018-09-12:
sure
I'm the one having trouble I have to help solve then in any way.

ascillato, 2018-09-12:
If helps, I'm using Arduino 1.8.2 in windows 7 and works perfect.

Also you can try platformio.

JPM-git, 2018-09-12:
well 1.8.7 from windows 10 store...no lucky
1.8.7 for windows installer...same lucky

1.8.2 compiled ok?!??!!?

uploading as we speak

JPM-git, 2018-09-12:
can't get connection so...no :(

ascillato, 2018-09-12:
Please, let's continue this troubleshooting in the Tasmota Chat

ascillato, 2018-09-12:
From the chat:

Installing Arduino 1.8.2 from scratch with esplib v2.3.0 and erase and upload to the device have solved the issue.

closed 2018-09-12 by ascillato2