arendst/Tasmota #3785: Nested Backlogs

Pewibe, 2018-09-13:
**Is your feature request related to a problem? Please describe.**
I like to create a RULE by using BACKLOG to apply more than one command. This rule should by part of an outer BACKLOG to send all commands in one step.
Backlog Command X; Command Y; Rule1 on Event#X do Backlog Command A; Command B; endon; Command Z;

**Describe the solution you'd like**
If the BACKLOG is parsed (search for semicolon) the rule command (on ... endon) should be skipped.

**Describe alternatives you've considered**
1. BACKLOG command gets a corresponding ENDBACKLOG to recognize inner / outer loops.
2. BACKLOG gets a second character like semicolon to concatinate commans, e.g. & or _
3. The command part of a RULE is put into ". . ." like Rule1 on Event#X do "Backlog Command A; Command B;" endon;

Frogmore42, 2018-09-13:
Why?
What problem are you trying to solve?
If you need complex rules, use a Home Automation system where you can have arbitrarily complex rules and at least the hope of being able to debug them to get them correct.

ascillato2, 2018-09-13:
Hi,

Your idea is good but it is complex to explain in the wiki and complex to add in the code. So, as we are trying to have Tasmota as simple as possible, a Label to this issue has been added for future reference to be taken into account.

Thanks for share your ideas.

closed 2018-09-13 by ascillato2

benzino77, 2018-09-13:
Hi,
Maybe something like three (or N) `backlog1, backlog2, backlog3` commands can be implemented?

Prickelpit, 2018-09-13:
@Frogmore42 
Let me answer philosophical: When I learned first programming language, beside the IF and kind of LOOP statement I felt that a FUNCTION (use code more than once) is a strong feaqture. The BACKLOG construct is nothing else and it's there. Only the implementation not nested was an unfortune step.

@ascillato2 
My suggested solution no. 0  (on ... endon) is not complex to explain because you do not have to explain something except "Inside a rule another BACKLOG can be used". About complexity in the code I cannot say anything. This is why I suggested even solution no. 2 which extends code to parallel parse for ";" and "&".

@benzino77 
I had the same idea but didn't described this because of my understanding of complexity. Because BACKLOG is missing most in a rule, a backlog2 would be sufficient, more even better.

Sorry that I was away of this discussion so long. @ascillato2: Maybe you can rewind your decision and let the associated developer decide?

Pewibe, 2018-09-13:
**Sorry, I wrote this answer first erronously under a wrong sing in. Here copied again with correct account **

@Frogmore42
Let me answer philosophical: When I learned first programming language, beside the IF and kind of LOOP statement I felt that a FUNCTION (use code more than once) is a strong feaqture. The BACKLOG construct is nothing else and it's there. Only the implementation not nested was an unfortune step.

@ascillato2
My suggested solution no. 0 (on ... endon) is not complex to explain because you do not have to explain something except "Inside a rule another BACKLOG can be used". About complexity in the code I cannot say anything. This is why I suggested even solution no. 2 which extends code to parallel parse for ";" and "&".

@benzino77
I had the same idea but didn't described this because of my understanding of complexity. Because BACKLOG is missing most in a rule, a backlog2 would be sufficient, more even better.

Sorry that I was away of this discussion so long. @ascillato2: Maybe you can rewind your decision and let the associated developer decide?

Frogmore42, 2018-09-13:
Anybody is free to implement your idea.  Maybe Theo will think it is an excellent idea and spend some time implementing it.  Or maybe you will figure out how to do it. 
 Based on some of his previous responses I wouldn't expect Theo to do it soon.  But, I was surprised he added rules in the first place.
My response was to provide a method to achieve your goals, given the feature you want isn't implemented currently.

Rules can be really powerful, but they can be hard to understand, write, and debug.  Doing it in Tasmota might have an advantage in some cases, but it comes with a higher cost in all the things that are hard.

In a distributed system it is always a design decision where to implement intelligence.  Over the last 30+ years there has been a push to move some intelligence to the edges.  I think that is great, but there is a cost in complexity and understanding, in addition to any gains it might provide.

ascillato, 2018-09-13:
Nice,

But when you type `Rule1 on Clock#Timer=1 do Backlog Power On; Power 2; Delay 10; Power 2; endon` it got stored in EEPROM and will run everytime it gets trigger. You don't need to type it everytime.

What you are requesting: `Backlog Command X; Command Y; Rule1 on Event#X do Backlog Command A; Command B; endon; Command Z;` is a configuration right?

So, as a workaround by now, you can send all that in two steps:
```
Backlog Command X; Command Y; Command Z;
Rule1 on Event#X do Backlog Command A; Command B; endon
```

From a rule you can call also a set of commands using the EVENT trigger (as it is a subroutine)

Examples of this are in the wiki at rules.

Hope this helps as a workaround for you.

Pewibe, 2018-09-14:
Yes, I know that there is no need to type again and again :-) 
And yes, I began with configuration work and thought: good work from these guys because BACKLOG is like a command file. As long as I reached the necessarity to use rules. But I'm convinced that doing configuration is not a shame.
@ascillato2, your suggested workaround is exactly like I started. But immediately I discovered that the 2 steps could be 1 if the implementation of BACKLOG or RULE was made more structured. 
Please take this not as negative critic. It is an amazing job what you Tasmotas do. 
A last question concerning this issue: How about 
a) Instead of a binary config dump an text dump containing the original commands. In this case I can use  "Konfiguration wiederherstellen" as command file input -> no BACKLOG outside RULE needed!
b) An additional action by the build-in webserver like "Kommandos lesen" like this:
[image]

ascillato, 2018-09-14:
Hi,

> Instead of a binary config dump an text dump containing the original commands

that is an open issue in discussion: #2800