arendst/Tasmota #3789: Firmware uploading problem

L00000813, 2018-09-13:
I have downloaded a fresh instance of Arduini IDE and the latest Tasmota version. (due to other issues I was having).

I am now faced with this error when trying to upload firmware..

Invalid library found in C:\Program Files (x86)\Arduino\libraries\esp-epaper-29-ws-20171230-gemu-1.0: no headers files (.h) found in C:\Program Files (x86)\Arduino\libraries\esp-epaper-29-ws-20171230-gemu-1.0

Any idea whats going on?

Thanks.

ascillato, 2018-09-13:
Hi,

Seems that your libraries are not complete. Please, copy all again using the instructions of the wiki. Thanks

L00000813, 2018-09-13:
Please dont give me a reference to a wiki.
If you know the solution let me know, (paste the text) otherwise your reply does nor help.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Sonoff-Tasmota/issues/3789#issuecomment-421052854",
"url": "https://github.com/arendst/Sonoff-Tasmota/issues/3789#issuecomment-421052854",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [arendst/Sonoff-Tasmota] Firmware uploading problem (#3789)",
"sections": [
{
"text": "",
"activityTitle": "**Adrian Scillato**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@ascillato",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"arendst/Sonoff-Tasmota\",\n\"issueId\": 3789,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"arendst/Sonoff-Tasmota\",\n\"issueId\": 3789\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/arendst/Sonoff-Tasmota/issues/3789#issuecomment-421052854"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 379453587\n}"
}
],
"themeColor": "26292E"
}
]

Jason2866, 2018-09-13:
Here is the link where is it described what you have to do.
If you do it exactly as described you will get a working enviroment.
But why compile yourself? You can download compiled binaries.

ascillato, 2018-09-13:
> Please dont give me a reference to a wiki.
> If you know the solution let me know, (paste the text) otherwise your reply does nor help.

Yes, you are right. Sorry. I'm copying and pasting just for you...

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

ascillato, 2018-09-13:
Or may be: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

ascillato, 2018-09-13:
Or just the copy and paste the whole page:

How to setup and configure Arduino IDE for Tasmota compilation and upload.

## Download Arduino IDE
- Download Arduino IDE from https://www.arduino.cc/en/Main/Software

I prefer a *dedicated standalone version* of the IDE allowing easy ESP8266 file manipulation and library management. This can be achieved by downloading the Arduino IDE ZIP file for non admin install.

## Install Arduino IDE
- Unzip the installation file to a known folder.

IMPORTANT: Before executing *arduino.exe* add an empty folder called *portable* in the known folder.

### Install ESP8266 board software
- Open Arduino IDE and select ``File`` - ``Preferences`` and add the following text for field *Additional Boards Manager URLs:* ``http://arduino.esp8266.com/stable/package_esp8266com_index.json`` and select *OK*.
- Open ``Tools`` - ``Boards ...`` - ``Boards Manager ...`` and scroll down and click on *esp8266 by ESP8266 Community*. Click the *Install* button to download and install the latest ESP8266 board software. Select *Close*.

## Download Tasmota
- Download the latest Tasmota release Source code from https://github.com/arendst/Sonoff-Tasmota/releases and unzip to another known folder.

### Copy files
#### ESP8266 Board version **before** 2.4.0
- If not available copy from the Tasmota release Source code folder *arduino\version 2.3.0\tools\sdk\ld* file *eagle.flash.1m0.ld* to Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld*.
- Replace in Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.3.0* file *boards.txt* with the Tasmota Source code file *arduino\version 2.3.0\boards.txt*.

#### ESP8266 Board version 2.4.0
Remove default compile options for floating point ``printf`` and ``scanf`` which are not being used by Tasmota and save 15k code space.
- Replace in Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.4.0* file *platform.txt* with the Tasmota Source code file *arduino\version 2.4.0\platform.txt*.

#### ESP8266 Board version 2.4.1
Remove default compile options for floating point ``printf`` and ``scanf`` which are not being used by Tasmota and save 15k code space.
- Replace in Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.4.1* file *platform.txt* with the Tasmota Source code file *arduino\version 2.4.1\platform.txt*. https://github.com/arendst/Sonoff-Tasmota/tree/development/arduino/version%202.4.1

#### For all versions
- Copy all files from the Tasmota release Source code folder *lib* into your *sketchbook\libraries* folder.
- Copy the Tasmota release Source code folder *sonoff* to your *sketchbook*.

## Configure Arduino IDE

- Select ``Tools`` and verify the following settings for **All Tasmota devices**.

#### ESP8266 Board version 2.3.0:

[image]

#### ESP8266 Board version 2.4.0:

[image]

#### ESP8266 Board version 2.4.1:

[image]

#### ESP8266 Board version 2.4.2:

[image]

### Optional: Prepare for OTA upload
Tasmota release Source code provides scripts to be installed in the Arduino IDE and your webserver to copy the compiled binary to your webserver. This webserver can then provide the firmware via OTA to the device.

- If not available install PHP on your webserver and copy the Tasmota release Source code folder *api* to the root of your webserver.
- If not already done in the previous steps replace in Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.3.0* file *boards.txt* with the Tasmota Source code file *arduino\version 2.3.0\boards.txt*.
- Replace in Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.3.0* file *platform.txt* with the Tasmota Source code file *arduino\version 2.3.0\platform.txt*.
- Copy from the Tasmota release Source code folder *arduino* file *espupload.py* to Arduino IDE folder *portable\packages\esp8266\hardware\esp8266\2.3.0\tools* and change in the script HOST_ADDR to point to your webserver ip address.

After restarting your Arduino IDE you now have an extra option for ``Tools`` - ``Upload Using: "OTA_upload"``.

## Compile Tasmota
Open Arduino IDE and select file *sonoff.ino* from your *sketchbook\sonoff* folder.

- Compile Tasmota with ``Sketch`` - ``Verify/Compile``.

## Upload Tasmota to device
Arduino IDE uses the serial interface to upload the firmware to your device. On Windows these interfaces are named COM1, COM2 etc. On Linux these interfaces are called /dev/ttyUSB0, /dev/ttyUSB1 etc.

In the following commands I use COM5 as an example.

Before using Arduino IDE upload make sure you know to which serial interface name your device is connected to. 

### Put device in firmware upload mode
When performing a firmware upload do **not connect the device to AC** but use the power supply provided by your (FTDI type) serial interface.

- Put the device in firmware upload mode by grounding pin GPIO00 while applying power.

Grounding pin GPIO00 can often be achieved by pressing button 1 on the Sonoff device or using a wire between GPIO00 and Gnd if the button is not available. Deviations may apply.

- Connect the serial interface of your PC to the device while GPIO00 to Gnd.

### Perform serial upload
- Make the correct serial interface selection in the Arduino IDE via ``Tools`` - ``Port: "COM5"``.
- Upload the compiled firmware with ``Sketch`` - ``Upload``.

NOTE: For a proper device initialization after first firmware upload power down and power up the device.

## Optional: Upload Tasmota to OTA server
If a webserver is available you can upload the compiled firmware using optional scripts and prepare it for OTA download by any Tasmota device using the MQTT ``upgrade 1`` or ``upgrade 5.1.2`` command.

- Make sure that ``Tools`` - ``Upload Using: "OTA_upload"`` is selected.
- Upload the compiled firmware to the OTA server with ``Sketch`` - ``Upload``.

meingraham, 2018-09-13:
@L00000813 Gavin,

I'll point you in the direction... I have also had Arduino IDE issues. I found that I had to follow the installation and configuration to the minutest detail in the wiki write-up. Following that, I found that when downloading a TASMOTA source package, the libraries within need to replace any and all libraries you might have downloaded from a previous TASMOTA package. My strategy has become to rename the libraries folder from previous TASMOTA packages to append the TASMOTA version. I then create a new 'libraries' folder and extract all the libraries from the TASMOTA 'lib' folder. The same follows for the sonoff folder contents. This needs to be completely replaced. Again, I rename the previous sonoff folder appending the version. This allows me to copy the user_config_override.h and to retain the previous user_config.h as reference in editing the new user_config.h. Once I have a "fresh" install with my user_config customizations, I usually can click compile and have no errors.

So, the links you need which I am now pasting per your request:
- https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE
- https://github.com/arendst/Sonoff-Tasmota/archive/development.zip

But, as @Jason2866 suggests, you could avoid all compilation issues by grabbing a pre-compiled binary, flashing that, and then going through the configuration at "runtime". Pasted for your convenience, here is the link to the releases of pre-compiled firmware binaries:
- https://github.com/arendst/Sonoff-Tasmota/releases

I'm afraid you're in for disappointment. The expectation is that once the problem is identified, it is up to you to continue the research and find the information you've been pointed to. I know it's frustrating because many of us are in the early learning stages of one piece of technology or another (or many more than one concurrently) and we just want the answer. This is my experience in practically every forum I've visited to make my way through a problem (Raspberry Pi, Linux, openHAB, Python, TASMOTA, and on and on). In the end I find that I was guided to find the answer and in turn I learned from the journey and I'm much better equipped to fix, or avoid, issues the next time.

Regards,

Mike

P.S. The quickest way to get people to "mute" you, "bypass" you, or, at best, drag their feed in responding to you, is to bitch at them for not solving YOUR problem. We are all here out of our own goodwill and perhaps even paying back for past assistance. None of us are being compensated for answering your questions. Come with a good attitude; accept what you're given; and be cordial.

P.P.S. @ascillato is much too kind... and patient to a fault!

ascillato, 2018-09-13:
Please, 

In order to help you, step by step find us in the Tasmota Chat.

Thanks.

ascillato2, 2018-09-13:
ups sorry. the link: https://discord.gg/Ks2Kzd4

closed 2018-09-13 by ascillato2

meingraham, 2018-09-13:
@ascillato - yes, must not forget to copy and paste the minute details ;-)

meingraham, 2018-09-13:
@Jason2866 - were you just trolling by not pasting the link? ;-)

Michael Ingraham
Spartanburg, SC

ascillato, 2018-09-13:
Ok, we are all joking. I hope @L00000813 don't get it personal. Just have fun. See us in the chat

L00000813, 2018-09-13:
Ascillato, Jason2866 and meingraham – I appreciate your time and efforts to help with my issue.

I had previously followed the Wiki to no avail as I could not get past the problem I was faced with. I did try several things for a period before reverting to ask the community.

With a combination of your comments and again the Wiki – I have finally got the software compiling and uploading as it should. 

I would conclude that the problem persisted because I had several versions of Tasmota downloaded over time within different directories. Perhaps “things” were not linking to the correct places.

I wiped my PC of everything Arduino and Tasmota and started from scratch.

The issue is resolved.

Thanks.

lincronin, 2019-02-01:
Sketch uses 582272 bytes (60%) of program storage space. Maximum is 958448 bytes.
Global variables use 50524 bytes (61%) of dynamic memory, leaving 31396 bytes for local variables. Maximum is 81920 bytes.
Uploading 586416 bytes from C:\Users\Linda\AppData\Local\Temp\arduino_build_724539/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 13% ]
................................................................................ [ 27% ]
................................................................................ [ 41% ]
................................................................................ [ 55% ]
................................................................................ [ 69% ]
................................................................................ [ 83% ]
................................................................................ [ 97% ]
.............                                                                    [ 100% ]
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib

ANyone have a simple fix to do this??