arendst/Tasmota #3790: Upgrade to v6.2.1 failure

dweston, 2018-09-13:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : 
Sonoff POW
- [x] Provide the output of command ``status 0`` :
```
Unavailable as module is no longer responding.
```
I have been using v5.13.0 for some time on a group of Sonoff POW modules with no problems.
As a precursor to upgrading the group to v6.2.1 I followed the upgrade migration path as advised in the Wiki [https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade#migration-path](url) on a test module I keep aside for this purpose.

OTA upgrade of the test module from v5.13.0 to v5.14.0 (minimal) was successful.
Firmware Source: [https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff-minimal.bin](url)
OTA upgrade from v5.14.0 (minimal) to v5.14.0 (default version) was successful.
Firmware Source: [https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff.bin](url)
OTA upgrade from v5.14.0 (default version) to v6.2.1 (minimal) was successful.
Firmware Source: [https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.2.1/sonoff-minimal.bin](url)
OTA upgrade from v6.2.1 (minimal) to v6.2.1 (default version) failed. The target module is no longer responding at all.
Firmware Source: [https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.2.1/sonoff.bin](url)
I'm presuming I will now have no choice but to resort to a direct connection flashing in order to 'revive' the module.

Is there something I may have missed in the OTA firmware migration sequence?

ascillato, 2018-09-13:
> I'm presuming I will now have no choice but to resort to a direct connection flashing in order to 'revive' the module.

I'm sorry, yes, you are right. You need to flash by cable.

> Is there something I may have missed in the OTA firmware migration sequence?

Your sequence was ok. Some other thing has failed in the last OTA.

ascillato, 2018-09-13:
Hi, I have just tested 6.2.1 minimal to 6.2.1 default and works. So, sorry, but could be a wifi miscommunication or something like that.

Hope that you could flash by cable easily.

Thanks for share your issue.

closed 2018-09-13 by ascillato2

dweston, 2018-09-14:
@ascillato To ascertain whether the first failure was a glitch I tried the upgrade sequence I related above on a second Sonoff POW previously running v5.13.0 with the same outcome. The transition from 6.2.1 minimal to 6.2.1 default also failed on this second module.

dweston, 2018-09-14:
@ascillato I just tried a third module and this time did the final 6.2.1 minimal to 6.2.1 default transition selecting the "OTA Url" option. Upgrade via that path was successful.