arendst/Tasmota #3793: trigger for dimmer?

Newz20, 2018-09-14:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules
Tell me, please, what kind of trigger you can use for a dimmer to create rules.

ascillato, 2018-09-14:
Please, explain exactly what you want to achieve

ascillato, 2018-09-14:
It is already there. When you change dimmer value, Tasmota sends a mqtt message

closed 2018-09-14 by ascillato2

Newz20, 2018-09-14:
But I need to redirect to another topic

ascillato, 2018-09-14:
From where you are sending the dimmer command?

ascillato, 2018-09-14:
https://discord.gg/Ks2Kzd4

For a more dynamic conversation, please join us at the Discord Chat

devanl, 2019-01-11:
It would be nice if you could trigger rules on the dimmer, I do not believe this is possible with current message?

ascillato, 2019-01-11:
Yes, it is possible and it is implemented. Please, read the wiki at rules. Thanks

devanl, 2019-01-11:
Got it, not sure how I missed that with how many times I read this. Thanks again!