arendst/Tasmota #3794: "Upgrade":"1 or >6.2.0 to upgrade"

MaxBec, 2018-09-14:
Hey guys,

i have my own build firmware on a WebServer. And I entered that web-url in the otaurl field.
If I now enter upgrade 1 in the web console everything works fine.
But if I send the command upgrade 1 via mqtt, there is always the following response and nothing happens:

"Upgrade":"1 or >6.2.0 to upgrade"

What am I doing wrong here?

curzon01, 2018-09-14:
I don't believe you send Update 1 via MQTT

it works as expected (first it's load sonoff-minimal via OtaMagic cause my bin is >512 kB)

`mosquitto_pub -h localhost -u mqttdev -P xxxx -p 1883 -t 'cmnd/dev/esp08/Upgrade' -m '1'
`

output on MQTT sub

`mosquitto_sub -h localhost -u mqttdev -P xxxx -p 1883 -t '#' -v | awk '{ print strftime("%c: "), $0; fflush(); }'`
```
Fr 14 Sep 2018 11:48:02 CEST:  stat/dev/esp08/RESULT {"Upgrade":"Version 6.2.0 from http://tasmota.dev.local/v6.2.0/sonoff.bin"}
Fr 14 Sep 2018 11:48:07 CEST:  tele/dev/esp08/LWT Offline
Fr 14 Sep 2018 11:48:22 CEST:  tele/dev/esp08/LWT Online
Fr 14 Sep 2018 11:48:22 CEST:  cmnd/dev/esp08/POWER (null)
Fr 14 Sep 2018 11:48:37 CEST:  tele/dev/esp08/INFO1 {"Module":"Generic","Version":"6.2.0-minimal","FallbackTopic":"DVES_0EBA54","GroupTopic":"esp"}
Fr 14 Sep 2018 11:48:37 CEST:  tele/dev/esp08/INFO2 {"WebServerMode":"Admin","Hostname":"phesp08","IPAddress":"172.18.10.98"}
Fr 14 Sep 2018 11:48:37 CEST:  tele/dev/esp08/INFO3 {"RestartReason":"Software/System restart"}
Fr 14 Sep 2018 11:48:37 CEST:  stat/dev/esp08/RESULT {"POWER1":"OFF"}
Fr 14 Sep 2018 11:48:37 CEST:  stat/dev/esp08/POWER1 OFF
Fr 14 Sep 2018 11:48:38 CEST:  tele/dev/esp08/LWT Offline
Fr 14 Sep 2018 11:48:55 CEST:  tele/dev/esp08/LWT Online
Fr 14 Sep 2018 11:48:55 CEST:  cmnd/dev/esp08/POWER (null)
Fr 14 Sep 2018 11:48:57 CEST:  stat/dev/esp08/UPGRADE Successful. Restarting
Fr 14 Sep 2018 11:49:13 CEST:  tele/dev/esp08/LWT Online
Fr 14 Sep 2018 11:49:13 CEST:  cmnd/dev/esp08/POWER (null)
Fr 14 Sep 2018 11:49:13 CEST:  tele/dev/esp08/INFO1 {"Module":"Generic","Version":"6.2.0","FallbackTopic":"DVES_0EBA54","GroupTopic":"esp"}
Fr 14 Sep 2018 11:49:13 CEST:  tele/dev/esp08/INFO2 {"WebServerMode":"Admin","Hostname":"phesp08","IPAddress":"172.18.10.98"}
Fr 14 Sep 2018 11:49:13 CEST:  tele/dev/esp08/INFO3 {"RestartReason":"Software/System restart"}

```

What's your console output on this device when you publish your MQTT upgrade command?

MaxBec, 2018-09-14:
That is my output for the following command via MQTT.fx:

tasmota/cmnd/MAXS20004/upgrade 1

```
12:05:47 SRC: MQTT
12:05:47 RSL: Group 0, Index 1, Command UPGRADE, Data 
12:05:47 MQT: tasmota/stat/MAXS20004/RESULT = {"Upgrade":"1 or >6.2.1 to upgrade"}
12:05:48 HTP: Command
```

MaxBec, 2018-09-14:
Found the error. Thank you!

closed 2018-09-14 by MaxBec