arendst/Tasmota #3795: Teleperiod not recognized

andreasloe, 2018-09-14:
**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
Although Teleperiod is set to 150 Blitzwolf reports within several seconds the state of the Sensor. 

**Screenshots**
This is the console report where the Blitzwolf wasn't touched:
```
10:45:36 MQT: tele/blitzwolf2/STATE = {"Time":"2018-09-14T10:45:36","Uptime":"0T00:33:17","Vcc":3.230,"POWER":"ON","Wifi":{"AP":1,"SSId":"WLAN-120954","RSSI":44,"APMac":"38:10:D5:1C:D5:B8"}}
10:45:36 MQT: tele/blitzwolf2/SENSOR = {"Time":"2018-09-14T10:45:36","ENERGY":{"Total":1.33981,"Yesterday":0.77064,"Today":0.56917,"Period":2.450,"Power":58.200,"Factor":0.50,"Voltage":225.900,"Current":0.515}}
10:47:47 MQT: tele/blitzwolf2/SENSOR = {"Time":"2018-09-14T10:47:46","ENERGY":{"Total":1.34189,"Yesterday":0.77064,"Today":0.57125,"Power":29.600,"Factor":0.25,"Voltage":227.500,"Current":0.515}}
10:47:48 MQT: tele/blitzwolf2/SENSOR = {"Time":"2018-09-14T10:47:48","ENERGY":{"Total":1.34192,"Yesterday":0.77064,"Today":0.57128,"Power":58.400,"Factor":0.50,"Voltage":227.500,"Current":0.515}}
10:48:09 MQT: tele/blitzwolf2/STATE = {"Time":"2018-09-14T10:48:09","Uptime":"0T00:35:50","Vcc":3.230,"POWER":"ON","Wifi":{"AP":1,"SSId":"WLAN-120954","RSSI":44,"APMac":"38:10:D5:1C:D5:B8"}}

```

curzon01, 2018-09-14:
Duplicate issue - see
#3778
#3660

and set **PowerDelta 0**

(unfortunately your Status 0 output copy here is cut, so I assume your Powerdelta is ;ne& 0 - see Status 9 which is part of Status 0)

andreasloe, 2018-09-14:
Sorry for the cut, here is status 0.
```
12:11:08 MQT: stat/blitzwolf2/STATUS = {"Status":{"Module":45,"FriendlyName":["blitzwolf2"],"Topic":"blitzwolf2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:11:08 MQT: stat/blitzwolf2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:58:49","StartupUTC":"2018-09-14T08:12:19","Sleep":50,"BootCount":17,"SaveCount":78,"SaveAddress":"F6000"}}
12:11:08 MQT: stat/blitzwolf2/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018.09.13 10:33:12","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:11:08 MQT: stat/blitzwolf2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXX",""],"TelePeriod":150,"SetOption":["00208009","569F8000"]}}
12:11:08 MQT: stat/blitzwolf2/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000407","0FDAE794","0C000000","23B6179E","00000000"]}}
12:11:08 MQT: stat/blitzwolf2/STATUS5 = {"StatusNET":{"Hostname":"blitzwolf2-5461","IPAddress":"192.168.X.X","Gateway":"192.168.X.X","Subnetmask":"255.255.255.0","DNSServer":"192.168.X.X","Mac":"DC:4F:22:2A:XX:XX","Webserver":2,"WifiConfig":3}}
12:11:08 MQT: stat/blitzwolf2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.X.X","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2AD555","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:11:08 MQT: stat/blitzwolf2/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 14 10:11:08 2018","Local":"Fri Sep 14 12:11:08 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:38","Sunset":"19:24"}}
12:11:08 MQT: stat/blitzwolf2/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:11:08 MQT: stat/blitzwolf2/STATUS10 = {"StatusSNS":{"Time":"2018-09-14T12:11:08","ENERGY":{"Total":1.42243,"Yesterday":0.77064,"Today":0.65179,"Power":58.400,"Factor":0.50,"Voltage":226.900,"Current":0.517}}}
12:11:08 MQT: stat/blitzwolf2/STATUS11 = {"StatusSTS":{"Time":"2018-09-14T12:11:08","Uptime":"0T01:58:49","Vcc":3.230,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXX","RSSI":42,"APMac":"38:10:D5:1C:D5:B8"}}}
```
And indeed PowerDelta:80

closed 2018-09-14 by andreasloe