arendst/Tasmota #3798: Wemos D1 loss of connection - Tasmota

francylife89, 2018-09-14:
Dear everyone,

I'm new and I desperately need your help. I'm using 8 wemos d1 mini with Tasmota software version 5.14.0. I installed the devices a few months ago with mqtt disabled to control the lights at home and everything seemed to work perfectly until I have enabled the mqtt and connected everything to hassio version 0.77.2. Suddenly the wemos are no longer reachable (randomly, not all at once) via the web server and I can not ping them. Although they are not reachable I can turn on the lights from the buttons. The only way to reach them via the network is to press the reboot button on the wemos or by restarting the router. When the wemos are reachable, the ping (which I attach) makes me understand that it is as if the router was assigning the same address to multiple devices. The wemos have a static IP address and I'm sure there are no devices in the network with the same IP. 
In the meantime, I realized that the wemos connect / disconnect from the mqtt broker (I attach screen).
I really tried all:
1. I have updated the tasmota software to version 6.2.1
2. I changed the MQTT_LIBRARY_TYPE from MQTT_PUBSUBCLIENT to MQTT_TASMOTAMQTT
3. I have modified the MQTT_KEEPALIVE from 15 to 60
4. I checked if the client IDs were correctly unique and I assigned a username and password to the individual wemos for the login to the broker
5. I changed configurations on the router

I do not know what else to do... 

Does anyone have any idea why this thing happens? Please help me
[image]
[image]
[image]

ascillato, 2018-09-14:
> I desperately need your help

Wow!! call 911 :wink: 

Please, Can you share your `status 0` _(as the troubleshooting template states)_ and also the output of your console when this disconnection happens?

francylife89, 2018-09-14:
Hi! Thank you for your reply. 
Status 0 
[image]
unfortunately I do not have
"output of your Tasmota's console when this disconnection happens"

francylife89, 2018-09-14:
I also send you the status 0 of a wemos that has software 5.14 and mqtt keepalive 15
[image]

ascillato, 2018-09-14:
Hi,

The issue you are experiencing is because of the esplib you are using in your own compilation (2.4.0) That is known to be very buggy and have lots of wifi issues.

Please update that to the latest esp8266 core libraries v2.4.2 in your compiler.

Or even better, use the precompiled bins from the release.

If you need further assistant, do not hesitate on ask.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-09-14 by ascillato2

francylife89, 2018-09-15:
Hi,
I did what you suggested, but the problem unfortunately persists:

11:27:02 CMD: status 0
11:27:02 MQT: stat/test/STATUS = {"Status":{"Module":18,"FriendlyName":["Test"],"Topic":"test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":0}}
11:27:02 MQT: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tests","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin","RestartReason":"External System","Uptime":"0T00:00:13","StartupUTC":"2018-09-15T10:26:49","Sleep":0,"BootCount":5,"SaveCount":13,"SaveAddress":"F7000"}}
11:27:02 MQT: stat/test/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-15T12:20:16","Boot":31,**"Core":"2_4_2"**,"SDK":"2.2.1(cfd48f3)"}}
11:27:02 MQT: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-44584285","LIFE"],"TelePeriod":300,"SetOption":["00008049","55818000","00000000"]}}
11:27:02 MQT: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":516,"Free":484,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000410","0FDAE794","00000001","23B617CE","00000000"]}}
11:27:02 MQT: stat/test/STATUS5 = {"StatusNET":{"Hostname":"test-0725","IPAddress":"192.168.1.134","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"5C:CF:7F:5A:02:D5","Webserver":2,"WifiConfig":4}}
11:27:02 MQT: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.128","MqttPort":1883,"MqttClientMask":"test_%06X","MqttClient":"test_5A02D5","MqttUser":"test","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":60}}
11:27:02 MQT: stat/test/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 15 10:27:02 2018","Local":"Sat Sep 15 11:27:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:41","Sunset":"18:07"}}
11:27:02 MQT: stat/test/STATUS10 = {"StatusSNS":{"Time":"2018-09-15T11:27:02"}}
11:27:02 MQT: stat/test/STATUS11 = {"StatusSTS":{"Time":"2018-09-15T11:27:02","Uptime":"0T00:00:13","Vcc":2.989,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-44584285","RSSI":70,"APMac":"E4:8F:34:0C:16:68"}}}

[image]