arendst/Tasmota #3801: Help to configre Basic sonoff with two windows senores

aihysp, 2018-09-15:
Hi i would like to config two windows sensors on the basic 

i was thinking about using GPIO 0,14

can anyone tell me if this is the right way to do that?
what config should I use for 
switchMode
switchTopic
buttuntopic

any ideas well be welcomed thanks!

andrethomas, 2018-09-15:
Have you considered reading [this](https://github.com/arendst/Sonoff-Tasmota/wiki)

Frogmore42, 2018-09-15:
Yes, while it can sometimes be hard to find what you want the wiki does have lots of good information about your questions as do many of the other issues.

GPIO 0 is a bad choice for an input switch, as stated in the wiki and other issues because it can render your device inoperable if the switch is closed during a power failure.

ascillato2, 2018-09-15:
Hi,

Closing issue as it has been answered. Please, ask to reopen if needed. Thanks.

closed 2018-09-15 by ascillato2

aihysp, 2018-09-15:
ascillato2 you are fast :)

one more Q...
can i use other GPIOs? 3,1,4?
do i need to put some kind of resistor?

andrethomas, 2018-09-15:
Yes you can use other GPIO - experiment with it a little to learn for yourself.