arendst/Tasmota #3802: MCP230XX lost LAST interrupt.

benzino77, 2018-09-15:
**Describe the bug**
MCP230XX sometimes lost information about last MCP interrupt (or maybe doesn't detect it)

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : 
- [x] _Provide the output of command_``status 0`` :
```
09:15:36 CMD: status 0
09:15:36 MQT: stat/test14-wm01/STATUS = {"Status":{"Module":18,"FriendlyName":["Test14 Goouuu"],"Topic":"test14-wm01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:15:36 MQT: stat/test14-wm01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:38:45","StartupUTC":"2018-09-15T06:36:51","Sleep":0,"BootCount":12,"SaveCount":94,"SaveAddress":"3F6000"}}
09:15:36 MQT: stat/test14-wm01/STATUS2 = {"StatusFWR":{"Version":"6.2.0.1","BuildDateTime":"2018-09-14T11:17:48","Boot":5,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:15:36 MQT: stat/test14-wm01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"log.lipowa54.local","LogPort":514,"SSId":["myiots",""],"TelePeriod":300,"SetOption":["00008019","55818000","00000000"]}}
09:15:36 MQT: stat/test14-wm01/STATUS4 = {"StatusMEM":{"ProgramSize":422,"Free":2648,"Heap":17,"ProgramFlashSize":4096,"FlashSize":2048,"FlashMode":3,"Features":["00000809","0F482984","240003A1","000000C6","00000011"]}}
09:15:36 MQT: stat/test14-wm01/STATUS5 = {"StatusNET":{"Hostname":"test14-wm01-7072","IPAddress":"10.10.30.134","Gateway":"10.10.30.3","Subnetmask":"255.255.255.0","DNSServer":"10.10.30.3","Mac":"5C:CF:7F:D6:7B:A0","Webserver":2,"WifiConfig":5}}
09:15:36 MQT: stat/test14-wm01/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.lipowa54.local","MqttPort":2883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D67BA0","MqttUser":"myiots","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:15:36 MQT: stat/test14-wm01/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 15 07:15:36 2018","Local":"Sat Sep 15 09:15:36 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:19","Sunset":"19:03"}}
09:15:36 MQT: stat/test14-wm01/STATUS10 = {"StatusSNS":{"Time":"2018-09-15T09:15:36","MCP230XX":{"D0":0,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":1}}}
09:15:36 MQT: stat/test14-wm01/STATUS11 = {"StatusSTS":{"Time":"2018-09-15T09:15:36","Uptime":"0T00:38:45","Vcc":3.144,"POWER":"OFF","Dimmer":27,"Color":"444444","HSBColor":"0,0,27","Channel":[27,27,27],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"myiots","BSSId":"E6:8D:8C:07:73:1C","Channel":7,"RSSI":62}}}
```

**To Reproduce**
Hi there,
It is not happening all the time but I can observe this behaviour from time to time: the information about LAST interrupt is lost. I understand that the frequency of checking interrupt information from the IC, is not more frequent than 20 times per second (default every 50ms). This is clear for me that some interrupts can be lost and the reported information is the LAST interrupt which is know to MCP when tasmota is doing its job every 50ms. 
The schematic:

[image]

Description of the schematic:
MCP23017 is connected to Wemos D1 Mini through level converter, so the Womos is working on its 3.3V logic and the MCP23017 is working on 5V logic.

 The test:
 I have performed three different tests. The test was very simple: take jumper wire connect one end to GND and the second end to MCP PIN. Then try to connect and disconnect the jumper wire from the pin as fast as I can (simulate pushing and releasing button). All three below test vere performed with `intpri` set to 20 (but I can reproduce the same effect with `intpri` set to 10 or to default 0)
 
 1. D1 PIN with external PULLUP (10k) and debouncing capacitor (22pF)
 ```
13:41:12 CMD: sensor29 1,?
13:41:12 MQT: stat/test14-wm01/SENSOR = {"Sensor29_D1":{"MODE":2,"PULL_UP":"OFF","INT_MODE":"ALL","STATE":"ON"}}
13:41:26 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:26","MCP230XX_INT":{"D1":1,"MS":11089}}
13:41:26 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:27 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:27","MCP230XX_INT":{"D1":0,"MS":993}}
13:41:27 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:28 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:28","MCP230XX_INT":{"D1":1,"MS":1168}}
13:41:28 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:29 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:29","MCP230XX_INT":{"D1":1,"MS":993}}
13:41:29 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:30 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:30","MCP230XX_INT":{"D1":0,"MS":1180}}
13:41:30 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:31 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:41:31","MCP230XX_INT":{"D1":0,"MS":993}}
13:41:31 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:41:39 MQT: tele/test14-wm01/STATE = {"Time":"2018-09-15T13:41:39","Uptime":"0T05:04:48","Vcc":3.147,"POWER":"OFF","Dimmer":27,"Color":"444444","HSBColor":"0,0,27","Channel":[27,27,27],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"myiots","BSSId":"E6:8D:8C:07:73:1C","Channel":7,"RSSI":66}}
13:41:39 MQT: tele/test14-wm01/SENSOR = {"Time":"2018-09-15T13:41:39","MCP230XX":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":1}}
```
As you can see the last reported by firmware interrupt was when the pin changed it's state to "0", but it is "1". Conclusion: information about interrupt when pin changed it's state to "1" is lost.
Pictures of what I see on the broker:
[image]
[image]

2. D2 Pin with external PULLUP (10k). No debouncing capacitor.
```
11:14:10CMD: sensor29 2,?
11:14:10 MQT: stat/test14-wm01/SENSOR = {"Sensor29_D2":{"MODE":2,"PULL_UP":"OFF","INT_MODE":"ALL","STATE":"ON"}}
11:14:16 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
11:14:17 MQT: stat/test14-wm01/{"TIME":"2018-09-15T11: = {"Time":"2018-09-15T11:14:17","MCP230XX_INT":{"D2":0,"MS":992}}
11:14:17 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
11:14:18 MQT: stat/test14-wm01/{"TIME":"2018-09-15T11: = {"Time":"2018-09-15T11:14:18","MCP230XX_INT":{"D2":1,"MS":1291}}
11:14:18 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
11:14:19 MQT: stat/test14-wm01/{"TIME":"2018-09-15T11: = {"Time":"2018-09-15T11:14:19","MCP230XX_INT":{"D2":1,"MS":992}}
11:14:19 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
11:14:21 MQT: stat/test14-wm01/{"TIME":"2018-09-15T11: = {"Time":"2018-09-15T11:14:21","MCP230XX_INT":{"D2":1,"MS":1187}}
11:14:21 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
[... 10 lines cut ...]
13:47:05 CMD: sensor29 3,?
13:47:05 MQT: stat/test14-wm01/SENSOR = {"Sensor29_D3":{"MODE":2,"PULL_UP":"ON","INT_MODE":"ALL","STATE":"ON"}}
13:47:13 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:13","MCP230XX_INT":{"D3":0,"MS":1199}}
13:47:13 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:14 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:14","MCP230XX_INT":{"D3":0,"MS":993}}
13:47:14 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:29 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:29","MCP230XX_INT":{"D3":1,"MS":14472}}
13:47:29 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:30 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:30","MCP230XX_INT":{"D3":1,"MS":1299}}
13:47:30 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:31 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:31","MCP230XX_INT":{"D3":1,"MS":992}}
13:47:31 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:32 MQT: stat/test14-wm01/{"TIME":"2018-09-15T13: = {"Time":"2018-09-15T13:47:32","MCP230XX_INT":{"D3":0,"MS":1197}}
13:47:32 MQT: stat/test14-wm01/EVENT = {"Event":"Done"}
13:47:39 CMD: status 8
13:47:39 MQT: stat/test14-wm01/STATUS8 = {"StatusSNS":{"Time":"2018-09-15T13:47:39","MCP230XX":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":1}}}
```
As you can see the last reported by firmware interrupt was when the pin changed it's state to "0", but it is "1". Conclusion: information about interrupt when pin changed it's state to "1" is lost.
Pictures of what I see on the broker:
[image]

**Expected behavior**
Tasmota detect/doesn't loose information about the LAST interrupt.

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-09-15:
[image]

It is clear that interrupts are lost - most likely because they fall outside of the ~50ms mark or your intpri setting is > 0 (which will make this worse)

Also adding to that you have interrupt reporting enabled for both EVENT and TELEMETRY - The extra CPU time needed to output this could cause further delays in reading interrupts from the MCP chip, especially if you are changing the pin state very quickly.

I cannot reproduce what you are getting whilst using an external oscillator as a signal source at a set frequency of 18hz (just to make sure that I stay below the 20hz margin)

benzino77, 2018-09-15:
But that's OK that 17 interrupts between last check and present check are lost. That's absolutely normal when there is check every 50ms... but the information about the last interrupt should be reported and as you can see it is not. It can even be reported with the delay (because CPU got some other stuff to do) but it should be reported. I've already check two MCPs and two Wemoses boards and the observation is the same (any hint what am I doing wrong, or where I make mistake?).

I can change reporting to "Only Event" but it won't change the situation significantly, because the driver is calling `'event` which results with `stat/test14-wm01/EVENT = {"Event":"Done"}` send to the broker (less information than `stat/test14-wm01/RESULT = {"Time":"2018-09-15T15:07:13","MCP230XX_INT":{"D2":0,"MS":4889210}}` but still it has to send some information to broker)

I'm changing the pin state by hand and I'm pretty sure that I cannot reach 18Hz :)

andrethomas, 2018-09-15:
At all depends at which time the interrupt register was cleared - switching by hand can cause hundreds of triggers as the pin is connected or disconnected - this is the reason why decoupling delays are usually used.

andrethomas, 2018-09-15:
> I can change reporting to "Only Event" but it won't change the situation significantly, because the driver is calling event which results with stat/test14-wm01/EVENT = {"Event":"Done"} send to the [...]

EVENT is only useful if you're going to use it on the device itself - use TELE intmode 2

andrethomas, 2018-09-15:
So proof of concept (removing the nasty effects of using hands to tie pins down and release)

I wrote a small sketch on Arduino Uno:
```
void setup() {
  pinMode(2,OUTPUT);
}

void loop() {
  uint8_t ca;
  digitalWrite(2,HIGH);
  delay(50);
  digitalWrite(2,LOW);
  delay(50);
  digitalWrite(2,HIGH);
  delay(5000);
  for (ca=0;ca<10;ca++) {
    digitalWrite(2,LOW);
    delay(50);
    digitalWrite(2,HIGH);
    delay(50);
  }
  while (1);
}
```

I then connect GND to ESP8266 and D2 of Arduino UNO to D0 of MCP23017 so from the code above its pretty clear what should happen when I hit reset on the Arduino UNO and this is what I get as output
```
18:32:06 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:06","MCP230XX_INT":{"D0":0,"MS":4990}}
18:32:06 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:06","MCP230XX_INT":{"D0":1,"MS":42}}
18:32:06 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:06","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:06 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:06","MCP230XX_INT":{"D0":1,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":42}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":44}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":36}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":38}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":42}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":44}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":42}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":40}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":39}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":0,"MS":39}}
18:32:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T18:32:07","MCP230XX_INT":{"D0":1,"MS":40}}
```
You will see another thing here also that it may actually poll faster than 50ms depending on what else the esp8266 is doing (I know @arendst is still working on finding a better timekeeping mechanism but for most purposes this is accurate enough.

From the delay(5000) we get the expected transition to 0 @ 4990ms which is the beginning of a loop consisting of 10 cycles of HIGH->LOW and LOW->HIGH

So there are 20 interrupts as expected (excluding the ones I use to initialise the Arduino UNO and put in the 5-second delay to show the start of the loop.

So I conclude that there is nothing wrong with the interrupts as such but recommend that you rethink how you're using the MCP23017 chip and the interrupt functionality, most specifically from a mechanical perspective to correctly provide HIGH and LOW states.

andrethomas, 2018-09-15:
Yeah, just confirmed - the millis is wrong because of my placement of where I reset it - I fixed this now and will make a PR

Below is a picture of what I see now on serial console

The blue is caused by this code from Arduino UNO
```
digitalWrite(2,HIGH);
delay(50);
digitalWrite(2,LOW);
delay(50);
digitalWrite(2,HIGH);
```

The red section is caused by the for loop

[image]

There are still some deviation but it evens out on the next interrupt - possibly because the ESP8266 is busy with WIFI or other stuff, but all in all it looks fine to me.

benzino77, 2018-09-15:
You read in my mind with Arduino concept. That was exactly what I was thinking to do! Really, really great support from you. I've got one more idea to test. I will let you know the result.

Offtopic: is there any option to turn off the message `stat/test14-wm01/EVENT = {"Event":"Done"}` every time the event is called?

andrethomas, 2018-09-15:
No, its a response from the firmware because an event was called from the mcp driver. If you do not want to use event's then just set the interrupt mode for telemetry only.

[image]

So for pin 0 and interrupt on a change you do
`sensor29 0,2,0,2`

Then it will only output mqtt JSON on interrupt

benzino77, 2018-09-15:
I will do my tests with arduino tomorrow. For now, I have changed the capacitor from 22pF to 5pF and I got the feeling that changing the pin state by hand, the problem occurs less often.

andrethomas, 2018-09-15:
Cool, no problem - I really do think its the way you're interacting with the MCP chip that is causing what you are getting.

I think we'll close this one for now but you can still post back here tomorrow or whenever you have more feedback without opening it - i will still receive the notification and respond.

Thanks

benzino77, 2018-09-15:
Well, I really don't think that the way I interact with MCP is somehow special or inappropriate ;)
It easy to imagine that we connect the button to the pin and someone just start pushing and releasing it. It is electronic device, that kind of behavior/treatment should not bother it. I do believe that your huge job to make MCP230XX works as extension for input/output in tasmota firmware, will be used mainly for:
1. Ralays
2. Buttons/Switches

I really don't think that there will be huge number of use cases when someone connect oscillator or arduino to input pin.

andrethomas, 2018-09-15:
Yeah, mostly push buttons and PIR sensors and such for input - at least thats how I use it.

By trying to close contact by hand your connection is not as immediate as you may think.

If you have a logic analyser and hooked it up you will see something like this

[image]

That disturbance as the button is being pushed (or in your case, as the jumper is connected) can cause false interrupts and because the polling is at 50ms may result in false last state - the only way to solve this is by externally biasing it correctly.

Read here for more ideas https://www.allaboutcircuits.com/technical-articles/switch-bounce-how-to-deal-with-it/

If you use the same arduino sketch and decrease the delays < 50ms you will see it will miss interrupts. This is a given and is already documented in the wiki.

andrethomas, 2018-09-15:
Ok, so I'll humor your comment on using switches. Check this.. normal push button connected directly to MCP23008 D0 pinmode=2, pullup=1 with no other external components

[image]

```
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":0,"MS":6600}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":1,"MS":59}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":0,"MS":141}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":1,"MS":100}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":0,"MS":100}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":1,"MS":100}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":0,"MS":100}}
22:31:07 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:07","MCP230XX_INT":{"D0":1,"MS":100}}
22:31:08 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:08","MCP230XX_INT":{"D0":0,"MS":100}}
22:31:08 MQT: stat/sonoff/MCP230XX_INT = {"Time":"2018-09-15T22:31:08","MCP230XX_INT":{"D0":1,"MS":100}}
```

benzino77, 2018-09-15:
As I can see you are really fast in pushing button :)

benzino77, 2018-09-16:
I've changed the concept of how to switch the pin state:
[image]

1. I'm far faster than I was thinking. I can generate even 13 changes per second :)
2. With the push buttons I was able to generate "last interrupt lost" only two times (on tens of attempts)

So, yes you ware absolutely right, it was wrong assumption/design/concept that I can use jumper wires to simulate "button" behavior. Especially when I carefully read and watch all those pictures on a allaboutcircuits.com site you provided.

Thanks for a great lesson and all your explanations!

benzino77, 2018-09-16:
https://www.best-microcontroller-projects.com/mcp23017.html here is a good elaboration what happens when you do not read (clear) INCAP or GPIO (50ms interval), which explains possibility of loosing information about last interrupt in Tasmota world.

andrethomas, 2018-09-16:
[image]

Correct, in the case of Tasmota we're not using the hardware interrupt pin of the MCP so we can only read INTCAP at a rate of 50ms (20hz) - Faster than this is not possible due to the fact that the Tasmota firmware is also doing other functions and its not an RTOS in itself.

benzino77, 2018-09-17:
I don't need something which will be better than 50ms check. To tell the truth `teleperiod X` will be enough for me, especially that it will give me information about **actual** state of input pins, by `gpio = MCP230xx_readGPIO(1);`. But I need to know **WHY?**. I've got my answer to that question. 
I think that it will be helpful to other users if such information (about read INTCAP or GPIO and what it leads to in Tasmota world) will be available in the wiki. 

Thanks again for your great support!