arendst/Tasmota #3803: Temp sensor gone Sonoff Basic

Prutsium, 2018-09-15:
**Describe the bug**
Version: 6.2.1.3
I noticed on my sonoff Basic with a DS18b20 sensor that in Domoticz the temperatur did not get updated anymore. So logged into the Sonoff directly and noticed on the main screen no temperature was shown. Checked the config and the GPIO was defined and also the sensor was defined.

When i give i2cscan: (via Console web)
15:27:09 CMD: i2cscan
15:27:09 MQT: domoticz/in/RESULT = {"Command":"Unknown"}

Tried various reboots but sensor did not come back.
I got my sensor back by going to the GPIO setting and save the same setting again.

Will monitor when issue occur again.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [X ] _Searched the problem in issues and in the wiki_
- [X ] _Hardware used_ : Sonoff Basic + Ds18b20 sensor
- [X ] _Provide the output of command_``status 0`` :
```
15:24:28 MQT: domoticz/in/STATUS = {"Status":{"Module":1,"FriendlyName":["Michelle_Aquarium"],"Topic":"Aquarium","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
15:24:28 MQT: domoticz/in/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:01","StartupUTC":"2018-09-15T13:21:27","Sleep":0,"BootCount":80,"SaveCount":180,"SaveAddress":"F7000"}}
15:24:28 MQT: domoticz/in/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-08T19:00:45","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:24:28 MQT: domoticz/in/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["404Domotics",""],"TelePeriod":60,"SetOption":["00008029","55818000","00000001"]}}
15:24:28 MQT: domoticz/in/STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000002C0"]}}
15:24:28 MQT: domoticz/in/STATUS5 = {"StatusNET":{"Hostname":"Michelle_Aquarium","IPAddress":"192.168.99.66","Gateway":"192.168.99.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.1","Mac":"EC:FA:BC:86:A6:17","Webserver":2,"WifiConfig":5}}
15:24:28 MQT: domoticz/in/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.99.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86A617","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:24:28 MQT: domoticz/in/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 15 13:24:28 2018","Local":"Sat Sep 15 15:24:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:26","Sunset":"20:04"}}
15:24:28 MQT: domoticz/in/STATUS10 = {"StatusSNS":{"Time":"2018-09-15T15:24:28","DS18B20":{"Temperature":25.3},"TempUnit":"C"}}
15:24:28 MQT: domoticz/in/STATUS11 = {"StatusSTS":{"Time":"2018-09-15T15:24:28","Uptime":"0T00:03:01","Vcc":3.183,"POWER":"ON","Wifi":{"AP":1,"SSId":"404Domotics","BSSId":"40:16:7E:31:A2:D2","Channel":10,"RSSI":82}}}
```

**To Reproduce**
Not figured out yet as issue occurred for first time.

**Expected behavior**
_A clear and concise description of what you expected to happen._

andrethomas, 2018-09-15:
```
15:27:09 CMD: i2cscan
15:27:09 MQT: domoticz/in/RESULT = {"Command":"Unknown"}
```

For starters - this means that I2C is not configured under device configuration

ascillato2, 2018-09-15:
Hi,

Closing as your question has been answered. Please, ask to reopen if needed. Thanks.

closed 2018-09-15 by ascillato2

Prutsium, 2018-09-16:
Ok today it happened again and sorry for the mistake on I2C.

Status 0 during the issue:
10:11:50 CMD: status 0
10:11:50 MQT: domoticz/in/STATUS = {"Status":{"Module":1,"FriendlyName":["Michelle_Aquarium"],"Topic":"Aquarium","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
10:11:50 MQT: domoticz/in/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-minimal.bin","RestartReason":"Power on","Uptime":"0T00:06:02","StartupUTC":"2018-09-16T08:05:48","Sleep":0,"BootCount":81,"SaveCount":181,"SaveAddress":"F6000"}}
10:11:50 MQT: domoticz/in/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-08T19:00:45","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:11:50 MQT: domoticz/in/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["404Domotics",""],"TelePeriod":60,"SetOption":["00008029","55818000","00000001"]}}
10:11:51 MQT: domoticz/in/STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","000002C0"]}}
10:11:51 MQT: domoticz/in/STATUS5 = {"StatusNET":{"Hostname":"Michelle_Aquarium","IPAddress":"192.168.99.66","Gateway":"192.168.99.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.1","Mac":"EC:FA:BC:86:A6:17","Webserver":2,"WifiConfig":5}}
10:11:51 MQT: domoticz/in/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.99.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86A617","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:11:51 MQT: domoticz/in/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 16 08:11:51 2018","Local":"Sun Sep 16 10:11:51 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"07:28","Sunset":"20:01"}}
10:11:52 MQT: domoticz/in/STATUS10 = {"StatusSNS":{"Time":"2018-09-16T10:11:51"}}
10:11:53 MQT: domoticz/in/STATUS11 = {"StatusSTS":{"Time":"2018-09-16T10:11:52","Uptime":"0T00:06:04","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"404Domotics","BSSId":"40:16:7E:31:A2:D2","Channel":10,"RSSI":70}}}

No DS18x20 is seen by the module even when configured (See screenshots)
[url=https://imgbb.com/][img]https://image.ibb.co/iq5s7K/S1.jpg[/img][/url]
[url=https://imgbb.com/][img]https://image.ibb.co/eq7iEz/S2.jpg[/img][/url]

After pressing just save (no changes made) in Configure module the sensor is back again.

Prutsium, 2018-09-16:
> The DS18B20 does not use i2c, so you don't need to set i2c up to use it and i2c scan won't find it even if it was set up.
> 
> There are several choices for configuring use with DS18B20 sensors. Some of them only support a single sensor, which will then always show up, but will report 0 if the sensor doesn't respond. The other ones will show no sensor if the sensor stops responding.
> 
> You did not indicate if you used a pull-up resistor or not. If you did not, you probably need to add one. 4K7 ohms is the normal size.

4k7 used as pullup and issue actually only appeared after the last upgrade to 6.2.1.3