arendst/Tasmota #3806: Support Domoticz Battery & RSSI for GPIO Switches

jannnfe, 2018-09-15:
Hello,
I would be happy if external switches (in my case Reed sensors) connected to a Sonoff via GPIO with "SwitchMode1 1" & "SwitchTopic Garage DoorSensor" also trigger the parameters RSSI and Battery to Domoticz. Is more of a blemish but would be happy.
Thank you and have a nice day :)

closed 2018-09-15 by ascillato2

jannnfe, 2018-09-16:
Thanks @ascillato 
Unfortunately, this does not work if I enter exactly these commands.
My configuration currently looks like this:

`MQT: stat/Garagentor/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":417,"Rules":"on switch1#state do publish domoticz/in {"idx":63"nvalue":%value%,"RSSI":9,"Battery":56} endon"}`

I checked the Domoticz logs after opening and closing. There, the normal MQTT message is sent without battery. Do you have to pay attention to something here? I do not quite understand it: [https://github.com/arendst/Sonoff-Tasmota/issues/2635#issuecomment-388630780](url)

ascillato, 2018-09-16:
Do you have that gpio configured as switch?

jannnfe, 2018-09-17:
Am I doing something wrong? I still have not got it yet.

jannnfe, 2018-09-21:
Hello @ascillato 
I found the mistakes. The first thing that got lost was a sign that Tasmota does not like.
It is the quotation mark -> “
The right one is here -> "

Another mistake is that I have "SwitchTopic GaragedoorSensor". Unfortunately, the rule only goes with "SwitchTopic 0". That's a pity because I need Switch1 and Switch2 independent of the relay, because these are only sensors.

Is there any Workaround?

ascillato, 2018-09-21:
>  the rule only goes with "SwitchTopic 0".  I need Switch1 and Switch2 independent of the relay, because these are only sensors.

if you want to do two mqtt publish, you can use backlog and run several commands like for example:

```
rule1 on Switch1#state do backlog publish stat/GaragedoorSensor/switch %value%; publish domoticz/in {"idx":61,"nvalue":%value%,"Battery":100} endon
```

jannnfe, 2018-09-21:
I don't need two MQTT publish (only the domoticz one) but setting SwitchTopic GaragedoorSensor was the only solution for me to not trigger the Relay with the Sensors.
Is there a other way to get the two Sensors independent of the relay?
Because the rule you have send is not working for me:
I set:
"SwitchMode1 1"
"SwitchMode2 1"
"SwitchTopic 0"
and your rule
`rule1 on Switch1#state do backlog publish stat/GaragedoorSensor/switch %value%; publish domoticz/in {"idx":63,"nvalue":%value%,"Battery":100} endon`
`rule2 on Switch2#state do backlog publish stat/GaragedoorSensor/switch %value%; publish domoticz/in {"idx":62,"nvalue":%value%,"Battery":100} endon`

This way the Relay will be triggered

ascillato, 2018-09-21:
If you have:

* set your sensor configured as `switch1` in module configuration.
* switchtopic1 0
* rule1 1
* rule1 on Switch1#state do publish domoticz/in {"idx":63,"nvalue":%value%,"Battery":100} endon

when pushing or toggling your switch1, your relay1 will not toggle.

SwitchtopicX 0 and rules (on switchX#state do ...) will detach the switchX from relayX.

_(the rule is an example, you have to adapt it to your needs)_

More information at: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

If you have further question remember that you can find us in Discord Chat.

jannnfe, 2018-09-21:
Thanks @ascillato 
I was still looking for a bit and I think I know why this does not work for me. I use the Sonoff 1 channel and this seems to be a problem with inputs on RX and TX. Can it be that it is the 1 Channel is switched every time the relay? I use the 1 Channel with Configure module sonoff basic and GPIO1 Serial Out Switch1 and GPIO3 Serial In Switch2 and the settings in your post above.

I found the informations here: https://github.com/arendst/Sonoff-Tasmota/issues/893#issuecomment-387842022