arendst/Tasmota #3808: More than 2 WiFi Networks

iwilkenloh, 2018-09-15:
Hi @all

I don't know, if this is a feature request. My Problem is, that I'm living in an House of 4Levels. Everry Level has is own WLAN. So every Time I program an TasmotaDevice I have to decide if it wll be in EG/OG or in KG/EG. I need more WLANs to use. I don't know if it is a Problem to get more. For me a Number of 4 whould be great.

Ingo

P.S.:: Sorry for my poor English, I hope U understand it! :-)

Frogmore42, 2018-09-15:
It currently only supports 2 different SSIDs, but you can have multiple APs on the same SSID.  Many people use the same SSID for different floors or even locations.  Why do you have different SSIDs for the different floors?

ascillato, 2018-09-16:
Hi,

Sorry, it is not supported by now.

closed 2018-09-16 by ascillato2