arendst/Tasmota #3809: H801: Configuration > Configure Module > GPIO00: 7 - WS2812 (not working)

gusman80, 2018-09-15:
I am using H801 analogue PWM outputs (thanks to tasmota) successfully for quite some time now. 
As there is one user configurable gpio (GPIO00) i tried to configure & use it as ws2812 digital data line.
After reboot no ws2812 special commands seem to work, as i always get {"Command":"Unknown"} as a console log.
Some flags inside tasmota prevent ws2812 being used with H801?

ascillato, 2018-09-16:
Hi,

Please, could you be so kind to complete the troubleshooting template in order to provide us with all the information needed to help you properly.

Thanks.

ascillato, 2018-09-16:
Hi,

Thanks for completing...

From your `status 0`:

```
{"Status":{"Module":20,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:55","StartupUTC":"2018-09-15T23:58:17","Sleep":0,"BootCount":13,"SaveCount":254,"SaveAddress":"F6000"},"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-11T21:36:01","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["toni3",""],"TelePeriod":300,"SetOption":["00018001","55818000","00000000"]},"StatusMEM":{"ProgramSize":506,"Free":496,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B617CE","00000000"]},"StatusNET":{"Hostname":"sonoff-7050","IPAddress":"192.168.2.20","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:4B:DB:8A","Webserver":2,"WifiConfig":5},"StatusTIM":{"UTC":"Sun Sep 16 00:00:12 2018","Local":"Sun Sep 16 01:00:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:28","Sunset":"19:01"},"StatusSNS":{"Time":"2018-09-16T01:00:12"},"StatusSTS":{"Time":"2018-09-16T01:00:12","Uptime":"0T00:01:55","Vcc":3.555,"POWER":"OFF","Dimmer":93,"Color":"00EC000000","HSBColor":"120,100,93","Channel":[0,93,0,0,0],"CT":499,"Scheme":4,"Fade":"OFF","Speed":10,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"****","RSSI":56,"APMac":"****"}}}
```
using _decode-status.py_ we obtain:

```
*** decode-status.py v20180730 by Theo Arends ***
Decoding information for device Sonoff from status report taken at 2018-09-16T01:00:12

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) Show value units in JSON messages
   3 (OFF) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (ON ) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Rules once mode until 5.14.0b
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  50 (OFF) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  MQTT_PUBSUBCLIENT
  USE_ADC_VCC
  USE_ARILUX_RF
  USE_BH1750
  USE_BMP
  USE_DHT
  USE_DISCOVERY
  USE_DOMOTICZ
  USE_DS18x20
  USE_EMULATION
  USE_ENERGY_SENSOR
  USE_HOME_ASSISTANT
  USE_HTU
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LM75AD
  USE_MHZ19
  USE_NOVA_SDS
  USE_PMS5003
  USE_PZEM004T
  USE_RULES
  USE_SENSEAIR
  USE_SERIAL_BRIDGE
  USE_SGP30
  USE_SHT
  USE_SHT3X
  USE_SR04
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_WEBSERVER
  USE_WS2812
  WEBSERVER_ADVERTISE
```

The driver for the **WS2812** is there. So, I don't know what is happening.

mike2nl, 2018-09-16:
15 (ON ) Do not allow PWM control ???

arendst, 2018-09-16:
It's either pwm OR 1 ws2812 string as they both ise the same color/dimmer resources.

gusman80, 2018-09-16:
can i offer you any more log info or tests on the device?

arendst, 2018-09-16:
It's not supported

closed 2018-09-16 by arendst

gusman80, 2018-09-16:
thx, last question: can i chose PWM or ws2812 at some point ? (linking/compilation)?