arendst/Tasmota #3810: Support for EXS RELAY V5.0

cheise, 2018-09-16:
Your Firmware Support the EXS RELAY V3.1, i use this Device configuration for the EXS Relay V5.0, but i can use only first Relay on the V5.0 Relay, LED1 not working, too. Can you support the second Relay and LED1 in your next Releases, please ?

cheise, 2018-09-16:
The V5.0 Relay have this Pinouts:

     GPIO_USER,        // GPIO00 Module Pin 4
     GPIO_USER,        // GPIO01 Module Pin 3
     0,                        // GPIO02
     0,                        // GPIO03
     GPIO_USER,        // GPIO04 Module Pin 2
     GPIO_USER,        // GPIO05 Module Pin 1
     0, 0, 0, 0, 0, 0,     // Flash connection
     GPIO_REL1,        // GPIO12 Relay1 ( 1 = Off)
     GPIO_REL2,        // GPIO13 Relay1 ( 1 = On)
     GPIO_REL3,        // GPIO14 Relay2 ( 2 = Off)
     GPIO_LED1,       // GPIO15 LED1
     GPIO_REL4,       // GPIO16 Relay2 ( 2 = On)
     0

Frogmore42, 2018-09-17:
You can copy the template for the v3.1 and add the extras for v5, by using the generic module type.  It looks like the v3.1 uses relay 1 and relay 2 for the single relay, so you will need to use relay 3 and 4 for the second one.  You will have to update both relays in Tasmota to get the real relay to turn on.  Perhaps there is a rule that can make it easier, so you only have to turn on one and the rule will take care of the other.

cheise, 2018-09-17:
I have do that bevor over the generic profil. I can‘t toggle the second relay. Than i look in the sonoff.ino and determine only the first relay is hardcoded by the Firmware for „latching relay“ use ( void setlatchingrelay) when you take the EXS Relay profil. My expertise to patch this issue are very bad. :(

Frogmore42, 2018-09-17:
Try set option 14
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
That might work.

cheise, 2018-09-19:
With "SetOption14=%20On" i can use both Relays with interlock Mode. I have 4 virtual Buttons. For interacting with the Relays i must use the Buttons pairwise (1-2  = Relay 1 and 3-4 = Relay 2), thats not what i need. With the "ESX Relay Profil" i can use the Relay 1 with 1 Button in Toggle Mode. Thats what i need for Relay 2, too, 1 Button for Relay 2 in Toggle Mode. You have any ideas?

Frogmore42, 2018-09-20:
You are correct that there is code that special cases the ESX.  If you use VS Code with PlatformIO it has a search capability which you could use to search the code to find it and duplicate it for the second relay.  Or, you could probably write a rule that would do something similar.

cheise, 2018-09-21:
Ok, i can wait, maybe, i hope the device becomes support with the next release. My expertise to patch the Firmware is very bad, thanks for your support.

cheise, 2018-09-23:
Here is picture from the Board with Pinouts:
[image]

arendst, 2018-09-29:
BTW do you have a link to this device? I cannot find it on ex-store.de

cheise, 2018-09-29:
Here is the link:

https://ex-store.de/2-Kanal-WiFi-WLan-Relay-V5-Blackline-fuer-Unterputzmontage

arendst, 2018-09-30:
In order to use EXS Relay V5.0 select the EXS Relay(s) module and add the two additional relays as used by the EXS Relay V5.0:

[image]

arendst, 2018-09-30:
Pls let me know if this works.

closed 2018-09-30 by ascillato2

cheise, 2018-09-30:
Everything is functional now, but two things are wrong. The output pins are reversed on the PCB. That's my mistake. You must change relay 1 into relay 2 and and the other way with relay 2, too. Second thing is, the state from relay is inverted. I must use 31=Relay3i and 32=Relay4i, see the picture below.....

[image]

arendst, 2018-09-30:
If relay1 and 2 need to be reversed than the v3.1 version won't work anymore.

Is anyone still using v3.1 as it doesn't appear on the sellers website anymore?

cheise, 2018-09-30:
> Is anyone still using v3.1 as it doesn't appear on the sellers website anymore?

You can leave it like that. Only the outputs on the PCB are reversed. No problem. For Relay 2 ( Relays 3+4 latching state) is the "i" option the right thing ;). The V3.1 are disappear, i think they sell only the v5.0 now.

arendst, 2018-10-01:
@cheise would you be so kind to add this info to the wiki to make sure users know how to add the second relay.

cheise, 2018-10-01:
@arendst 
I have add a Info in the wiki under "other devices" ;)