arendst/Tasmota #3812: Idea: Second MQTT Provider

HarryHase11, 2018-09-16:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

HarryHase11, 2018-09-16:
Is it possible to get connected to a second MQTT broker, so that the devices ca listend on two broker at the same time?

curzon01, 2018-09-16:
If you need to handle with topics/payloads of two MQTT broker use a bridge between both and connect Tasmota to only one as is. Search for "MQTT bridge" with your search engine of your choice

HarryHase11, 2018-09-16:
MQTT bridging did not solve my issue, I know that's currently not possible!  Maybe feature request?!

HarryHase11, 2018-09-16:
The benefit, 
one is to use it as backup broker,
simultan two broker, one inside the house, the other outside,
third no change to actual solution
... maybe more
---
nearly same as secondary Accesspoint.
More benefits then negativ aspects.

ascillato, 2018-09-16:
Hi duplicated issue:
#2937