arendst/Tasmota #3814: Module POW - Power thresholds usage

Pewibe, 2018-09-16:
Make sure these boxes are checked [x] before submitting your issue - Thank you!
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/.../wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Hardware used : Sonoff POW R2
- [ ] Provide the output of command ``status 0`` : not necessary - V6.2.0

My problem is to understand the purpose of the commands "PowerLow" / "PowerHigh". I set them as example both to 3 watts to see if an external device is running (consuming more than 3 watts) or is off (less than 3 watts). The console shows the line
`15:14:08 RSL: MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}`
if the transition appears. I'm convinced that there must be more than showing this console line. 
Thanks in advance for filling up my mind gap!

ascillato2, 2018-09-17:
Hi,

Please read the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-pow-sonoff-s31-and-pzem004t

Thanks.

closed 2018-09-17 by ascillato2

Pewibe, 2018-09-17:
@ascillato2 thanks for your time to answer, but a quick answer like this doesn't really help. I have read the whole doc you are poiting to, and this chapter especially. I used the search command of the browser and got "margin" no hits and "powerlow" 3 hits. An explanation like "PowerLow - Set power low threshold value" helps nothing to understand for what purpose to use it. A sentence like "I have no time to update / extend the wiki" or "no time to answer" I would understand.
And let me touch a second item: It's not a good behaviour to close an issue without giving the initiating  party the chance to accept. A timeout of 3..5 days is a practical approach to close an issue without asking. I hope you agree.

meingraham, 2018-09-17:
I had been following this issue because I recently put an S31 to use as an appliance monitor. I wanted to read current issues to see if I needed to be aware of any problems or ran across any tips.

In any case, in setting up my scenario, I ran across this: https://community.openhab.org/t/washing-machine-state-machine/15587/80

The way I read this, PowerLow/PowerHigh/PowerDelta are trigger thresholds for TASMOTA to report power usage for the device "out of TelePeriod" cycle. Perhaps those parameters are used for other purposes, but for my setup, that's how I make use of them.

Without looking through the source code, I'm not sure I've seen an explanation of what they are used for outside of this reference.

Regards.

Mike

ascillato, 2018-09-17:
Hi,

@Pewibe 

> a quick answer like this doesn't really help

Sorry, most of the time a short quick answer helps a lot. May be in this case it didn't. That's ok. no problem with that. The issues are not being locked, so anyone can continue posting. An Open Issue means being tracked actively for follow up.

>  I have read the whole doc you are poiting to, and this chapter especially.

Yes, the wiki still needs improvements but it is open to everyone to contribute to it. The wiki is being made by the people of this community.

> An explanation like "PowerLow - Set power low threshold value" helps nothing to understand for what purpose to use it.

Ok, should be: **PowerLow - Set power low threshold value to send extra MQTT telemetry.**

> A sentence like "I have no time to update / extend the wiki" or "no time to answer" I would understand

Sorry, That would be very useless. In that case it is better is not to answer anything.

> It's not a good behaviour to close an issue without giving the initiating party the chance to accept. A timeout of 3..5 days is a practical approach to close an issue without asking. I hope you agree.

Please, do not take it personal. It is just being practical. 1/3 of issues are being left behind from its initiators (not your case). On January we had 1248 open issues. That was impossible to follow up.
So, now, we are doing more less the following:

* **General Question:** Answer and Close.
* **Troubleshooting:** Answer and try to solve the issue in order to enhance the user experience with Tasmota (Traking to solve and with 3 to 5 days timeout, the same you suggested and if not the initiator can continue, then goes to Hold On label)
* **Feature Request:** If that is in the actual Scope of Tasmota (readme) it will be added Enhancement label and anyone who want can provide a PR or solution. These ones are not being closed by the Stale BOT.
* **Bug Report:** Answer and try to solve the issue in order to enhance Tasmota. These ones are not being closed by the Stale BOT.

Hope you don't take it personal.

General questions are better to be addressed in the Tasmota Chat or in the Forum and leave Github issues for software issues (bugs, enhancements). And remember that all this is being supported in our free time. So, not everything posted will be answered and not everything posted will be solved/implemented. Anyone can answer and can make pull requests.

meingraham, 2018-09-17:
I went ahead and made edits to the wiki.

ascillato, 2018-09-17:
Great! Thanks a lot!

charettepa, 2020-02-03:
This still does not explain if you need to be below or above the threshold.
one would think that low threshold is below and high is above.
ie if you are below low or above high updates will be continuous but not in between.
however its not clear.

further explanation would be appreciated.