arendst/Tasmota #3815: Tasmota and TPlink kasa making tasmota crash/reboot

Chipz0r, 2018-09-16:
**Describe the bug**
Hi
I have a number of TP link HS101 homeplugs that use the Kasa software to control them, i am getting something strange , when i am in the kasa software and i pull down to refresh the units ALL my tasmota units crash/reboot.
I thought it was maybe something to do with alexa so today i flashed a fresh WemosD1 with tasmota and started to test and this reboots as well when refreshing in the KASA software, I do NOT have homeplug emulation enabled , the only things i setup was the wifi and a mqtt topic.
I have included output from serial at the time of the crash/reboot.

If you do not have access to tp link kasa i am willing to help you in any way i can .

Many thanks

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : Wemos D1 Mini
- [x ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```
15:47:19 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2","Sonoff3"],"Topic":"test1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:47:19 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:06:52","StartupUTC":"2018-09-16T14:40:27","Sleep":0,"BootCount":13,"SaveCount":30,"SaveAddress":"F6000"}}
15:47:19 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-16T15:13:36","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:47:19 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["hax0r5",""],"TelePeriod":300,"SetOption":["00008001","55818000","00000000"]}}
15:47:19 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","240003A0","23B617CE","000002C0"]}}
15:47:19 RSL: STATUS5 = {"StatusNET":{"Hostname":"test1-5457","IPAddress":"192.168.3.42","Gateway":"192.168.2.1","Subnetmask":"255.255.254.0","DNSServer":"8.8.8.8","Mac":"80:7D:3A:5D:35:51","Webserver":2,"WifiConfig":5}}
15:47:19 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 16 14:47:19 2018","Local":"Sun Sep 16 15:47:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:28","Sunset":"19:01"}}
15:47:19 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-09-16T15:47:19"}}
15:47:19 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-09-16T15:47:19","Uptime":"0T00:06:52","Vcc":2.748,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"hax0r5","BSSId":"00:50:7F:E2:AD:5C","Channel":3,"RSSI":28}}}

**To Reproduce**
You would need tp link software to recreate i think

**Expected behavior**
Tasmota not to crash/reboot on refresh

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
I "presume" that tp link is sending some sort of broadcast over the network to get a reply from the homeplugs , i "presume" that tasmota is also seeing this broadcast and not knowing what to do with it , it crashes

Here is serial output

16T15:36:42","Uptime":"0T00:00:14","Vcc":2.748,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"hax0r5","BSSId":"00:50:7F:E2:AD:5C","Channel":3,"RSSI":50}}

Exception (0):
epc1=0xe7612b3c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  00304020 00100000 ffff0000 e730ffff  
3ffffeb0:  52dc4020 a39e3fff 00003ffe e8a40000  
3ffffec0:  ffff3ffe 00007fff de000000 e1b83ffe  
3ffffed0:  00053ffe 00000000 00010000 a39e0000  
3ffffee0:  00003ffe 00010000 00300000 00050000  
3ffffef0:  00080000 03960000 00054010 00000000  
3fffff00:  00200000 08360000 a7a54010 00053ffe  
3fffff10:  00000000 00200000 08360000 00050010  
3fffff20:  00000000 00200000 08360000 a7a54010  
3fffff30:  3b1b3ffe e2304010 00163ffe 3fff5d00  
3fffff40:  3ffec9a6 00000000 3fff52dc 40246f39  
3fffff50:  d303a8c0 00000062 00000000 00000017  
3fffff60:  00000002 00000018 40235eff 3ffee230  
3fffff70:  3ffec980 3fffdcc0 3ffea898 3ffea898  
3fffff80:  40235e72 3ffee230 00000000 3fff5294  
3fffff90:  3fffdc80 00000000 3fff52dc 4023e34f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff35f0
~ld

00:00:00 Project sonoff Sonoff (Topic test1, Fallback DVES_5D3551, GroupTopic sonoffs) Version 6.2.1.3-2_3_0
00:00:00 WIF: Connecting to AP1 hax0r5 in mode 11N as test1-5457...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on test1-5457.local with IP address 192.168.3.42
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/test1/LWT = Online (retained)
00:00:05 MQT: cmnd/test1/POWER = 
00:00:05 MQT: tele/test1/INFO1 = {"Module":"Generic","Version":"6.2.1.3","FallbackTopic":"DVES_5D3551","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/test1/INFO2 = {"WebServerMode":"Admin","Hostname":"test1-5457","IPAddress":"192.168.3.42"}
00:00:05 MQT: tele/test1/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0xe7612b3c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

15:40:07 RSL: STATE = {"Time":"2018-09-16T15:40:07","Uptime":"0T00:00:14","Vcc":2.748,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"hax0r5","BSSId":"00:50:7F:E2:AD:5C","Channel":3,"RSSI":52}}

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3fff35f0
~ld

00:00:00 Project sonoff Sonoff (Topic test1, Fallback DVES_5D3551, GroupTopic sonoffs) Version 6.2.1.3-2_3_0
00:00:00 WIF: Connecting to AP1 hax0r5 in mode 11N as test1-5457...
00:00:09 WIF: Connect failed with AP incorrect password
00:00:10 WIF: Connect failed with AP incorrect password
00:00:11 WIF: Connect failed with AP incorrect password
00:00:11 RSL: INFO1 = {"Module":"Generic","Version":"6.2.1.3","FallbackTopic":"DVES_5D3551","GroupTopic":"sonoffs"}
00:00:11 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"test1-5457","IPAddress":"192.168.3.42"}
00:00:11 RSL: INFO3 = {"RestartReason":"Hardware Watchdog"}
00:00:11 RSL: RESULT = {"POWER1":"ON"}
00:00:11 RSL: POWER1 = ON
00:00:11 RSL: RESULT = {"POWER2":"OFF"}
00:00:11 RSL: POWER2 = OFF
00:00:11 RSL: RESULT = {"POWER3":"OFF"}
00:00:11 RSL: POWER3 = OFF
00:00:12 WIF: Connected
00:00:12 DNS: Initialized

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-09-17:
Hi,

Seems that your firmware has something wrong.

Try to recompile but using esplib v2.4.2 instead of 2.3.0.

For more support please see us in Tasmota Chat.

Thanks

closed 2018-09-17 by ascillato2

ascillato, 2018-09-17:
From Tasmota Chat.

At startup, when Tasmota search for MQTT server using mDNS, it receives data and crash with exception.

As workaround, disabling mDNS in user_config.h file prevents this issue.

ciB89, 2018-10-08:
Experiencing the same issue. Although for me it seems that only one Tasomota device is restarting while trying to add the plug. Adding the device in Kasa also fails which gives me the feeling that it tries to add the Tasmota and not the TP-Link plug.

xnaron, 2018-12-01:
I also have the same issue. I'll try disabling mDNS as suggested.

FryGuy1981, 2019-06-03:
Having the same problem. Using Tasmota on S31 plugs. Looks like the latest versions (setoption55) mDNS is off by default. Anyone resolve this issue or is it still happening?