arendst/Tasmota #3817: Sonoff Fails to go into AP mode or connect to WPA2 PSK AP

Paulf007, 2018-09-16:
**Describe the bug**
Sonoff-Tasmota-6.2.1 - fails to connect to Wifi Network and AP mode does not start. I then reverted back to Sonoff-Tasmota-5.12.0 and all worked as expected. Uploaded 6.2.1 to 3 x SonOffs basic with the same results. 
I also upgraded my AP to a new'ish Edimax and the SonOff Basic refuses to connect to it even it if has a signal strenth of almost 98% It Only diference between the old AP and the new AP is that it is running 
WPA2 PSK at channel 8 (Saw a previous note on it) vs the old AP
WPA PSK is there a fix to this?

Best regards

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : Sonoff Basic

```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Uploaded the new firmware to 3 x Sonoff Basic 
None of my current Sonoff's want to connect to the new AP

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-09-17:
Please fill out the requesting information (in particular the results of status 0, if you don't know how to do that, look at the wiki for instructions on how to do it).  Without that we would only be guessing and that is likely to waste everyone's time.

Most people are using WPA2 PSK (aka WPA-Personal), which is the current standard for security in Wi-Fi.  WPA3 is not likely to be generally available for some time.  https://www.howtogeek.com/339765/what-is-wpa3-and-when-will-i-get-it-on-my-wi-fi/

The esp8266 libraries support WPA2-PSK and so does Tasmota.

However, Sonoff Basic devices seem to sometimes have trouble with Wi-Fi.  With Wi-Fi (and RF in general) more signal is not always better.

ascillato2, 2018-09-17:
Hi, seems that you have a problem with your firmware. Try to download again from release or download from http://thehackbox.org/tasmota/

closed 2018-09-17 by ascillato2

Josov, 2018-09-17:
I have the same problem, can't get my Sonoff's connected to the WiFi  using WPA2 PSK  (tried with 3).
Back to release 6.1.1: no trouble, connecting immediately. Downloading a pre-compiled version, flashing it does not do the job either.

thenameiwantedwastaken, 2018-09-17:
I can conform that this is a problem. For me it always occurs while on 240v, but only sometimes occurs while connected to 3.3v via the pin headers. I cannot show Status 0 at times when the problem is occurring as there is no way to access the console. Status 0 while connected to 3.3v is below.

After powering-up the Sonoff via 240v for a short while, my router log reports multiple connection attempts:
```
18.09.18 08:59:47 New wireless device registered itself (2,4 GHz), 72 Mbit/s, Snf11, IP 192.168.178.62, MAC 80:7D:3A:18:30:FA.
18.09.18 08:59:33 Wireless device registered (2,4 GHz), 72 Mbit/s, Snf11, IP 192.168.178.62, MAC 80:7D:3A:18:30:FA.
18.09.18 08:59:31 Wireless LAN registration failed(2,4 GHz): Authentication error. MAC address: 80:7D:3A:18:30:FA.
18.09.18 08:59:17 New wireless device registered itself (2,4 GHz), 72 Mbit/s, Snf11, IP 192.168.178.62, MAC 80:7D:3A:18:30:FA.
18.09.18 08:58:49 Wireless device registered (2,4 GHz), 72 Mbit/s, Snf11, IP 192.168.178.62, MAC 80:7D:3A:18:30:FA.
```
... and so on...

I can confirm that there are no IP conflicts or anything like that. It works on 3.3v with the same IP address. I can also confirm that there is no password problem - again it works on 3.3v with no changes.

Status 0 while connected to 3.3v is...

```
00:01:46 CMD: Status 0
00:01:46 RSL: stat/Snf11/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"Snf11","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":
1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:01:46 RSL: stat/Snf11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","Restar
tReason":"Software/System restart","Uptime":"0T00:01:46","StartupUTC":"","Sleep":0,"BootCount":382,"SaveCount":429,"SaveAddress":"F7000"}}
00:01:46 RSL: stat/Snf11/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018-09-18T09:37:20","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"
}}
00:01:46 RSL: stat/Snf11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"Se
tOption":["00008009","55818000","00000000"]}}
00:01:46 RSL: stat/Snf11/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":
["00000809","0FDAE794","240003A1","23B617CE","000002C0"]}}
00:01:46 RSL: stat/Snf11/STATUS5 = {"StatusNET":{"Hostname":"Snf11-4346","IPAddress":"0.0.0.0","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNS
Server":"192.168.178.1","Mac":"80:7D:3A:18:30:FA","Webserver":2,"WifiConfig":3}}
00:01:46 RSL: stat/Snf11/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi.xxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1830F
A","MqttUser":"xxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:01:46 RSL: stat/Snf11/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:46 1970","Local":"Thu Jan 01 00:01:46 1970","StartDST":"Thu Jan 01 00:00:00 1970"
,"EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:01:46 RSL: stat/Snf11/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:46"}}
00:01:46 RSL: stat/Snf11/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:46","Uptime":"0T00:01:46","Vcc":2.813,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"44:D9:E7:FD:75:C0","Channel":1,"RSSI":100}}}
```

This is occurring with three different Sonoff Basic boards. They are all flashed with the same firmware which I have compiled myself with PlatformIO. I have tried restoring one to firmware 6.1.1 and it works fine.

Frogmore42, 2018-09-18:
Many people have reported issues with Sonoff Basic devices.  Your results indicate that the power supply of the Basic really isn't up to the task of work with the latest code.  You are using core 2.3.0.  You might try it with core 2.4.2 to see if that makes a difference.  You should also try completely erasing the flash and reflashing it, if you have not already done that.  If none of that works, you are probably stuck using the older version.  I am using old versions on most of my many devices, so that isn't as bad of a solution as it might seem.

Frogmore42, 2018-09-18:
I will say, you status 0 log indicates it is not connected to MQTT, so it might not be completely working on 3v3 either.

Josov, 2018-09-18:
I completely erased the sonoff using esptool 2.5.0:
[image]

I flashed firmware version 6.2 using esptool:
[image]

It did not work out well, could not get wifi connected, neither through providing power through the pin header nor using the 240V. However I see the sonoff appearing in the wifi settings on Android phone and on my iPad. But cannot manage to get a connection. It looks like the wifi signal from the sonoff is coming an going. The green led is flashing.

I erased the flash another time and downloaded the sonoff-NL.bin from Release 6.2. The result is the same as with my own compiled version. It is not possible to connect although it looks like the sonoff is broadcasting the SSID. The iPad show the message "Unable to join the network "sonoff-2495""

With release 6.1.1 I do not face problems.

Frogmore42, 2018-09-18:
What did the log show on the console (serial since WiFi is not connected)?

Josov, 2018-09-18:
[image]

Josov, 2018-09-18:
I just compiled a new 6.2 version and increased the debug info level to the maximum. First of all the Sonoff did not connect to the wifi network. I checked the ssid and password, both are correct. The password is a simple string of only Alpha's. Here is the result:

00:01:31 APP: DrukKnop1 meervoudig 1
00:01:32 APP: DrukKnop1 meervoudig 2
00:01:32 APP: DrukKnop1 meervoudig 3
00:01:32 APP: DrukKnop1 meervoudig 4
00:01:33 SRC: Button
00:01:33 RSL: Ontvangen topic /WifiConfig, Data lengte 1, Data 2
00:01:33 RSL: Groep 0, Index 1, Opdracht WIFICONFIG, Data 2
00:01:33 RSL: stat/home/office/tasmotatest/RESULT = {"WifiConfig":"WifiManager selected"}

Exception (0):
epc1=0x4020e608 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3fff4640 end: 3fff4a50 offset: 01a0

>>>stack>>>
3fff47e0:  05a0fd27 00000004 0000001d 0200bce3
3fff47f0:  05a0fd44 00000000 00000000 40102322
3fff4800:  ffffffff 3fffc6fc 00000001 3ff20a00
3fff4810:  00000001 40101989 00000104 7fffffff
3fff4820:  3ffeb28c 3ffeb28c 00000008 7fffffff
3fff4830:  3fff4840 00000064 00000000 4000050c
3fff4840:  3fffc278 401018b4 3fffc200 00000022
3fff4850:  3fff4860 00000001 3ffec56c 4020e610
3fff4860:  400043d5 00000030 00000018 ffffffff
3fff4870:  400044ab 3fffc718 3fff4960 08000000
3fff4880:  60000200 08000000 00000003 00000000
3fff4890:  0000ffff 3fffc6fc 00000001 000fa000
3fff48a0:  3fff29f8 3fff3940 00000000 00000030
3fff48b0:  4021168c 00000030 00000013 ffffffff
3fff48c0:  4021c14c 00002eec 000009ad 00000000
3fff48d0:  00000e00 3fff29f8 00000000 0000ffff
3fff48e0:  ffffffff 3fffc6fc 00000001 3fff28c4
3fff48f0:  3fff2892 3fff29f8 00000000 00000030
3fff4900:  40106a21 00000030 0000000f ffffffff
3fff4910:  40106e7c 00002e3e 00001f40 00000003
3fff4920:  00000005 00001b0c 00000000 fffffffe
3fff4930:  ffffffff 3fffc6fc 00000001 00000000
3fff4940:  00000000 3fffdad0 3fff3a1c 00000030
3fff4950:  ffffffff 3fffc6fc 00000001 00000000
3fff4960:  00000002 4000410f 3fff3a1c 00000030
3fff4970:  3fffc718 40004a3c 000000fa 3fff29f8
3fff4980:  3fffc718 401047d2 000000fa 00000030
3fff4990:  4020acb0 000000fa 4020acb9 00000030
3fff49a0:  3fff28bc 3fff2bf8 3fff0e90 4021c1b3
3fff49b0:  00000e00 3fff2885 3fff0a24 401072e9
3fff49c0:  3fff6084 3fff2884 3fff28c4 00000001
3fff49d0:  3fff29f8 3fff2892 3fff28c4 4022c050
3fff49e0:  3fff2730 0000002c 00000000 3fff3a1c
3fff49f0:  00000002 0000002c 00000004 4022f650
3fff4a00:  3ffe85c4 3fff3a1c 000172f7 40212743
3fff4a10:  3fff28e8 0000002c 000000fa 3fff3a1c
3fff4a20:  3fffdad0 00000000 3fff291c 4022c1b2
3fff4a30:  3fffdad0 00000000 3fff3a15 4020c900
3fff4a40:  feefeffe feefeffe 3fff3a30 4020e8b4
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v3fff3808
~ld

00:00:00 APP: Herstarts 59
00:00:00 SRC: Restart
[... 85 lines cut ...]
00:00:39 WIF: Controleer verbinding...
00:00:39 WIF: Verbinden...
00:00:40 WIF: Controleer verbinding...
00:00:40 WIF: Verbinden...
00:00:41 WIF: Controleer verbinding...
00:00:41 WIF: Verbinden...
00:00:42 WIF: Controleer verbinding...
00:00:42 WIF: Verbinden...
00:00:43 WIF: Controleer verbinding...
00:00:43 WIF: Verbinden...
00:00:44 WIF: Controleer verbinding...
00:00:44 WIF: Verbinden...
00:00:45 WIF: Controleer verbinding...
00:00:45 WIF: Verbinden...
00:00:46 WIF: Controleer verbinding...
00:00:46 WIF: Verbinden...
00:00:47 WIF: Controleer verbinding...
00:00:47 WIF: Verbinden...
00:00:48 WIF: Controleer verbinding...
00:00:48 WIF: Verbinden...
00:00:49 WIF: Controleer verbinding...
00:00:49 WIF: Verbinden...
00:00:50 WIF: Controleer verbinding...
00:00:50 WIF: Verbinding mislukt door AP time-out
00:00:50 WIF: SmartConfig 3 minuten actief