arendst/Tasmota #3819: Sonoff Dual R2 keep booting in programming mode

errik97, 2018-09-17:
Hi, as described in the title i have one tasmotized sonoff dual R2 that keep booting in programming mode even if i don't press gpio0 and gnd simultaneously. 

First of all, i have this device since some months and everything was working fine with tasmota (flashed with arduino IDE), starting from 5.14 version to 6.2.1 (i always update the device). One day we had a blackout but when the power came back the sonoff looked dead. I unmounted it from the rolling shutter, tryed to power it up with 3.3 volt but nothing happend, so i decided to open up Arduino IDE and i tryed to flash the lastest tasmota version and... It was flashing! Without first pressing gnd and gpio0. Once the flash was finished the device started and worked fine, but at the first restart... nothing! So, everytime i put some voltage on it, it goes in programming mode. With nothing on it pressing gnd + gpio0. 

Video of the issue: https://www.dropbox.com/s/d969oiltjxju0lk/sonoff.mp4?dl=0

Anyone know how i can solve this? Even soldering something on it, if necessary.

Thanks to anyone will help me.

Frogmore42, 2018-09-17:
While I don't doubt what you say, the only way an esp8266 is supposed to go into bootloader mode is when gnd is applied to GPIO 0.  That means one of the following:
1. There really is gnd connected to GPIO 0
1. The esp8266 is defective

The first issue can be verified by checking with an ohm meter.  The 2nd can be inferred by eliminating the 1st.

Since you had a power failure, it is possible the esp8266 (or some other part of the circuit) got damaged during the outage (or restoration) of power.  I have had a couple of devices fail on me so it is not like it never happens.

If it happened to me, I would just replace the device.  They are very cheap.  I had a THxx fail on me that had been working fine.  While I probably could have fixed it by replacing some component.  I decided it was better to just replace the whole device.

On the other I hand, I just discovered that a NodeMCU device that I thought was defective, is perhaps less defective than I thought it was.  For the past few months it has been unable to stay running for more than an hour or two, sometimes no more than a few mins (but it was still functioning well enough to be useful).  Yesterday I noticed it was running core 2.4.0.  At the time I flashed it, core 2.4.0 was not know to be bad.  So I flashed it with the latest with core 2.4.2 and lwip2.  So far it has been up for over 23 hours.

pszafer, 2018-09-17:
That's weird, same thing with my Sonoff Dual R2. I didn't notice any blackout or problems, just updated it, first time it started to switching on by itself and after restart it is turning on with programming mode on.
There is no connection between Button0 and GND, so defect of esp8266?

ascillato2, 2018-09-17:
> here is no connection between Button0 and GND, so defect of esp8266?

Seems to be. Please, try to flash again.

closed 2018-09-17 by ascillato2

pszafer, 2018-09-17:
Tried all of things you advices in different topics:
- erased flash,
- flash again,

After reboot is always going into programming mode.

ascillato, 2018-09-17:
If you don't have GND to ESP8266 GPIO0 then that I/O port is burned. Sorry.

errik97, 2018-09-17:
> 
> 
> Tried all of things you advices in different topics:
> 
>     * erased flash,
> 
>     * flash again,
> 
> 
> After reboot is always going into programming mode.

Same for me... So there is nothing we can do? maybe replacing the esp8285 on the board?