arendst/Tasmota #3820: 6.2 smarthings

xHirscHx, 2018-09-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

xHirscHx, 2018-09-17:
Im running smarthings hubs and about 20 Sonoffs with 6.1 Hue Bridge all working great thought i would try to upgrade to 6.2 . Now the 2 i upgraded arent  taking to the smart things hub, still works with alexa but my automatons wont( motion turning on the sonoff t1)or (t3 mimicking other lights) work. st new app and old. any ideas. I killed the bathroom and kitchen (wife not happy)
Help
thx

Frogmore42, 2018-09-17:
Start with filling out the information from the template, otherwise everyone will have to guess exactly what is wrong.

ascillato2, 2018-09-17:
Hi,

A lot of information is missing. Please, when you have time, complete all information requested in order to properly help you, Thanks.

closed 2018-09-17 by ascillato2

xHirscHx, 2018-09-17:
Sorry first post