arendst/Tasmota #3821: Sonoff-Tasmota Error compiling for board ESP32 Dev Module

dhernandezgt, 2018-09-17:
**Describe the bug**
Trying to use Tasmota on ESP23 Dev module it display this error `Sonoff-Tasmota-6.2.1\sonoff\sonoff.ino:29:143: fatal error: core_version.h: No such file or directory` with Arduino IDE

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ X] _Searched the problem in issues and in the wiki.  
- [ X] _Hardware used_ :  ESP32 Dev Module kit
- [ ] _Provide the output of command_``status 0`` :
```
C:\Users\danni\Documents\Sonoff-Tasmota-6.2.1\sonoff\sonoff.ino:29:143: fatal error: core_version.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.
```

**To Reproduce**
_Steps to reproduce the behavior:_
Open the sonoff.ino file. It will open all the necessary files in next tabs in Arduino IDE. Now open the userconfig.h tab to enter our Wi-Fi Credentials.
Edit the SSID and Password fields with your WiFi network SSID and Password. Also, give a unique project name by your choice in #define PROJECT “sonoff” line.
Select correct NodeMCU board  
Connect your NodeMCU and choose the correct COM Port and Board Version
**Expected behavior**
_A clear and concise description of what you expected to happen._
flash the ESP board.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

[image]

**Additional context**
_Add any other context about the problem here._

Thanks in advance for your help

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-09-17:
@dhernandezgt I think this issue is misplaced - perhaps it should be at https://github.com/JacekDob/Sonoff-Tasmota

Jason2866, 2018-09-17:
Here is Tasmota (original) and it doesnt support ESP32.
Please post your Issue to the owner of the fork. This is the link @andrethomas  you gave.
Please close this missplaced issue. 
THX

ascillato2, 2018-09-17:
Hi,

Tasmota does not support ESP32.

See https://github.com/JacekDob/Sonoff-Tasmota that jacekDob is porting that.

Thanks

closed 2018-09-17 by ascillato2

atrovato, 2019-12-15:
Hello,
is it planned to support ESP32 ?

meingraham, 2019-12-15:
Unfortunately, no.