arendst/Tasmota #3824: Thermostat rule testing example missing "on"

MagnusThome, 2018-09-17:
The Thermostat rule example on https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#examples is missing an "on"

NOTE: The error is not in the actual thermostat rule but in the test rule just below it, the one for testing the thermostat rule without a temp sensor:

**CURRENT:**
EXAMPLE RULES WITHOUT TEMP SENSOR TO TEST THE THERMOSTAT RULES

rule on system#boot do ruletimer1 70 endon on Switch1#State do event toggling1=%mem1% endon on event#toggling1=0 do mem 1 endon on event#toggling1=1 do mem 0 endon on rules#timer=1 do backlog var1 0; ruletimer1 70; power1 0; endon on event#temp do backlog var1 1; ruletimer1 70; event ctrl_ready=1; event temp_demand=%value%; endon event#ctrl_ready>%mem1% do var1 0 endon on event#temp_demand>%mem2% do power1 0 endon on event#temp_demand<%mem3% do power1 %var1% endon

**SHOULD BE:**
EXAMPLE RULES WITHOUT TEMP SENSOR TO TEST THE THERMOSTAT RULES

rule on system#boot do ruletimer1 70 endon on Switch1#State do event toggling1=%mem1% endon on event#toggling1=0 do mem 1 endon on event#toggling1=1 do mem 0 endon on rules#timer=1 do backlog var1 0; ruletimer1 70; power1 0; endon on event#temp do backlog var1 1; ruletimer1 70; event ctrl_ready=1; event temp_demand=%value%; endon **on** event#ctrl_ready>%mem1% do var1 0 endon on event#temp_demand>%mem2% do power1 0 endon on event#temp_demand<%mem3% do power1 %var1% endon

MagnusThome, 2018-09-17:
I suspect there is one too many "backlog" in the actual rules in the example too. But I guess that doesn't break anything?

**RULES:**
rule1 on system#boot **backlog** do ruletimer1 70 endon on Switch1#State do event toggling1=%mem1% endon on event#toggling1=0 do mem 1 endon on event#toggling1=1 do mem 0 endon on rules#timer=1 do backlog var1 0; ruletimer1 70; power1 0; endon on tele-SI7021#temperature do backlog var1 1; ruletimer1 70; event ctrl_ready=1; event temp_demand=%value%; endon on event#ctrl_ready>%mem1% do var1 0 endon on event#temp_demand>%mem2% do power1 0 endon on event#temp_demand<%mem3% do power1 %var1% endon

curzon01, 2018-09-17:
> **RULES:**
> rule1 on system#boot **backlog** do ruletimer1 70...

should be
rule1 on system#boot do **backlog** ruletimer1 70...

MagnusThome, 2018-09-17:
When just writing "rule" instead of "rule1" it automatically sets it to rule1. I have no idea if that's seen as best practices though?

MagnusThome, 2018-09-17:
Ah, misunderstood. So there are sort of three errors on the page?

- The first "backlog" should not be there at all? 
- But now that it is there it's actually in the wrong place?
- And in the test code there's a missing "do"

So just removing that first occurence of "backlog" in the thermostat code is the proper fix I guess. And adding the "do" in the test example code.

ascillato2, 2018-09-17:
Thanks for pointing this out.

I will correct that soon.

closed 2018-09-17 by ascillato2

ascillato, 2018-09-17:
Corrected. Thanks.

jas8472, 2019-11-04:
I can't get this to work at all the button remains on turning the thing on/off and not toggling the mem1