arendst/Tasmota #3826: Two Sonoff Dual R2 with GPIO broken for the same reason?

errik97, 2018-09-17:
Hi everyone, recently i had one issue with one of my sonoff dual R2 (see #3819 ). Now another one have another problem, but seem that the main reason is the same.

The one that keep booting in programming mode have this problem: if i flash tasmota i can use it till i don't power it off. When i can use it if i apply gnd to "button 0" (that is the GPIO0) the relay won't turn on (of course i configured in "module parameters" it as switch1), but if i apply gnd to "button 1" (that is the GPIO9) the second relay turns on. 

On the other side we have the second broken sonoff dual R2. The strange thing, is that we are in the same situation, but this time the "button 0" work, and the "button 1" doesen't. 

So, i tought: Of course the blackout was the reason of the damge. And it's interesting to note that both sonoffs were configured exactly the same, connected to a rolling shutter motor, and on the GPIO0 and GPIO9 connected to an external interblocked biticino's switch (the keys of this switch doesen't come back up after one press, but they keep the pression until you don't press the other button). Maybe, when the blackout happened, one sonoff had the GPIO9 applyed to one GND and the other one had the GPIO0 applyed to one GND (even with both relays off, because i setted up the pulsetime option). In fact, one have a broken GPIO9 pin and the other one (that keep booting in programming mode) have the GPIO0 broken! 

Now i'm thinking: 
1) if this could happen would it be better to connect an external switch to the GPIO1 and GPIO3 (they are the TX and RX pins), or is better to keep connecting it to the GPIO0 and the GPIO9? 
2) And, there is a way to fix the two broken GPIO? 

The configuration from the console of both sonoff dual (running tasmota 6.2.0) was:
seriallog0
SetOption14 1
PulseTime1 150
PulseTime2 150
SwitchMode1 2
SwitchMode2 2

Thanks to everyone that will help me!

ascillato, 2018-09-17:
1- Yes, but see https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs
2- If your I/O is burnt, there is nothing to do. Sorry.

closed 2018-09-17 by ascillato2

errik97, 2018-09-17:
@ascillato it say 

> Don't use GPIO: 0, 1, 2, 6-11, 15, 16 if you can avoid it. 3 is RX, so best to avoid also. That leaves 4, 5, 12, 13, 14 that don't have constraints. The others can be used, but you have to mind the constraints. 

So, witch GPIO should i use for one external switch?

Thanks for your help