arendst/Tasmota #3827: Blitzwolf SHP2 somtimes reports wrong zero values for factor and power via MQTT

helmutGitHub, 2018-09-17:
With the SHP2 i control my fridge. Sometimes it reports current > 0 (2018-09-16 23:25:33) but the factor and power is reported = 0.

| 2018-09-16 23:24:52 | ENERGY_Current   | 0     |      
| 2018-09-16 23:24:56 | ENERGY_Current   | 1.03  |     
| 2018-09-16 23:25:33 | ENERGY_Current   | 0.982 |      
| 2018-09-16 23:25:34 | ENERGY_Current   | 0.983 |      
| 2018-09-16 23:29:56 | ENERGY_Current   | 1.004 |      
| 2018-09-16 23:34:56 | ENERGY_Current   | 0.967 |      
| 2018-09-16 23:39:56 | ENERGY_Current   | 0.946 |      
| 2018-09-16 23:44:44 | ENERGY_Current   | 0.929 |      
| 2018-09-16 23:44:54 | ENERGY_Current   | 0     |      

| 2018-09-16 23:24:52 | ENERGY_Factor    | 0     |      
| 2018-09-16 23:24:56 | ENERGY_Factor    | 0.62  |      
| 2018-09-16 23:25:33 | ENERGY_Factor    | 0     |      
| 2018-09-16 23:25:34 | ENERGY_Factor    | 0.59  |      
| 2018-09-16 23:29:56 | ENERGY_Factor    | 0.61  |      
| 2018-09-16 23:34:56 | ENERGY_Factor    | 0.58  |      
| 2018-09-16 23:39:56 | ENERGY_Factor    | 0.56  |      
| 2018-09-16 23:44:44 | ENERGY_Factor    | 0.16  |      
| 2018-09-16 23:44:54 | ENERGY_Factor    | 0     |      

| 2018-09-16 23:19:56 | ENERGY_Power     | 0     |      
| 2018-09-16 23:24:51 | ENERGY_Power     | 602.1 |      
| 2018-09-16 23:24:52 | ENERGY_Power     | 245.7 |      
| 2018-09-16 23:24:56 | ENERGY_Power     | 148.7 |      
| 2018-09-16 23:25:33 | ENERGY_Power     | 0     |      
| 2018-09-16 23:25:34 | ENERGY_Power     | 136   |      
| 2018-09-16 23:29:56 | ENERGY_Power     | 143.9 |      
| 2018-09-16 23:34:56 | ENERGY_Power     | 131.3 |      
| 2018-09-16 23:39:56 | ENERGY_Power     | 124   |      
| 2018-09-16 23:44:44 | ENERGY_Power     | 34.1  |      
| 2018-09-16 23:44:54 | ENERGY_Power     | 0     |

ascillato, 2018-09-17:
Hi,

Please, complete the troubleshooting template in order to help you properly.

Also inform us if you have powerdelta command different to zero.

Thanks

ascillato2, 2018-09-17:
Hi, from the time difference from your reports, it seems that you have powerdelta defined different than zero, so, you will have a mqqt message every delta. Power Factor is calculated from the teleperiod time.

closed 2018-09-17 by ascillato2