arendst/Tasmota #3828: WS2812 led strip on wemos

prankousky, 2018-09-17:
Hi all,

I just connected a WS2812 led strip to a Wemos D1 mini like device (see below). It receives 5V from an external power source and is only connected to D8 (GPIO15).

led | other device(s)

VIN | 5V power supply
DIN | nodeMCU GPIO15 / D8
GND | GND power supply

Firmware is Sonoff-Tasmota 6.2.1.3. 

The LED strip is a cut off piece with only 8 led on it. This is an end piece from a longer, fully functioning ws2812 strip. It seems to work (turns on, off, changes colors), just not the way I need it to...

By default, it is off, so I start it by `color 1`. Result:
Led 1 White
Led 2 White
Led 3 White
Led 4 White
Led 5 Red
Led 6 off
Led 7 off
Led 8 off

Next try `color 2`
All led display various colors very quickly, then **all** led turn red. Now, I am aware that I can change the RGB setting to, for example, BRG in user_config.h for something like this. Color1 should be red and color2 should not be red.

However, how come color1 displays different colors while color2 lights up all led? But then color2 will produce a completely different result the next time...

If I type `color1` again, I get a completely different result
Led 1 Green
Led 2 White
Led 3 White
Led 4 White
Led 5 Light Yellow
Led 6 Red
Led 7 off
Led 8 off

`color 000000` works, it turns all led off.

`Scheme<n>` works, but I cannot tell if correctly because I do not know what color output to expect from those schemes.

Would somebody be so kind and explain to me how I can

[for **all** led at the same time]
* set a specific color via mqtt (or console), like `color red` or `color FF0000` (I know how to do this with my regular LED strip connected to an `h801` flashed with tasmota; however, it must be working differently when using a ws2812 or similar)
* have current color fade to new color (for example, if the device is currently set to `color 1` (red), can i send a command to fade to another color rather than just switching from red to blue?

[for some led]
* create an led pattern, for example send `pattern 1`, have led 1-3 turn red, led 4+5 turn blue, led 6-8 turn orange
* create a custom scheme

but most importantly:
have the led strip itself act the way i need it to (like mentioned above: not just *some* led being one/different color/s, but unless explicitly instructed so, always switch **all** led when receiving a command

Development/Compiler/Upload tools used :
Platformio with modified user_config.h
Hardware used : 
nodeMCU v2 (esp8266 development board)
Provide the output of command ``status 0`` :
```
19:39:01 CMD: status 0
19:39:02 MQT: sonoff-6040/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["DVES_%06X","DVES_%06X2"],"Topic":"sonoff-6040","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":1,"PowerRetain":1}}
19:39:02 MQT: sonoff-6040/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://localserver.van/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:20","StartupUTC":"2018-09-17T17:38:42","Sleep":0,"BootCount":33,"SaveCount":155,"SaveAddress":"F9000"}}
19:39:02 MQT: sonoff-6040/stat/STATUS2 = {"StatusFWR":{"Version":"6.2.1.3","BuildDateTime":"2018.09.17 15:10:50","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:39:02 MQT: sonoff-6040/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<my-wifi>",""],"TelePeriod":300,"SetOption":["000080E9","55818000","00000000"]}}
19:39:02 MQT: sonoff-6040/stat/STATUS4 = {"StatusMEM":{"ProgramSize":467,"Free":536,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000407","0F8AE794","240003A0","23B617CE","000002C0"]}}
19:39:02 MQT: sonoff-6040/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6040-6040","IPAddress":"192.168.100.61","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"2C:3A:E8:0E:97:98","Webserver":2,"WifiConfig":5}}
19:39:02 MQT: sonoff-6040/stat/STATUS6 = {"StatusMQT":{"MqttHost":"<mqttip>","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E9798","<username>":"<username>","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:39:02 MQT: sonoff-6040/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 17 17:39:02 2018","Local":"Mon Sep 17 19:39:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"06:54","Sunset":"19:29"}}
19:39:02 MQT: sonoff-6040/stat/STATUS10 = {"StatusSNS":{"Time":"2018-09-17T19:39:02","BME280":{"Temperature":22.1,"Humidity":47.6,"Pressure":1013.1},"TempUnit":"C"}}
19:39:02 MQT: sonoff-6040/stat/STATUS11 = {"StatusSTS":{"Time":"2018-09-17T19:39:02","Uptime":"0T00:00:20","Vcc":2.788,"POWER1":"OFF","POWER2":"OFF","Dimmer":0,"Color":"000000","HSBColor":"0,0,0","Channel":[0,0,0],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"<my-wifi>","BSSId":"8A:8A:20:4C:B4:B8","Channel":11,"RSSI":68}}}
```

Thank you in advance for your help 👍

ascillato, 2018-09-17:
Hi,

Custom schemes are not supported by now.

Please see the ongoing open issue (https://github.com/arendst/Sonoff-Tasmota/issues/3480) about that and the referenced issues in that one.

Thanks.

prankousky, 2018-09-17:
Thank you @ascillato2.

Schemes are not *that* important, as long as I get to control all led (and possibly fade colors).