arendst/Tasmota #3830: Add support i2c 16-Channel PWM Driver PCA9685

dr-apple, 2018-09-17:
**Have you look for this feature in other issues and in the wiki?**
YES

Please add support i2c 16-Channel PWM Driver PCA9685 :-)

ascillato, 2018-09-17:
Hi,

What are you planning to do?

You have 5 PWM channels already with Tasmota.

dr-apple, 2018-09-17:
I have 16 LED Lamps ;-)

andrethomas, 2018-09-18:
Will write driver this weekend ;)

digiblur, 2018-09-18:
Nice!!  Huge thumbs up for this. It will definitely extend the capabilities of the esp8266 given these will be hardware based and not software based PWM.

dr-apple, 2018-09-18:
> Will write driver this weekend ;)

Thank you very much !!

andrethomas, 2018-09-21:
Proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/3866

This adds support for the PCA9685 hardware PWM board which adds 16 x 12bit PWM capable outputs via I2C.

The device resets to 50hz (commonly used by servos) but can be changed with commands as indicated below:

From console, commands:
```
driver15 pwmf,x // Where X is PWM frequency from 24Hz to 1526Hz
driver15 pwm,pin,pwmvalue // Where pin = 0 to 15 and pwmvalue = 0 to 4096
driver15 pwm,pin,ON // Turns pin ON
driver15 pwm,pin,OFF // Turns pin OFF

```
pwmvalue of 0 will turn the pin off completely whereas 4096 will turn the pin on completely.

Settings are not perpetual (yet, perhaps if there is a demand for this we can add it) so will revert to default on reset/power-on.

The enable device the PCA9685 defines need to be uncommented - please remember to use the correct I2C address of your board - if you are unsure, use i2cscan from console to see I2C address. Please do not use the broadcast address of 0x70 as this is not implemented.

andrethomas, 2018-09-21:
For those of you who are curious what it looks like on a logic analyzer, here is a sample after
```
driver15 pwmf,1000 // 1khz
driver15 pwm,0,2048 // ~50% duty cycle
```

[image]

andrethomas, 2018-09-21:
Tested servo on 50hz and works fine

andrethomas, 2018-09-22:
@ascillato2 I think we can close this one since the driver has been merged. If there are further feature requests it may be logged as new issues.

dr-apple, 2018-09-22:
dimming on / off a dimming value in a certain time and fade on "on / off" would be a great feature

digiblur, 2018-09-22:
@andrethomas thanks for your time on this. Definitely going to be picking a few of these chips up in my next order of goodies.

closed 2018-09-22 by ascillato2

andrethomas, 2018-09-22:
Thanks guys, wiki created at https://github.com/arendst/Sonoff-Tasmota/wiki/PCA9685

@dr-apple Your request has been noted and listed as to be implemented in future.

rollercontainer, 2018-09-25:
I appreciate your work @andrethomas, thanks a lot!
The advantage of a 16bit PWM is a smoother fading ability. Maybe the cap to 4096 steps should be expanded to 2^16?
Are you aware of the non-linear dimming curves of the human eye? Adafruit has a good article and fix with a gamma table: https://learn.adafruit.com/led-tricks-gamma-correction/the-quick-fix

rollercontainer, 2018-09-25:
I tried it onece with a normal arduino: https://www.youtube.com/watch?v=TOKxMpAs_OU

andrethomas, 2018-09-25:
The driver is 12bit... so 2^12 is the maximum.
@ascillato2 please ;)