arendst/Tasmota #3833: Tasmota esp32 shows error as **'const char HTTP_HEAD []' redeclared as different kind of symbol**

rulshid, 2018-09-18:
hi, i'm using  tasmota in my esp32(with ardino ide 1.8.7), i have downloaded it from  https://github.com/JacekDob/Sonoff-Tasmota
and moved(copy) all the files inside the /lib into my ardino/library folder but compiling the program shows this error 
and i don't know how to solve the issue  please help me ..

_Originally posted by @rulshid in https://github.com/arendst/Sonoff-Tasmota/issue_comments#issuecomment-422252443_

Frogmore42, 2018-09-18:
Since the issue is with code that you downloaded from a different project, you probably should ask the question there.

Jason2866, 2018-09-18:
@rulshid 
Tasmota does not Support ESP32!
You have to make an issue at this project: https://github.com/JacekDob/Sonoff-Tasmota
Plz close this issue.

closed 2018-09-18 by ascillato2