arendst/Tasmota #3834: Gosund SP111 LED1 only on for Relais1 not

matrix1000, 2018-09-18:
**Have you look for this feature in other issues and in the wiki?**
yes

**Is your feature request related to a problem? Please describe.**
Yes, I try to use the gosund SP111 as a small nightlight with blue color.
But unfortunately the LED1 goes on too. But LED1 should only go on if Relais1 is on.

**Describe the solution you'd like**
Therefore I have modifyied the "BlitzWolf SHP2" section in sonoff_template.h.

`{ "BlitzWolf SHP2",  // BlitzWolf BW-SHP2 (ESP8285 - BL0937 or HJL-01 Energy Monitoring)
     GPIO_LED1_INV,    // GPIO00 Red Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_USER,    // GPIO02 Blue Led (1 = On, 0 = Off)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_HLW_CF,      // GPIO05 BL0937 or HJL-01 CF power
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_HLW_SEL,     // GPIO12 BL0937 or HJL-01 Sel output
     GPIO_KEY1,        // GPIO13 Button
     GPIO_HLW_CF1,     // GPIO14 BL0937 or HJL-01 CF1 voltage / current
     GPIO_REL1,        // GPIO15 Relay (0 = Off, 1 = On)
     0, 0
},`

In the configuration I chaned GPIO02 to PWM1i

Now I can control the blue LED and I can adjust the brightness.

My problem now is, that if I turn on PWM1i the red led (LED1 goes on too).
How can I change this, that LED1 only goes on if I turn on Relay1?
Or can you show me the part of the Code which control LED1 if PWM1 goes on?

Many thanks for your Information.

Frogmore42, 2018-09-18:
Try changing it to PWM2 in the module definition.  That should use a different channel and likely not get confused.

matrix1000, 2018-09-18:
Ok,

now I can't use GPIO02 any longer, the LED does not turn on or off. But the Red (GPIO00) goes on and off.

Firmware Version:  6.2.1.4

Frogmore42, 2018-09-18:
You probably have it misconfigured or are using the wrong command.

Show the output from status 0 and also from your attempt to control it.

matrix1000, 2018-09-18:
Status0:
`
10:55:58 CMD: status0
10:55:58 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:56:00 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:56:00 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"","RestartReason":"Power on","Uptime":"0T00:07:11","StartupUTC":"2018-09-18T08:48:49","Sleep":0,"BootCount":13,"SaveCount":43,"SaveAddress":"F7000"}}
10:56:00 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1.4","BuildDateTime":"2018-09-17T15:18:55","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:56:00 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["...",""],"TelePeriod":300,"SetOption":["00008001","55818000","00000001"]}}
10:56:00 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F8AA584","000003A0","21B617CE","000002C0"]}}
10:56:00 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-4476","IPAddress":"192.168.0.242","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.5","Mac":"B4:E6:2D:42:51:7C","Webserver":2,"WifiConfig":5}}
10:56:00 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 18 08:56:00 2018","Local":"Tue Sep 18 10:56:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"06:53","Sunset":"19:21"}}
10:56:00 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:56:00 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-09-18T10:56:00","ENERGY":{"Total":0.001,"Yesterday":0.000,"Today":0.001,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
10:56:00 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-09-18T10:56:00","Uptime":"0T00:07:11","Vcc":3.502,"POWER1":"OFF","POWER2":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"...","BSSId":"...","Channel":6,"RSSI":84}}}
`

Frogmore42, 2018-09-18:
Even when you control it from the web UI there will be logging in the console.  You can switch back to the console view or use two browser windows.

Also, show the module configuration screen.

matrix1000, 2018-09-18:
Sorry, I don't quite understand what I'm supposed to do.

My modified Template at the top.
And  GPIO02 is changed to PWM1i in the config page. 
GPIO1 and GPIO3 are not configured.

matrix1000, 2018-09-18:
Ok, new try. 

I have changed the template again. Now gpio00 is user editable.
GPIO00 -> GPIO_USER
I've uploaded the new Firmware to the sp111.

Now I have configured (via Webinterface): 
GPIO0 (red LED) -> led1i
GPIO2(blue LED) -> Relay2i

If I now turn on the blue LED the red turns on too (some milliseconds later). 
(by the way: console -> ledstate -> 1)
Because I want the red LED turn on, IF relay1 is on. :-)

But why goes led1  (red LED) on too, if I turn Relay 2 (blue LED) to on?

matrix1000, 2018-09-18:
It shows only my on off commands:
```
20:40:58 RSL: RESULT = {"POWER2":"ON"}
20:40:58 RSL: POWER2 = ON
20:41:40 RSL: RESULT = {"POWER2":"OFF"}
20:41:40 RSL: POWER2 = OFF
```

Frogmore42, 2018-09-19:
That would indicate that perhaps there is a hardware connection between the LEDs.

matrix1000, 2018-09-19:
No, if I put LED blue to relay2i and LED red to relay3i, than I can switch them individually.

Frogmore42, 2018-09-19:
Well, it seems you have a solution then.

matrix1000, 2018-09-19:
Not really, I would like to have the red led as led1 (status led) for the relay1. But not for relay2. I am looking since same days to the code but until now I can not find the part which is responsible for this behavior.

Frogmore42, 2018-09-19:
It looks like the LED feature does not behave the way you expect.  I am not sure what it is really supposed to do.  Since there are multiple LED definitions one might think that they should work as a channel, i.e. LED1 reflects the state of Relay1 and LED2 reflects the state of Relay2. 

https://github.com/arendst/Sonoff-Tasmota/issues/1227 says the LED will reflect the state of the OR of the relays

Most of the Sonoff devices have hard wired LEDs for the relay, so they already have that control.

matrix1000, 2018-09-19:
Yes, #1227 is exactly the same problem. 
How can I find the responsible code part, to change it?

ascillato, 2018-09-19:
> if I put LED blue to relay2i and LED red to relay3i, than I can switch them individually.

Great! now you can make the behaviour you want using rules.

closed 2018-09-19 by ascillato2

matrix1000, 2018-09-19:
hm, ok, this is a other way as I expected. Ok, I will think about it. 
Originally I did not want to do without the flashing of the WLAN registration. 
I thought I can use the function of LED1, which it is a little bit more than state of Relay1.
OK. Thanks.

Frogmore42, 2018-09-19:
Rules to the rescue again.  But, I am unsure why there are multiple LEDn positions available if they all do the same thing.

weini22, 2019-11-02:
Just for the records: I was able to get the same scenario working with v6.7.1 by assigning PWM1i to GPIO02 using sonoff.bin (no individual build necessary).

matrix1000, 2019-11-02:
Inbetween I found a perfect solution too. :-)
# Template
{"NAME":"Gosund SP111","GPIO":[56,255,46,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
# Console
ledmask 0xFFFD