arendst/Tasmota #3837: Support for generic Rx sensor

afaucogney, 2018-09-18:
**Describe the solution you'd like**
I would like to connect my connected energy counter (linky) to Tasmota. We can find on the web adaptative circuit to adapt signal outing from linky to a serial signal compatible with Rx of ESP2286  or Raspebery....

Here is a version you can order : https://hallard.me/pitinfov12-light/
Or here is DIY version : http://sarakha63-domotique.fr/nodemcu-teleinformation-wifi/

Is there already a way to connect such circuit to Tasmota, inside a Sonoff Basic, and configure it to the sensor input to automatically sending teleinfo TTL/UART message over MQTT ?

**Describe alternatives you've considered**
I assume we can create a specific sensor for this device, but as this is not of the shelf, it may not be accepted. A generic UART to Mqtt gateway would easily allow to get the data and perform the data functionnal parser over the home automation solution.

**Additional context**
_Add any other context or screenshots about the feature request here._

lalo-uy, 2018-09-18:
Check the Wiki:

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#serial-bridge

closed 2018-09-18 by ascillato2

afaucogney, 2018-09-25:
Hello @lalo-uy, thanks for giving me the link, there are so many thing here, that is sometimes not so easy to find what we are looking for.

- So I plugged the circuit to GPIO3 of the sonoff with tasmota. https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic
- The GPIO3 configuration has been connected to SerBr Rx 
- I setup the baudrate to 1200 
- How can I setup the following configuration

Configuration : 1200 bps, 7 data bits, even parity, 1 stop bit
Split input : chr 0x3 

Info may be found here

http://hallard.me/pitinfo/

Bry50, 2019-02-17:
Hi
I too would like to connect linky to a nearby tasmota flashed sonoff.
Did you get this to work ?

Bry50, 2019-03-15:
I couldn't find the exact code you mentioned.
I changed SERIAL_8 to SERIAL_7 in the following, but i only receive the last character of a message when using 7 data bit transmission from termite. Strangely, using 8 bit transmission still receives the whole message.
I'm no programming expert ... can you please tell me if the change I made should work. If so I'll test more.

bool TasmotaSerial::begin(long speed, int stop_bits) {
  m_stop_bits = ((stop_bits -1) &1) +1;
  if (m_hardserial) {
    Serial.flush();
    if (2 == m_stop_bits) {
      Serial.begin(speed, SERIAL_8N2);
    } else {
      Serial.begin(speed, SERIAL_8N1);
    }
    if (m_hardswap) {
      Serial.swap();
    }

Bry50, 2019-03-15:
Sorry.
It is not totally clear
I changed to
Serial.begin(speed, SERIAL_7E2);
} else {
Serial.begin(speed, SERIAL_7E1);

hallard, 2020-06-15:
I'm opening this case because I did not succeded to open tasmota serial with 7E1 in the new teleinfo energy module. Since I'm using hardware module RX pin I needed to do this to get it working, once TasmotaSerial opened, use `SetSerialConfig(TS_SERIAL_7E1);`to get correct configuration. Not sure it will works with ESP32.

My issue is why using `TInfoSerial->begin(TINFO_SPEED, TS_SERIAL_7E1)` the serial in console is configured into 8N1 and I need to use `SetSerialConfig` justt after init to avoid that? Did I missed something ?

See on my PR #8695 

```cpp

        TasmotaSerial *TInfoSerial = nullptr;

        //  here using  SERIAL_7E1 or  TS_SERIAL_7E1 does not works
        if (TInfoSerial->begin(TINFO_SPEED, TS_SERIAL_7E1))
        {
            // This is a dirty hack, looks like begin does not take into account
            // the TS_SERIAL_7E1 configuration so on ESP8266 this is 
            // working only on Serial RX pin (Hardware Serial) for now
            SetSerialConfig(TS_SERIAL_7E1);

            if (TInfoSerial->hardwareSerial()) {
                ClaimSerial();
            }
            TInfoSerial->setTimeout(TINFO_READ_TIMEOUT);

            // Init teleinfo
            tinfo.init();
            tinfo_found = true;
            AddLog_P2(LOG_LEVEL_INFO, PSTR("TIC: Ready"));
        }

```

s-hadinger, 2020-06-15:
@hallard Why bother? Since you to RX only, just keep the default 8N1, and discard the parity bit with `byte & 0x7F`. The parity bit will show as Most Significant Bit (MSB).

hallard, 2020-06-15:
humm, need to be tested but this teleinfo is very sensible and had lot os issue without 7E1 and since HW serial can do parity check, I strongly prefer using it.

Anyway, I solved the issue adding a way to configure serial as we need from Even to Odd, 5 to 8 databits with 4 lines of code, I'll submit the PR ASAP.

Bry50, 2020-06-15:
My use case was :
1) A Sonoff 4 Ch pro Sonoff in the electrical panel next to the Linky counter. The Sonoff was reflashed with Tasmota. It controls electrical radiators ( fil pilot noir) and hot water.
2) A home made optocopler between the Linky and serial pins of the Tasmota.
3) I needed to modify the Tasmota code to handle the 7E1 serial protocol. 
4) Tasmota transmitted the serial data automatically to my MQTT host.
5) I used a Home Assistant sensor template to extract the data I wanted. 
If  power exceeded the subscribed maximum, my intention was to send MQTT message back to stop heating.

This worked OK untill my poorly home made optocopler broke. 
I was going to buy one of your Tindie couplers, but they went out of stock. ( Back in stock now but I haven't got round to restarting the project)

I thought that Tasmota could now read 7E1 as standard, but haven't looked further.