arendst/Tasmota #3838: Unable to access web interface when powered by AC

JohnOH, 2018-09-18:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : OTA update from website
- [x] Hardware used : Pow 2
- [x] Provide the output of command ``status 0`` : Unable to access web interface
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
I am new to Sonoff so there could be newbie issues. I have loaded up the latest firmware using serial connected via VTDI to my MacBook. All appears to work fine while connected to the FTDI with perhaps one odd symptom. The blue led does not flash regularly it stays off for longish periods but can be forced to blink by toggling the relay. It then stops flashing again until the next toggle. When I connect up a 2m AC cable to the inputs E/L/N without any outputs connected and connect with AC power the web interface no longer responds. The unit is logging to MQTT and the blue led flashes irregularly. The same symptoms happen on both of my new POW 2 units. 

It feels like some interrupt issue, any help would be most welcome.

JohnOH, 2018-09-18:
I have since connected a 60W bulb to the output pins, L/N. The symptoms reported above have not changed.

andrethomas, 2018-09-18:
Are you sure you did not mix the AC in with AC out?

ascillato, 2018-09-18:
> The blue led does not flash regularly it stays off for longish periods but can be forced to blink by toggling the relay.

As designed

> The unit is logging to MQTT and the blue led flashes irregularly.

As designed.

ascillato, 2018-09-18:
> Unable to access web interface when powered by AC

The power regulator seems to be faulty

JohnOH, 2018-09-18:
I have two units exhibiting the the same condition. The units appeared to operate normally when initially tested using EWelink.

JohnOH, 2018-09-18:
I have just noted the following from the MQTT log:
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
tele/sonoff/LWT Offline
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)
tele/sonoff/LWT Offline
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)

Does this indicate the Wifi dropping perhaps?

ascillato, 2018-09-18:
2 units would be weird to be faulty. you are right.

yes, that means wifi disconnections

JohnOH, 2018-09-18:
The units are also 4KB flash units.

ascillato, 2018-09-18:
> The units are also 4KB flash units.

What that means?

JohnOH, 2018-09-18:
The units have 4MB flash chips - loaded your standard 1MB code.

andrethomas, 2018-09-18:
> The units have 4MB flash chips - loaded your standard 1MB code.

That is fine, makes no difference it will still work.

andrethomas, 2018-09-18:
Is your device a POW or POW R2 ?

andrethomas, 2018-09-18:
Also, did you try to configure wifi from the console using command
`backlog ssid1 accesspointname ; password1 wifipassword`

ascillato, 2018-09-18:
http://thehackbox.org/tasmota/

Sym-Link, 2018-09-18:
I had the same issue with 3 out of 14 of my Sonoffs with 6.2.1. The troublesome ones would connect to wifi right away if I flashed back down to 6.1.1. Also, on 6.2.1 if they are powed by 3.3 dc the connect just fine, just on AC does it not connect as title states

andrethomas, 2018-09-18:
There was a batch of 6.2.1 ( < 9pm on 9 Sept I think) which still had wifi issues but those were sorted and reposted by Theo

JohnOH, 2018-09-18:
My units are Pow rev2. I didn't try POW settings, just POW2. I configured the wifi using a web browser wifi connected directly to the unit - whilst powered by the FTDI.

JohnOH, 2018-09-18:
@Sym-Link interesting similarities. Are you still running them on 6.1.1?

Jason2866, 2018-09-18:
Erase the pow completly with esptool. Fragments left behind can do weired things.
After that flash a prebuild bin from here.

JohnOH, 2018-09-18:
@Jason2866 do you have a link to a prebuild bin in English?

Jason2866, 2018-09-18:
https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.2.1/sonoff.bin

JohnOH, 2018-09-18:
Johns-MacBook-Pro:6.1.1 john$ esptool.py --port /dev/cu.usbserial-A9I55BZJ  erase_flash
esptool.py v2.5.0
Serial port /dev/cu.usbserial-A9I55BZJ
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:b1:e4:70
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.1s
Hard resetting via RTS pin...
Johns-MacBook-Pro:Tasmota john$ cd 6.2.1/
Johns-MacBook-Pro:6.2.1 john$ ls
sonoff.bin
Johns-MacBook-Pro:6.2.1 john$ /Users/johare/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13/esptool -vv -cd none -cb 115200 -cp /dev/cu.usbserial-A9I55BZJ -ca 0x00000 -cf sonoff.bin 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A9I55BZJ
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A9I55BZJ at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 482512 bytes from sonoff.bin to flash at 0x00000000
	erasing flash
	size: 075cd0 address: 000000
	first_sector_index: 0
	total_sector_count: 118
	head_sector_count: 16
	adjusted_sector_count: 102
	erase_size: 066000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 1000
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 16% ]
................................................................................ [ 33% ]
................................................................................ [ 50% ]
................................................................................ [ 67% ]
................................................................................ [ 84% ]
........................................................................         [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader

No change I'm afraid.

andrethomas, 2018-09-18:
Can you get a output from command
status 0
on serial?

JohnOH, 2018-09-18:
Running on FTDI it should be possible - is that what you want?

JohnOH, 2018-09-18:
I will do it first thing in the morning.

Thanks everyone for taking the time to help.

andrethomas, 2018-09-18:
> Running on FTDI it should be possible - is that what you want?

Yup, might give us insight on what the configuration is and possible keep an eye on it whilst it is open to see if it maybe reboots and hits at a reboot reason... any output from serial could be useful.

JohnOH, 2018-09-18:
Here is some serial data from my recent attempt:

00:00:29 WIF: Connect failed with AP incorrect password
00:00:30 WIF: Connect failed with AP incorrect password
00:00:31 WIF: Connect failed with AP incorrect password
00:00:32 WIF: Connect failed with AP incorrect password
00:00:33 WIF: Connect failed with AP incorrect password
00:00:34 WIF: Connect failed with AP incorrect password
00:00:35 WIF: Connect failed with AP incorrect password
00:00:36 WIF: Connect failed with AP incorrect password
00:00:37 WIF: Connect failed with AP incorrect password
00:00:47 WIF: Connect failed with AP timeout
00:00:48 WIF: Connecting to AP1 it-aide.co.uk in mode 11N as sonoff-1136...
00:00:58 WIF: Connect failed with AP timeout
00:01:02 WIF: Connect failed with AP incorrect password
00:01:03 WIF: Connect failed with AP incorrect password
00:01:04 WIF: Connect failed with AP incorrect password
00:01:05 WIF: Connect failed with AP incorrect password
00:01:06 WIF: Connect failed with AP incorrect password
00:01:07 WIF: Connect failed with AP incorrect password
00:01:17 WIF: Connect failed with AP timeout
00:01:18 WIF: Connecting to AP1 it-aide.co.uk in mode 11N as sonoff-1136...
00:01:28 WIF: Connect failed with AP incorrect password
00:01:29 WIF: Connect failed with AP incorrect password
00:01:30 WIF: Connect failed with AP incorrect password
00:01:31 WIF: Connect failed with AP incorrect password
00:01:32 WIF: Connect failed with AP incorrect password
00:01:33 WIF: Connecting to AP1 it-aide.co.uk in mode 11N as sonoff-1136...
00:01:43 WIF: Connect failed with AP timeout
00:01:44 WIF: Connected
00:01:44 DNS: Initialized
00:01:44 HTP: Web server active on sonoff-1136.local with IP address 192.168.1.45
00:01:44 MQT: Attempting connection...
00:01:45 DNS: Query done. MQTT services found 0
00:01:45 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:01:56 MQT: Attempting connection...
00:01:57 DNS: Query done. MQTT services found 0
00:01:57 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:02:00 RSL: tele/sonoff/UPTIME = {"Time":"1970-01-01T00:02:00","Uptime":"0T00:01:59"}
21:44:13 MQT: Attempting connection...
21:44:14 DNS: Query done. MQTT services found 0
21:44:14 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:44:16 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
21:44:16 RSL: stat/sonoff/POWER = ON
21:44:18 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
21:44:18 RSL: stat/sonoff/POWER = OFF
21:44:25 MQT: Attempting connection...
21:44:26 DNS: Query done. MQTT services found 0
21:44:26 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:44:36 MQT: Attempting connection...
21:44:37 DNS: Query done. MQTT services found 0
21:44:37 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:44:48 MQT: Attempting connection...
21:44:49 DNS: Query done. MQTT services found 0
21:44:49 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:00 MQT: Attempting connection...
21:45:01 DNS: Query done. MQTT services found 0
21:45:01 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:12 MQT: Attempting connection...
21:45:13 DNS: Query done. MQTT services found 0
21:45:13 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:23 MQT: Attempting connection...
21:45:24 DNS: Query done. MQTT services found 0
21:45:24 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:35 MQT: Attempting connection...
21:45:36 DNS: Query done. MQTT services found 0
21:45:36 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:47 MQT: Attempting connection...
21:45:48 DNS: Query done. MQTT services found 0
21:45:48 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:45:59 MQT: Attempting connection...
21:46:00 DNS: Query done. MQTT services found 0
21:46:00 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:46:10 MQT: Attempting connection...
21:46:11 DNS: Query done. MQTT services found 0
21:46:11 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:46:22 MQT: Attempting connection...
21:46:23 DNS: Query done. MQTT services found 0
21:46:23 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:46:34 MQT: Attempting connection...
21:46:35 DNS: Query done. MQTT services found 0
21:46:35 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:46:46 MQT: Attempting connection...
21:46:47 DNS: Query done. MQTT services found 0
21:46:47 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:46:57 MQT: Attempting connection...
21:46:58 DNS: Query done. MQTT services found 0
21:46:58 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
21:47:01 MQT: MqttHost 192.168.1.36, MqttPort 1883, MqttClient DVES_%06X, MqttUser DVES_USER, MqttPassword DVES_PASS, Topic sonoff, FullTopic %prefix%/%topic%/
21:47:03 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_B1E470, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 WIF: Connecting to AP1 it-aide.co.uk in mode 11N as sonoff-1136...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-1136.local with IP address 192.168.1.45
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff/POWER = 
00:00:05 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1","FallbackTopic":"DVES_B1E470","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1136","IPAddress":"192.168.1.45"}
00:00:05 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/sonoff/POWER = OFF
21:47:18 MQT: tele/sonoff/STATE = {"Time":"2018-09-18T21:47:17","Uptime":"0T00:00:14","Vcc":3.229,"POWER":"OFF","Wifi":{"AP":1,"SSId":"it-aide.co.uk","RSSI":80,"APMac":"90:21:06:7A:07:72"}}

andrethomas, 2018-09-18:
Issue command
`module 43`
This will cause a reboot, copy and paste everything that comes out until where it fails.
Then when it boots again issue command
`status 0`
and copy all the output back here

Sym-Link, 2018-09-18:
Yes, on those 3 that I had issues with I went back to 6.1.1. I was doing all of this today so I am sure I got the latest pre-built build for 6.2.1

JohnOH, 2018-09-19:
A little strange, module 43 did trigger a reboot and set the device type to POW2. However this appears to have disabled most of the logging to serial. I still get this:

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 APP: Set Baudrate to 4800
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 [... cut ...]
07:28:07 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:28:07 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-09-19T07:28:07","ENERGY":{"Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
07:28:08 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-09-19T07:28:07","Uptime":"0T00:04:47","Vcc":3.235,"POWER":"OFF","Wifi":{"AP":1,"SSId":"it-aide.co.uk","RSSI":84,"APMac":"90:21:06:7A:07:72"}}}

Frogmore42, 2018-09-19:
POW2 uses the serial port to talk to the energy monitoring chip so I believe that is expected.  It also looks like it is working properly.  Also, you should NEVER connect a POW or POW2 to mains and serial at the same time.  There will be sparks and smoke, at least, if you do.

JohnOH, 2018-09-19:
The status 0 posted earlier was when the unit was powered by FTDI. When moved to AC the web interface is not available. Thanks for the insight on the serial when in module 43 mode.

andrethomas, 2018-09-19:
> The status 0 posted earlier was when the unit was powered by FTDI. When moved to AC the web interface is not available. Thanks for the insight on the serial when in module 43 mode.

So the web interface is working now from AC ?

JohnOH, 2018-09-19:
No it isn't. When connected to my MAC via FTDI the unit works on wifi. When the FTDI is disconnected and the AC used it is not possible to connect to the web interface.

andrethomas, 2018-09-19:
Can you send a high resolution photo of the bottom of the pc board?

JohnOH, 2018-09-19:
[image]

JohnOH, 2018-09-19:
I share your view that there is probably an issue with the 3v3 rail. As an aside I have loaded 6.1.1 and am experiencing similar symptoms. However now I can provoke some wifi activity by using the button to toggle the output. During one of these active periods I pulled the information page, see attached pdf. 
[Sonoff1 - Information.pdf](https://github.com/arendst/Sonoff-Tasmota/files/2396540/Sonoff1.-.Information.pdf)

JohnOH, 2018-09-19:
> Well if it works for 6.1.1 then the 3V3 rail is probably OK.

A little strong, it doesn't work but I can get odd packets through by pressing the button. My money is still on the 3v3 rail. I do have a DNS server locally but not sure if is serving anything to the Sonoff. I need to code the MQTT host IP so I guess it isn't being served with that.

Sym-Link, 2018-09-19:
I wouldn't really think it is the rail. My Sonoffs were fine before 6.2.1 and are now fine again by going back to 6.1.1. Like I said I even downgraded my other Sonoffs because 6.1.1 seemed more stable even on the ones that were connecting.

JohnOH, 2018-09-19:
Can you share a link to the 6.1.1 code that you are running?

The version I have tried shows the same problem.

Sym-Link, 2018-09-19:
I am using this - [6.1.1](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v6.1.1) but you are right that even on 6.1.1 they do drop off here and there but come back. See my screen shot. The switches in red are unreachable at the time of the snapshot but they come right back

[image]

JohnOH, 2018-09-19:
Does anyone have a copy of the original EWelink code for the POW R2?

Frogmore42, 2018-09-19:
On wierd things that happen with networking with the various libraries.  I was pretty sure that one of my NodeMCU devices was bad.  So sure that I started buy Wemos D1 minis instead. The device would reset every hour or so, but appeared to be working fine until it wasn't.  It still worked well enough for MQTT though so I didn't do anything about it.  This weekend I spent some time on it.  I noticed that it had the problem of not responding to pings sometimes, even while the web UI was responding just fine.  I would have expected the opposite, with ping (once a second, standard size packet) to be the more reliable.  I then noticed it was running core 2.4.0.  I updated the firmware to what was current then with core 2.4.2 high bandwidth.  Since then, the device has been a picture of health and has been up for days.

Now, this case certainly seems like it could be caused by a somewhat defective 3v3 regulator.

On that device I have an ADS1115 ADC for some ratiometric pressure sensors.  So I had one channel on the 3v3 regulator.  That was nowhere near constant.  I even put an oscilloscope on it and it looked like it might be defective.  I then switched to using a separate 3v3 regulator for the sensors.  That was much better.  But, over time I could see that the new regulator was also not constant.  Since I got 5 of them, I swapped it out.  The new one on the sensors was rock solid (+/-1mV) over a long period.

So, both problems could true.  Your devices have marginally functional regulators and there is something in your network that is causing it to behave badly.

I have been happy with 2.4.2 with lwip2 high bandwidth.  But, my environment is not particularly challenging for WiFi.  My barnyard has been the most problematic because of distance and signal strength.  I had to change out the router to get stuff to work.  Even with that they report less than 50% and go as low as 30%.  Environments that have many competing networks especially on the same or adjacent channels can be really hard for the Wi-Fi code to handle.

So, if you haven't already, try using core 2.4.2 high bandwidth and see if it improves things.  If not, I am not sure what will.

JohnOH, 2018-09-19:
My wifi environment is quite challenging I think, a dozen or so networks. The router used by the Sonoff is a dual radio 2.4g & 5g. Do you have a link for a prebuilt bin file with 2.4.2 high bandwidth.

andrethomas, 2018-09-19:
> Do you have a link for a prebuilt bin file with 2.4.2 high bandwidth.

Here you go 
[sonoff_2_4_2.zip](https://github.com/arendst/Sonoff-Tasmota/files/2397948/sonoff_2_4_2.zip)

JohnOH, 2018-09-19:
It may be a code issue after all: I have loaded up ESPURNA 1.13.2 to one unit and the web access is rock solid...

andrethomas, 2018-09-19:
You mean the 2.4.2 bin works?

ascillato, 2018-09-19:
@JohnOH 

Hi,

have you managed to solve your issue with esplib v2.4.2 ?

JohnOH, 2018-09-19:
> You mean the 2.4.2 bin works?

I have not tried 2.4.2 yet, probably in the morning now.

JohnOH, 2018-09-19:
I have loaded up ESPURNA 1.13.2 from https://github.com/xoseperez/espurna to one of my units. This is another hack for the Sonoff devices. It is working flawlessly at the moment connected to the same network and in the same position as was causing problems with TASMOTA. 

I wonder if ESPURNA uses a different wireless library to TASMOTA.

huramentzefix, 2018-09-19:
there is something fundamentally wrong with tasmota.
at least since may.
See my issue #2999 from may and all the other issues and complains of others.
it's never been stable, always wifi issues, several reboots per day, clicking of the relay, slow pings, ...
with esp-easy I have ping times of 1-2ms when I am directly connected. I like tasmota because the integration of alexa is easier, ssl mqtt and much more.

it all works with esp-easy as well but it's more work to do it by hand.
I had nothing but issues with the wireless functionality and stability on more than 6 devices.
there is a lot of smart people contributing to the tasmota project and one will eventually catch the cause one day.
I read something yesterday in the 6.2.1 version that someone discovered some error with the correct bit alignment. But the image wouldn't allow any wifi connectivity.
probably I will be back once it's being resolved. till then I have to put more time in esp-easy, espurna, ...

ascillato2, 2018-09-19:
Closing this issue as there were hateful comments. Please, @JohnOH if you need further assistant, contact us in Discord Chat

closed 2018-09-19 by ascillato2