arendst/Tasmota #3839: Sono basic can't connect to AP

SSkull995, 2018-09-18:
Hi all! 
I have quite an annoying issue with my 2 sonoff basics. As soon as I flash the tasmota firmware, the sonoff can't connect to my wifi. I tried every firmware version until 5.12, nothing works. I even change the CFG_HOLDER, nothing.
Sometimes it connects, then disconnects and same issue as above, I attach the code I get with the master version, which is the only one I can connect to my wifi:

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_A70A11, GroupTopic sonoffs)
 Version 6.2.1.4-2_3_0
00:00:00 WIF: Connecting to AP1 Home WiFi in mode 11N as sonoff-2577...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-2577.local with IP address 192.168.21.148
00:00:04 MQT: Attempting connection...
00:00:05 DNS: Query done. MQTT services found 0
00:00:05 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
00:00:25 WIF: Connect failed as AP cannot be reached
00:00:25 WIF: Connecting to AP1 Home WiFi in mode 11N as sonoff-2577...
00:00:32 WIF: Connect failed as AP cannot be reached
00:00:32 WIF: Serial active for 3 minutes

After serial everything restarts or the sonoff just stalls. 
I have other 3 sonoff, never found this problem before. The password is correct as the wifi. I also got a message after a while stating that the RSSI is 100. 
Really don't know what to do, can you help me please?

ascillato, 2018-09-18:
Hi,

Seems that your version has been modified from the standard in the Tasmota repository

the line
00:00:04 HTP: Web server active on sonoff-2577.local with IP address 192.168.21.148
is not what is in the actual code.

If you have modified Tasmota there isn't nothing we can do.

Jason2866, 2018-09-18:
Erase device completly and use a precompiled firmware.
If it works you have a error in your setup...

SSkull995, 2018-09-18:
@ascillato if there is a modified part of the code is not by my hand. I actually make changes to wifi and CFG stop. 
@Jason2866 I have already erased it with Esptool, actually like 5 times, each time after a different firmware was uploaded. Which firmware version would you suggest me?

SSkull995, 2018-09-18:
of course, thank you both for you replies