arendst/Tasmota #3840: Command to disable relay but still sends MQTT messages

Kiritzai, 2018-09-18:
Hi!

First off I want to thank you ( Theo Arends ) for this wonderful software you've been providing us.
This is a feature request so lets get down down to my story.

I've got an doorbell that is connected with 2 wires to my Sonoff Basic ( GPIO14 and GND ), GPIO14 module is configured as: **12 switch4** and it's working perfectly and to top it off, if the doorbell rings I get an telegram message with a snapshot of the person at the door, pretty handy!

Now occasionally my doorbell will ring late in the evening or very early in the morning. It's pretty loud and my cats will get scared when the doorbell rings, so what I thought is to disable only the GPIO12 relay ( with a command ) on a certain time and enable it again later on the next day, i'm unfortunately unable to find such solution.

I've seen people suggest ButtonTopic or PowerOnState in which case won't work for me, I still want to send MQTT messages.

It's possible that I overlooked something but I'm unable to find this solution so far.

ascillato, 2018-09-18:
Hi, you can achieve this behaviour using the rules feature.

Please, check the examples in the wiki.

Thanks.

Kiritzai, 2018-09-18:
Thank you!

Although I wish I could've done this with my domoticz rules ( DzVents ) instead of rules on my device itself.

Kiritzai, 2018-09-18:
Not sure why you've closed it, i'm requesting a new feature which clearly isn't available 'yet' in tasmota.
Feature request: turning off relay with a command that can be changed through URL.

ascillato, 2018-09-18:
Hi,

It is already available. You need to use rules in order to make a logic enable/disable for relays. There are several examples in the wiki about the use of rules.

Kiritzai, 2018-09-18:
That wasn't very clear for me.
Though I didn't found it on the Tasmota wiki rules what I was looking for.
I did found it on ESPEasy wiki that its called HTTP Call, so i'm not sure it will work for tasmota.

So does it work for tasmota?
http://<someip>/cm?cmnd=event,someevent

Source: ESPeasy

### HTTP call

When you enter this first command with the correct IP address in the URL of your browser:
```
 http://<espeasyip>/control?cmd=event,startwatering
 http://<espeasyip>/control?cmd=event,stopwatering
```

And have this rule in the addressed ESP:

```
On startwatering do 
  gpio,12,1 //start watering (open valve)
  timerSet,1,600 //timer 1 set for 10 minutes
endon
 
On stopwatering do 
  timerSet,1,0 //timer 1 set to halt, used to stop watering before the timer ends!
  gpio,12,0 //stop watering (close valve)
endon
 
On Rules#Timer=1 do  
   gpio,12,0 //stop watering (close valve)
endOn
```

Provided that you also have the valve etc, the plants will be happy.

Kiritzai, 2018-09-19:
Seemed the rules didn't work for me at all.

Simple solution for me was ( if anybody is interested ):
GPIO set on **12 switch4** ( I'm using a **PUSHBUTTON** as doorbell )

I've set de switchmode to 2 for switch 4 ( _Set switch mode to inverted FOLLOW (0 = On, 1 = Off)_ )
```
switchmode4 2
```

### Relay off/disable
To disable Relay GPIO12, type the following in console
```
switchtopic4 1
```
or HTTP Call/Command:
```
http://<tasmotaIP>/cm?cmnd=switchtopic4%201
or
http://<tasmotaIP>/cm?&user=<username>&password=<password>&cmnd=switchtopic4%201
```

### Relay on/enable
To enable Relay GPIO12, type the following in console
```
switchtopic4 0
```
or HTTP Call/Command
```
http://<tasmotaIP>/cm?cmnd=switchtopic4%200
or
http://<tasmotaIP>/cm?&user=<username>&password=<password>&cmnd=switchtopic4%200
```

Now I can keep it centralized by using this in DzVents in Domoticz.
Just wanted to share this, cause I wasn't the only one that was searching for this.

gericho, 2019-07-25:
Thank you to publish your solution but on my Sonoff doesn't work

What it is working:
Sonoff v6.60
- configuration/module set it to "generic" save it reboots
- once rebooted, return in the same window, and be sure all is set to generic, all the GPIO to none(0)
- go to configure template, and set Based On / Sonoff Basic (1)
- GPIO0 Button1(17)
- GPIO1->GPIO4 User 255
- GPIO12-GPIO14 User 255 (this will also disable the led permanently
- Save it reboots
- Go to the Console
- `PowerOnState 0 `(optional set the relay of at boot)
- `Switchmode3 0` (set the switchmode3 puch button not inverted to toggle)
- `Switchmode1 3` (set the MQTT command on ON and OFF)
- Now press the phisical button on the sonoff
- verify MQTT messages works as follow

> - 08:45:26 MQT: stat/yourtopic/RESULT = {"POWER":"ON"}
> - 08:45:26 MQT: stat/yourtopic/POWER = ON
> - 08:45:28 MQT: stat/yourtopic/RESULT = {"POWER":"OFF"}
> - 08:45:28 MQT: stat/yourtopic/POWER = OFF

- put your ear on the Sonoff and be sure the relay stays off

- set an HA automation like this:

```
- alias: 'Push to ON'
  hide_entity: true
  initial_state: true
  trigger:
    platform: mqtt
    topic: stat/yourtopic/POWER
    payload: 'ON'
  action:
    - service: light.turn_on
      entity_id: light.kitchen

- alias: 'Push to OFF'
  hide_entity: true
  initial_state: true
  trigger:
    platform: mqtt
    topic: stat/yourtopic/POWER
    payload: 'OFF'
  action:
    - service: light.turn_off
      entity_id: light.kitchen
```